ArduPilot/apm_planner #728: Add button for MNT_MODE on flight data screen

sgofferj, 2015-06-05:
I have been scratching my head today on a filming trip, wondering why I can't suddenly control the gimbal via the TX. I quickly rewired my IRIS for CH5 passthru in the field. At home I figured it out. When setting up the gimbal, I used the testing method described in the wiki and used the "Point camera here function" which set MNT_MODE to 4 (GPS). The parameter is persistent over a reboot and hence the RC-control didn't work.
Please add a button to conveniently choose MNT_MODE from the flight data screen to allow for easy transitions.

billbonney, 2015-06-05:
This is not an ideal situation, It would be much better if the message was just obeyed, and the mount_mode was not changed. Let's check with the ArduPilot team on abetter behaviour here, as this is not the best currently.
It doesn't make sense to add the mount_configs to the Flight Data page. Maybe a 'Camera Control' widget would be helpful?

sgofferj, 2015-06-05:
Jup, a widget would be great. I did open an issue for copter on this also: https://github.com/diydrones/ardupilot/issues/2376

rmackay9, 2015-06-06:
I think we've fixed the behaviour in AC3.3 and I've commented on the ardupilot issue linked above.