ArduPilot/apm_planner #729: APM application start problem

katrichnikitos, 2015-06-07:
APM doesn't start, shows cyclically reapeating messages in console (see below).
APM is built from git sources() on Fedora 22(all dependencies are met according to spec file from redhat directory).

$ ./release/apmplanner2 
 INFO 2015-06-07T15:08:18.037 QGCCore::initialize() 
 INFO 2015-06-07T15:08:18.037 Current Build Info 
 INFO 2015-06-07T15:08:18.037 Git Hash: 7934a289278a24977e9d8516a4fff8f7ccf2c761 
 INFO 2015-06-07T15:08:18.037 Git Commit: 2.0.17-61-g7934a28 
 INFO 2015-06-07T15:08:18.037 APPLICATION_NAME: "APM Planner" 
 INFO 2015-06-07T15:08:18.037 APPLICATION_VERSION: "v2.0.17" 
 INFO 2015-06-07T15:08:18.037 APP_PLATFORM: debian32 
 INFO 2015-06-07T15:08:18.037 APP_TYPE: stable 
 INFO 2015-06-07T15:08:18.087 Start Link Manager 
DEBUG 2015-06-07T15:08:18.095 Create MAVLinkDecoder:  MAVLinkDecoder(0x1a72150) 
 INFO 2015-06-07T15:08:18.104 Start UAS Manager 
DEBUG 2015-06-07T15:08:18.384 Creating MainWindow 
 INFO 2015-06-07T15:08:19.067 Creating SubMainWindow:  SubMainWindow(0x1bcfac0) 
 INFO 2015-06-07T15:08:19.067 setting objectName:  "VIEW_MISSION" 
 INFO 2015-06-07T15:08:19.340 Creating SubMainWindow:  SubMainWindow(0x1c7ba80) 
 INFO 2015-06-07T15:08:19.340 setting objectName:  "VIEW_FLIGHT" 
 INFO 2015-06-07T15:08:19.372 Creating SubMainWindow:  SubMainWindow(0x1ccb490) 
 INFO 2015-06-07T15:08:19.372 setting objectName:  "VIEW_HARDWARE_CONFIG" 
DEBUG 2015-06-07T15:08:20.405 fillPortsInfo  
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
DEBUG 2015-06-07T15:08:20.638 Creating MainWindow 
 INFO 2015-06-07T15:08:20.988 Creating SubMainWindow:  SubMainWindow(0x32a4ce0) 
 INFO 2015-06-07T15:08:20.988 setting objectName:  "VIEW_MISSION" 
 INFO 2015-06-07T15:08:21.019 Creating SubMainWindow:  SubMainWindow(0x336b5d0) 
 INFO 2015-06-07T15:08:21.019 setting objectName:  "VIEW_FLIGHT" 
 INFO 2015-06-07T15:08:21.045 Creating SubMainWindow:  SubMainWindow(0x33bda30) 
 INFO 2015-06-07T15:08:21.046 setting objectName:  "VIEW_HARDWARE_CONFIG" 
DEBUG 2015-06-07T15:08:21.268 fillPortsInfo  
DEBUG 2015-06-07T15:08:21.429 Creating MainWindow 
 INFO 2015-06-07T15:08:21.749 Creating SubMainWindow:  SubMainWindow(0x48d9390) 
 INFO 2015-06-07T15:08:21.749 setting objectName:  "VIEW_MISSION" 
 INFO 2015-06-07T15:08:21.779 Creating SubMainWindow:  SubMainWindow(0x49489c0) 
 INFO 2015-06-07T15:08:21.779 setting objectName:  "VIEW_FLIGHT" 
 INFO 2015-06-07T15:08:21.804 Creating SubMainWindow:  SubMainWindow(0x49e8180) 
 INFO 2015-06-07T15:08:21.804 setting objectName:  "VIEW_HARDWARE_CONFIG" 
DEBUG 2015-06-07T15:08:22.066 fillPortsInfo

froller, 2015-06-10:
void ApmFirmwareConfig::hideEvent(QHideEvent *) calls MainWindow::instance() and it repeats infinitely until stack overflow. Endless recursion.

billbonney, 2015-06-10:
This probably because you haven't installed the code to it's destination location and it's failing to load the QML files.
try make install and run from the installed location

froller, 2015-06-11:
[froller@flyingpenguin APMPlanner2]$ ls -R /usr/share/APMPlanner2/qml
/usr/share/APMPlanner2/qml:
ApmToolBar.qml  components  PrimaryFlightDisplayQML.qml  resources

/usr/share/APMPlanner2/qml/components:
AltitudeIndicator.qml  CompassIndicator.qml  HeartbeatDisplay.qml             ModeDisplay.qml     RollPitchIndicator.qml  StatusDisplay.qml           TextButton.qml
Button.qml             DigitalDisplay.qml    InformationOverlayIndicator.qml  PitchIndicator.qml  SpeedIndicator.qml      StatusMessageIndicator.qml

/usr/share/APMPlanner2/qml/resources:
apmplanner  components

/usr/share/APMPlanner2/qml/resources/apmplanner:
toolbar

/usr/share/APMPlanner2/qml/resources/apmplanner/toolbar:
connect.png  disconnect.png  donate.png  flightdata.png  flightplanner.png  hardwareconfig.png  helpwizard.png  light_initialsetup_icon.png  light_tuningconfig_icon.png  simulation.png  softwareconfig.png  terminal.png

/usr/share/APMPlanner2/qml/resources/components:
rollPitchIndicator

/usr/share/APMPlanner2/qml/resources/components/rollPitchIndicator:
artGroundSky.svg  background_old.png  background.png  compassIndicator.svg  compass.svg  crossHair.svg  needle.png  needle_shadow.png  overlay.png  pitchGraticule.png  quit.png  rollGraticule.svg  rollPointer.svg

katrichnikitos, 2015-06-11:
Made "make install". All the same.

katrichnikitos, 2015-06-11:
Have built rpm, installed - the same issue.

katrichnikitos, 2015-06-25:
Hey, dear team,
Do you have any ideas?

froller, 2015-06-26:
Works for me in 2.0.18-rc2

dsborets, 2015-06-27:
I'm having the same problem:
INFO 2015-06-27T00:07:39.259 QGCCore::initialize() 
 INFO 2015-06-27T00:07:39.259 Current Build Info 
 INFO 2015-06-27T00:07:39.259 Git Hash: 47bd731d16cbe8892200e553d364ebe71a9d30b6 
 INFO 2015-06-27T00:07:39.259 Git Commit: 2.0.17-DEV 
 INFO 2015-06-27T00:07:39.259 APPLICATION_NAME: "APM Planner" 
 INFO 2015-06-27T00:07:39.259 APPLICATION_VERSION: "v2.0.17" 
 INFO 2015-06-27T00:07:39.260 APP_PLATFORM: ubuntu_trusty64 
 INFO 2015-06-27T00:07:39.260 APP_TYPE: stable 
kf5.kiconthemes: "Theme tree: (Breeze)"
 INFO 2015-06-27T00:07:39.306 Start Link Manager 
DEBUG 2015-06-27T00:07:39.306 Create MAVLinkDecoder:  MAVLinkDecoder(0x1d17b80) 
 INFO 2015-06-27T00:07:39.307 Start UAS Manager 
DEBUG 2015-06-27T00:07:39.379 Creating MainWindow 
 INFO 2015-06-27T00:07:39.450 Creating SubMainWindow:  SubMainWindow(0x1dafbc0) 
 INFO 2015-06-27T00:07:39.450 setting objectName:  "VIEW_MISSION" 
 INFO 2015-06-27T00:07:39.475 Creating SubMainWindow:  SubMainWindow(0x1ebb2f0) 
 INFO 2015-06-27T00:07:39.475 setting objectName:  "VIEW_FLIGHT" 
 INFO 2015-06-27T00:07:39.488 Creating SubMainWindow:  SubMainWindow(0x1f14170) 
 INFO 2015-06-27T00:07:39.488 setting objectName:  "VIEW_HARDWARE_CONFIG" 
DEBUG 2015-06-27T00:07:39.639 fillPortsInfo
.................

kubuntu 15.04 3.19.0-21-generic 

Any ideas how to fix it?

katrichnikitos, 2015-06-28:
Thanks, @froller. Share it, please. However it's better to complain how have you built it.
I tried to build rpm from the latest git source and got the same problem.

froller, 2015-06-29:
@katrichnikitos I've build it as usual with my scripts. Fortunately they're already in repo so you're free to use them.

If you have no rpm-build yet:
```
yes | yum install rpm-build
mkdir -p ~/rpmbuild/SOURCES
mkdir -p ~/rpmbuild/SPECS
```

If you have no source yet:
```
mkdir -p ~/git
pushd ~/git
git clone https://github.com/diydrones/apm_planner
popd
```

Every time you decide you need new rpm
```
pushd ~/git/apm_planner/scripts
./prepare_rpm.pl
./build_rpm.sh
popd
```

froller, 2015-06-29:
@katrichnikitos 

```
[diydrones]
name=DIY Drones - $basearch
failovermethod=priority
baseurl=http://www.froller.net/diydrones/$basearch
#metalink=https://mirrors.fedoraproject.org/metalink?repo=fedora-$releasever&arch=$basearch
enabled=1
metadata_expire=7d
gpgcheck=0
gpgkey=file:///etc/pki/rpm-gpg/RPM-GPG-KEY-diydrones-$basearch
skip_if_unavailable=False

[diydrones-debuginfo]
name=DIY Drones - $basearch - Debug
failovermethod=priority
baseurl=http://www.froller.net/diydrones/$basearch/debug/
#metalink=https://mirrors.fedoraproject.org/metalink?repo=fedora-debug-$releasever&arch=$basearch
enabled=0
#metadata_expire=7d
gpgcheck=0
gpgkey=file:///etc/pki/rpm-gpg/RPM-GPG-KEY-diydrones-$basearch
skip_if_unavailable=False

[diydrones-source]
name=DIY Drones - Source
failovermethod=priority
baseurl=http://www.froller.net/diydrones/SRPMS/
#metalink=https://mirrors.fedoraproject.org/metalink?repo=fedora-source-$releasever&arch=$basearch
enabled=0
#metadata_expire=7d
gpgcheck=0
gpgkey=file:///etc/pki/rpm-gpg/RPM-GPG-KEY-diydrones-$basearch
skip_if_unavailable=False
```

billbonney, 2015-06-29:
I've cherry-picked your changes into master with 54fd6cd 54fd6cd 54fd6cd

froller, 2015-06-29:
Note that I don't perform automated builds. At least for now.

dsborets, 2015-06-29:
I was able to make it working for me by applying the next changes:
```diff
diff --git a/src/ui/MainWindow.cc b/src/ui/MainWindow.cc
index 36317bc..90bf9c5 100644
--- a/src/ui/MainWindow.cc
+++ b/src/ui/MainWindow.cc
@@ -88,9 +88,14 @@ This file is part of the QGROUNDCONTROL project
 MainWindow* MainWindow::instance(QSplashScreen* screen)
 {
     static MainWindow* _instance = 0;
-    if(_instance == 0)
+    static bool initializing = true;
+
+    if(_instance == 0 && initializing)
     {
-        _instance = new MainWindow();
+        initializing = false;
+
+        _instance = new MainWindow();        
+        
         if (screen) connect(_instance, SIGNAL(initStatusChanged(QString)), screen, SLOT(showMessage(QString)));
 
         /* Set the application as parent to ensure that this object
diff --git a/src/ui/configuration/ApmFirmwareConfig.cc b/src/ui/configuration/ApmFirmwareConfig.cc
index 9264d9e..0fff992 100644
--- a/src/ui/configuration/ApmFirmwareConfig.cc
+++ b/src/ui/configuration/ApmFirmwareConfig.cc
@@ -245,7 +245,9 @@ void ApmFirmwareConfig::hideEvent(QHideEvent *)
     m_timer->stop();
     if(ui.stackedWidget->currentIndex() == 0)
     {
-        MainWindow::instance()->toolBar().disableConnectWidget(false);
+        if (MainWindow::instance() != 0) {
+            MainWindow::instance()->toolBar().disableConnectWidget(false);
+        }
     }
     //MainWindow::instance()->toolBar().disableConnectWidget(false);
 }```

katrichnikitos, 2015-06-29:
@froller, thanks a lot. Scripts! That's what i needed. And, yes, i found
after that the repo file. Thanks again.

Nikita

billbonney, 2015-06-29:
@dsborets That strange code, you are guarding against a condition that doesn't exist on my linux builds with xubuntu 15.04 Qt5.4.1 What's changed in the MainWindow() constructor that is recursively calling it?

mrpilot2, 2015-06-29:
@dsborets I'm sorry to say, but I'm pretty sure your code does not change the behaviour at all.

In MainWindow instead of using one variable (_instance) to check if the instance is already created you're now using two variables doing the same thing.

The if statement in ApmFirmwareConfig is also useless, because the  MainWindow::instance() always returns a valid (not null) object.

If your code really changes the behaviour than there are really, really strange side effects in the application that should be fixed instead of doing a workaround in this way.

katrichnikitos, 2015-06-29:
@froller seems the problem is still persist with my own built rpm as well as with yours.
@billbonney maybe the problem is somehow related to QT or libs version? The app was working fine for me in Fedora 21, and it have stopped after updgrade to F22.

dsborets, 2015-06-29:
@billbonney Sorry, I have no idea why it's happening on my machine (BTW my two machines. Just tested on my another laptop with fresh kubuntu installation). 
I'm not Qt developer. I will be happy if you can provide me any suggestion how I can check my Qt installation/configuration etc.

This is my current Qt version:
Using Qt version 5.4.1 in /usr/lib/x86_64-linux-gnu
Kubuntu 15.04 3.19.0-21-generic

dsborets, 2015-06-29:
@mrpilot2 
> In MainWindow instead of using one variable (_instance) to check if the instance is already created you're now using two variables doing the same thing.

Actually not. For some reason (again I'm not Qt dev) during MainWindow object creation, the function 'hideEvent' from ApmFirmwareConfig is calling (base on gdb trace). And in that function we have 'MainWindow::instance()' which is null at that time. By adding extra var 'initializing' we can controll recursively calling '_instance = new MainWindow()' when the MainWindow still is creating.

> The if statement in ApmFirmwareConfig is also useless, because the MainWindow::instance() always returns a valid (not null) object.

You are right if MainWindow object has been successfully created.

billbonney, 2015-06-29:
@mrpilot. It won't return a valid instance if you are in side main windows constructor. It's actually a good example of why using singletons is a bad idea and having proper parent sibling relationships between objects is important.

froller, 2015-06-29:
@katrichnikitos I haven't tested it on F22 yet.

mrpilot2, 2015-06-29:
@dsborets @billbonney 
Thanks for clarification. I'm currently downloading and installing Fedora to reproduce the problem and in the meantime I had a closer look at the code and have found the cycle. So I agree with both of you. 

So the root cause is that the singleton creation is not thread safe. One thread is the main application that calls all the constructors, the other one is the Qt event loop that calls the hideEvent and both are calling MainWindow::instance().  The problem is probably not related to a specific operating system or QT version but to the machine that runs the application. It seems to be a timing issue.

A better solution would be to make the singleton thread safe using a lock. Unfortunately I've never done this using Qt libraries. But it should be possible using a QMutex.

dsborets, 2015-06-29:
@billbonney Yes, you are right. 
But I disagree that the singleton is a bad idea. There are a lot of cases where the singleton is required, but it should be implemented correctly (in Qt probably using mutex)

billbonney, 2015-06-30:
@dsroberts I mean the use of the singletons in APM Planner, as there are many. Singletons are a good idea, and in general, you only need one in an App, which is the App object itself.

@mrpilot there is only one thread in this case. even if you made it thread safe, you would get a lock. The problem is one of ownership, which complicates the initialization routines. The way the UI has been written in APM Planner 2.0 (i.e. QGC which is was based) is that all the views are created at startup. It would be much better it lazily loading of the other screens, and this issue would be avoided.

If anybody wants to improve that part of the code, it would be most welcome :+1:

dsborets, 2015-06-30:
@billbonney Sure, I agree.

martinrunge, 2015-07-20:
I have the same issue on KUbuntu 15.04 / Qt 5.4.1. Exactly like described above, c-tor of MainWindow called recursively by hideEvent.

I noticed an interesting detail:  It does not happen, if I start apmplanner2 with sudo. Logged in as my usual user, but started with sudo it runs outside of my user's session and does not inherit some of the desktop's theme. In that case, also the trace line
*'kf5.kiconthemes: "Theme tree: (Breeze)"'*
from @dsborets is not shown.

This is not as unusual as it might seem: for example Eclipse Luna crashes on KDE based desktops when the GTK theme 'oxygen-gtk' is used. Change the theme and it runs fine.

billbonney, 2015-07-21:
@martinrunge do you have your user added to the dialout group?
`sudo adducer <username> dialout`

martinrunge, 2015-07-21:
Yes, I am member of the dialout group. I also uninstalled modemmanager, but without success.
I assumed a missing permission first, too. If anything works with sudo that doesn't work without, it is most often a permissions problem. But I could not find any "permission denied" with strace and co. 
Then I debugged the startup and saw exactly the problem described by @mrpilot2

billbonney, 2015-07-21:
It works for me on xubuntu 15.04... I'll have to find some time to look at this issue.

martinrunge, 2015-07-21:
depending on the desktop environment used, a different Qt style is active. On my machine (KUbuntu 15.04) the breeze theme is used by Qt apps by default:
./apmplanner2 -style=breeze   #  -> the problem appears
./apmplanner2 -style=fusion    #  -> starts up normally
./apmplanner2 -style=gtk         #  -> starts up normally

maybe the breeze style is not available on xubuntu, should be installable with: 
sudo apt-get install kubuntu-desktop