ArduPilot/apm_planner #730: APM Planner creates the folder ~/apmplanner2 even if the AppData location is changed

pepak, 2015-06-07:
I changed the folder locations of App Data, Log output, tlog output and Parameters, but APM Planner still creates these folders:
~/apmplanner2
~/apmplanner2/dataflashLogs
~/apmplanner2/missions
~/apmplanner2/parameters
~/apmplanner2/tlogs

billbonney, 2015-06-07:
The folder mapping are absolute. Not relative. This is more a 'feature' than a defect. This is so you can for example have logs in another location entirely from the app directory.

Having paths able to be set both relative to the App Directory setting or an absolute path would be ideal.

pepak, 2015-06-07:
For clarification:
I changed the folder locations to:
APP_DATA_DIRECTORY=/home/user/.config/apmplanner2
LOG_DIRECTORY=/home/user/.config/apmplanner2/dataflashLogs
MAVLINK_LOG_DIRECTORY=/home/user/.config/apmplanner2/tlogs
MISSION_DIRECTORY=/home/user/.config/apmplanner2/missions
PARAMETER_DIRECTORY=/home/user/.config/apmplanner2/parameters

APM Planner still creates these empty folders that are not used:
/home/user/apmplanner2
/home/user/apmplanner2/dataflashLogs
/home/user/apmplanner2/missions
/home/user/apmplanner2/parameters
/home/user/apmplanner2/tlogs

During startup, the application should create the folder: /home/user/.config/apmplanner2 (if not exist), not /home/user/apmplanner2. I hope this is understandable, my English is not good.

mrpilot2, 2015-06-13:
Hi, I have fixed this issue: 

The default directory passed when reading the settings file is only created if no other directory is found in the ini file (via the getters)

Furthermore I have added support to change the missions directory via GUI

commit 87d11d37 (2015-06-15): General Settings: Fixes #730 default directory always created

There is no need to create the default directory when the settings are read, because each getter also creates the
directory which is either the default directory or the directory read from the settings file
closed 2015-06-15 by billbonney