ArduPilot/apm_planner #731: Pixhawk firmware update does not work on Ubuntu

sgofferj, 2015-06-09:
Generally, connecting to Pixhawk is only possible about 10-15 seconds after connecting USB on Ubuntu which also leads to FW updates failing. I suspect, the UDEV daemon grabs the device first and only after it cannot find a suitable driver, it releases the device.
Strictly this is probably not an AP2 bug, but there should either be a solution in the documentation or maybe a suitable blacklist file in the Linux package.
I will try to investigate what exactly causes the problem and post a solution as I find it.

sgofferj, 2015-06-09:
Here's a DMESG output from a USB connect:

[ 2416.321902] usb 3-1: new full-speed USB device number 11 using xhci_hcd
[ 2416.338836] usb 3-1: string descriptor 0 malformed (err = 2), defaulting to 0x0409
[ 2416.339194] usb 3-1: New USB device found, idVendor=26ac, idProduct=0011
[ 2416.339200] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2416.339204] usb 3-1: Product: PX4 BL FMU v2.x
[ 2416.339208] usb 3-1: Manufacturer: 3D Robotics
[ 2416.339211] usb 3-1: SerialNumber: 0
[ 2416.339442] usb 3-1: ep 0x83 - rounding interval to 1024 microframes, ep desc says 2040 microframes
[ 2416.556374] cdc_acm 3-1:1.0: This device cannot do calls on its own. It is not a modem.
[ 2416.556402] cdc_acm 3-1:1.0: ttyACM0: USB ACM device
[ 2416.556765] usbcore: registered new interface driver cdc_acm
[ 2416.556769] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[

billbonney, 2015-06-09:
Have you uninstalled the modemmanager? Flashing FW was working for me on Linux just the other day.

sgofferj, 2015-06-09:
Uninstalling modemmanager solves it BUT then my laptop doesn't recognize the 4G stick anymore...
I'll put this on my todo list for next weekend - finding a workaround - and will report back.

gmorph, 2015-06-09:
I have the same problem with ModemManager.  I tried to exclude the pixhawk devices from ModemManager using UDEV but couldn't get it to work.  I just mv /usr/sbin/ModemManager out of the way and then kill any running ModemManager tasks and do what I need to do and put ModemManager back and restart it when I need Wireless Broadband.  Annoying but ModemManager is a fairly stupid program and just grabs any serial device and starts shoving modem commands down it.
If you can figure out something better I'd love to know.
Thanks, Grant.

billbonney, 2015-06-10:
AP2 used to warn the user to remove the modem manager. Is that not happening anymore.  I'll try and remember to test. (My phone is my WiFi access point for 3G/4G connections so I don't need modem manager)

lostcontrol, 2015-06-10:
I think I got it working with ModemManager installed but I have to check again. I did not install a new firmware for some time and I don't remember right now if I have ModemManager installed or not. What I did is that I put a sleep(1 second) somewhere in AP2 serial code to let the things settle down before uploading the firmware. Never had a issue with firmware upload so. But again, I have to check first if I did not remove ModemManager.

I also had a lot of connection issues but these seem to be solved with 3.3 (and/or recent AP2 master).

I'm using Ubuntu 15.04.

sgofferj, 2015-06-10:
Here's a first take on a solution:

Create /etc/udev/rules.d/99-APM.rules with content
ATTRS{idVendor}=="26ac", ATTRS{idProduct}=="0011", ENV{ID_MM_DEVICE_IGNORE}="1"

Restart service udev

Works so far here. I will continue testing and report back. If it works, the rules file could be included in the Ubuntu package with the respective post-install triggers to restart the udev daemon.
I only have a genuine 3DR Pixhawk to test with. As far as other devices need the same, their VID and PID can be added to the same rules file, accordingly.

sgofferj, 2015-06-10:
Confirmed working here. Without the rules file:
Jun 10 10:34:37 enterprise kernel: [ 2121.112015] usb 3-1: new full-speed USB device number 21 using xhci_hcd
Jun 10 10:34:37 enterprise kernel: [ 2121.128972] usb 3-1: string descriptor 0 malformed (err = 2), defaulting to 0x0409
Jun 10 10:34:37 enterprise kernel: [ 2121.129378] usb 3-1: New USB device found, idVendor=26ac, idProduct=0011
Jun 10 10:34:37 enterprise kernel: [ 2121.129384] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jun 10 10:34:37 enterprise kernel: [ 2121.129389] usb 3-1: Product: PX4 BL FMU v2.x
Jun 10 10:34:37 enterprise kernel: [ 2121.129392] usb 3-1: Manufacturer: 3D Robotics
Jun 10 10:34:37 enterprise kernel: [ 2121.129396] usb 3-1: SerialNumber: 0
Jun 10 10:34:37 enterprise kernel: [ 2121.129607] usb 3-1: ep 0x83 - rounding interval to 1024 microframes, ep desc says 2040 microframes
Jun 10 10:34:37 enterprise kernel: [ 2121.130117] cdc_acm 3-1:1.0: This device cannot do calls on its own. It is not a modem.
Jun 10 10:34:37 enterprise kernel: [ 2121.130146] cdc_acm 3-1:1.0: ttyACM0: USB ACM device
Jun 10 10:34:37 enterprise mtp-probe: checking bus 3, device 21: "/sys/devices/pci0000:00/0000:00:14.0/usb3/3-1"
Jun 10 10:34:37 enterprise mtp-probe: bus: 3, device: 21 was not an MTP device
Jun 10 10:34:42 enterprise kernel: [ 2125.879326] usb 3-1: USB disconnect, device number 21
Jun 10 10:34:42 enterprise ModemManager[4911]: <info>  (tty/ttyACM0): released by modem /sys/devices/pci0000:00/0000:00:14.0/usb3/3-1
Jun 10 10:34:43 enterprise ModemManager[4911]: <warn>  (Plugin Manager) (Nokia) [ttyACM0] error when checking support: '(tty/ttyACM0) failed to open port: Could not open serial device ttyACM0: No such file or directory'
Jun 10 10:34:43 enterprise ModemManager[4911]: <warn>  (Plugin Manager) (Iridium) [ttyACM0] error when checking support: '(Iridium) Missing port probe for port (tty/ttyACM0)'
Jun 10 10:34:43 enterprise ModemManager[4911]: <warn>  (Plugin Manager) (Via CBP7) [ttyACM0] error when checking support: '(Via CBP7) Missing port probe for port (tty/ttyACM0)'
Jun 10 10:34:43 enterprise ModemManager[4911]: <warn>  (Plugin Manager) (Generic) [ttyACM0] error when checking support: '(Generic) Missing port probe for port (tty/ttyACM0)'
Jun 10 10:34:43 enterprise ModemManager[4911]: <warn>  Couldn't find support for device at '/sys/devices/pci0000:00/0000:00:14.0/usb3/3-1': not supported by any plugin
Jun 10 10:34:44 enterprise kernel: [ 2127.673856] usb 3-1: new full-speed USB device number 22 using xhci_hcd
Jun 10 10:34:44 enterprise kernel: [ 2127.877559] usb 3-1: New USB device found, idVendor=26ac, idProduct=0011
Jun 10 10:34:44 enterprise kernel: [ 2127.877568] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jun 10 10:34:44 enterprise kernel: [ 2127.877573] usb 3-1: Product: PX4 FMU v2.x
Jun 10 10:34:44 enterprise kernel: [ 2127.877577] usb 3-1: Manufacturer: 3D Robotics
Jun 10 10:34:44 enterprise kernel: [ 2127.877581] usb 3-1: SerialNumber: 0
Jun 10 10:34:44 enterprise kernel: [ 2127.877796] usb 3-1: ep 0x81 - rounding interval to 1024 microframes, ep desc says 2040 microframes
Jun 10 10:34:44 enterprise kernel: [ 2127.878591] cdc_acm 3-1:1.0: ttyACM0: USB ACM device
Jun 10 10:34:44 enterprise mtp-probe: checking bus 3, device 22: "/sys/devices/pci0000:00/0000:00:14.0/usb3/3-1"
Jun 10 10:34:44 enterprise mtp-probe: bus: 3, device: 22 was not an MTP device
Jun 10 10:35:07 enterprise ModemManager[4911]: <info>  Creating modem with plugin 'Generic' and '1' ports
Jun 10 10:35:07 enterprise ModemManager[4911]: <warn>  Could not grab port (tty/ttyACM0): 'Cannot add port 'tty/ttyACM0', unhandled serial type'
Jun 10 10:35:07 enterprise ModemManager[4911]: <warn>  Couldn't create modem for device at '/sys/devices/pci0000:00/0000:00:14.0/usb3/3-1': Failed to find primary AT port
Jun 10 10:35:18 enterprise kernel: [ 2162.134996] usb 3-1: USB disconnect, device number 22
Jun 10 10:35:18 enterprise ModemManager[4911]: <info>  (tty/ttyACM0): released by modem /sys/devices/pci0000:00/0000:00:14.0/usb3/3-1

With the rules file:
Jun 10 10:36:40 enterprise kernel: [ 2243.704407] usb 3-1: new full-speed USB device number 23 using xhci_hcd
Jun 10 10:36:40 enterprise kernel: [ 2243.725184] usb 3-1: string descriptor 0 malformed (err = 2), defaulting to 0x0409
Jun 10 10:36:40 enterprise kernel: [ 2243.725579] usb 3-1: New USB device found, idVendor=26ac, idProduct=0011
Jun 10 10:36:40 enterprise kernel: [ 2243.725587] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jun 10 10:36:40 enterprise kernel: [ 2243.725593] usb 3-1: Product: PX4 BL FMU v2.x
Jun 10 10:36:40 enterprise kernel: [ 2243.725598] usb 3-1: Manufacturer: 3D Robotics
Jun 10 10:36:40 enterprise kernel: [ 2243.725602] usb 3-1: SerialNumber: 0
Jun 10 10:36:40 enterprise kernel: [ 2243.725858] usb 3-1: ep 0x83 - rounding interval to 1024 microframes, ep desc says 2040 microframes
Jun 10 10:36:40 enterprise kernel: [ 2243.726387] cdc_acm 3-1:1.0: This device cannot do calls on its own. It is not a modem.
Jun 10 10:36:40 enterprise kernel: [ 2243.726422] cdc_acm 3-1:1.0: ttyACM0: USB ACM device
Jun 10 10:36:40 enterprise mtp-probe: checking bus 3, device 23: "/sys/devices/pci0000:00/0000:00:14.0/usb3/3-1"
Jun 10 10:36:40 enterprise mtp-probe: bus: 3, device: 23 was not an MTP device
Jun 10 10:36:40 enterprise kernel: [ 2244.397972] usb 3-1: USB disconnect, device number 23
[... cut ...]
Jun 10 10:36:48 enterprise kernel: [ 2251.706698] usb 3-1: Manufacturer: 3D Robotics
Jun 10 10:36:48 enterprise kernel: [ 2251.706699] usb 3-1: SerialNumber: 0
Jun 10 10:36:48 enterprise kernel: [ 2251.706775] usb 3-1: ep 0x81 - rounding interval to 1024 microframes, ep desc says 2040 microframes
Jun 10 10:36:48 enterprise kernel: [ 2251.707368] cdc_acm 3-1:1.0: ttyACM0: USB ACM device
Jun 10 10:36:53 enterprise mtp-probe: checking bus 3, device 25: "/sys/devices/pci0000:00/0000:00:14.0/usb3/3-1"
Jun 10 10:36:53 enterprise mtp-probe: bus: 3, device: 25 was not an MTP device

froller, 2015-06-10:
Same issue on Fedora

sgofferj, 2015-06-11:
@froller Could you try adding the rules file and restart the udev service as I described above in https://github.com/diydrones/apm_planner/issues/731#issuecomment-110632463 ? Does this fix the issue also on Fedora?

froller, 2015-06-26:
@sgofferj sorry for delay. I'm gonna test it this weekend

HarrisonHeli, 2015-09-05:
Thanks @sgofferj sgofferj 
I have just obtained a clone Pixhawk and was trying to update the firmware to 3.3-rc10.
I am using Mint 17 x64  with locally complied APM Planner v2.0.18.
At first the firmware would download but not flash to the device.

I followed your instructions and I had success.

> Create /etc/udev/rules.d/99-APM.rules with content
> ATTRS{idVendor}=="26ac", ATTRS{idProduct}=="0011", ENV{ID_MM_DEVICE_IGNORE}="1";

> sudo service udev restart

Here is my dmeg output after the udev rule change:

[188714.998238] systemd-udevd[4600]: starting version 204
[188768.451278] usb 5-1: USB disconnect, device number 14
[188772.507941] usb 5-1: new full-speed USB device number 15 using uhci_hcd
[188772.870059] usb 5-1: string descriptor 0 malformed (err = 2), defaulting to 0x0409
[188772.881081] usb 5-1: New USB device found, idVendor=26ac, idProduct=0011
[188772.881087] usb 5-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[188772.881091] usb 5-1: Product: PX4 BL FMU v2.x
[188772.881095] usb 5-1: Manufacturer: 3D Robotics
[188772.881098] usb 5-1: SerialNumber: 0
[188772.883087] cdc_acm 5-1:1.0: This device cannot do calls on its own. It is not a modem.
[188772.883110] cdc_acm 5-1:1.0: ttyACM0: USB ACM device
[188776.885355] usb 5-1: USB disconnect, device number 15

PR #1218 (2021-04-03): Firmware Install : Work around non-instant modem availability for PX4
During firmware updates, one is typically requested to replug the device.

On my system, the newly inserted device is nearly immediately detected by the planner, but the permissions on that device are not immediately set the right way (probably because `mtp-probe` needs to check it first). For this reason, firmware flashing is basically impossible.

This pull request introduces waiting for the device to become available with one-second sleeps between the attempts. While this is a dirty-ish hack, and something should have may [... cut]
closed 2021-04-03 by Arne-W