ArduPilot/apm_planner #732: Link changing announced, but fails.

AndKe, 2015-06-09:
Just build from master:
Upon connecting a telemetry radio:
```
INFO 2015-06-09T14:31:55.753 Inserting  "ttyUSB0" 
 INFO 2015-06-09T14:31:55.754 Changed Link to: "ttyUSB0" 
 INFO 2015-06-09T14:31:58.756 Inserting  "ttyUSB0" 
 INFO 2015-06-09T14:31:58.758 Changed Link to: "ttyUSB0" 
 INFO 2015-06-09T14:32:01.751 Inserting  "ttyUSB0" 
 INFO 2015-06-09T14:32:01.753 Changed Link to: "ttyUSB0" 
 INFO 2015-06-09T14:32:04.758 Inserting  "ttyUSB0" 
 INFO 2015-06-09T14:32:04.759 Changed Link to: "ttyUSB0" 
 INFO 2015-06-09T14:32:07.752 Inserting  "ttyUSB0" 
 INFO 2015-06-09T14:32:07.753 Changed Link to: "ttyUSB0" 
 INFO 2015-06-09T14:32:10.760 Inserting  "ttyUSB0" 
 INFO 2015-06-09T14:32:10.761 Changed Link to: "ttyUSB0" 
 INFO 2015-06-09T14:32:13.753 Inserting  "ttyUSB0" 
 INFO 2015-06-09T14:32:13.754 Changed Link to: "ttyUSB0" 
 INFO 2015-06-09T14:32:16.762 Inserting  "ttyUSB0" 
 INFO 2015-06-09T14:32:16.763 Changed Link to: "ttyUSB0" 
 INFO 2015-06-09T14:32:19.753 Inserting  "ttyUSB0" 
 INFO 2015-06-09T14:32:19.754 Changed Link to: "ttyUSB0" 
 INFO 2015-06-09T14:32:22.761 Inserting  "ttyUSB0" 
 INFO 2015-06-09T14:32:22.762 Changed Link to: "ttyUSB0" 
 INFO 2015-06-09T14:32:25.752 Inserting  "ttyUSB0" 

```

but the selected link does not change.
No similar response logged, and nothing changed when connecting a Pixhawk (/dev/ttyACM0)

billbonney, 2015-06-09:
Not a defect. It's just scanning for ports. It noisy I have an improvement coming

AndKe, 2015-06-09:
great - please auto-set port to ttyACMx or ttyUSBx  if seen on startup, or - if not connected , autoset it to an device that just spawned.   - default speed for ttyACM would be 115.2k , ttyUSB 57k.

billbonney, 2015-06-09:
That's the current behaviour, is to default to ttyACM. I'm going to close this as defect for now

closed 2015-06-09 by billbonney

AndKe, 2015-06-09:
- currently it does not auto-chanage port , or speed, to anything, based on plugging in, or seeing a device after start - (in case you think it currently works on Linux)

billbonney, 2015-06-09:
It should remember the setting for a serial port and the last used one. it does not switch automatically to a new serial link. UDP link will remain reading and waiting. You can have one or more serial links

AndKe, 2015-06-09:
It does not, connecting Pixhawk I need to select 115k, and connecting telemetry 57k, then same all over again next time

billbonney, 2015-06-09:
I doing some rework of the port handling and I am developing jointly on Unix/OSX these days. I'll make sure that gets fixed.