ArduPilot/apm_planner #734: Extended tuning screen throttle PIDs not refreshed

sgofferj, 2015-06-13:
I yesterday had a crash with my IRIS. After detailed analysis of the logfile and params I found that all throttle PIDs were 0. Before the flight, I set the CH7 option through AP2's extended tuning screen.
I did some testing today and found that when clicking refresh, the PIDs in the extended tuning screen were not updated from the vehicle reliably, leading to the throttle PIDs being 0 when I wrote the data back to my IRIS. There should be some kind of safeguard e.g. prevent editing of the values (grey out) until all PIDs have been reliably loaded from the vehicle and there should also be an indicator indicating if all values have been reliably written to the vehicle. Maybe even a verify after write?

HarrisonHeli, 2015-10-22:
Yip, this one got me a few days ago,. No crash but I nearly had a fly-away. Got it back under acro mode from 150m as it was trying to disappear into the clouds!
The quad appeared to ignore the RTL command and GEO-fence.

The problem is in the UAS.cc
See my branch commit for the solution.
https://github.com/HarrisonHeli/apm_planner/commit/b7e2a6853b220a99784d5a7e24d30c7e6a78d9ce

In addition the order the that the parameters are requested can have have an affect on if all the parameters arrive.
I have been playing with different combinations but have not found a foolproof combination yet.
https://github.com/HarrisonHeli/apm_planner/commit/4d1e087cebc1a08a2c63780d25bf6c5bd6ace3ea

billbonney, 2015-10-22:
I fixed this here https://github.com/diydrones/apm_planner/commit/d7c904ad1fe1c1e5d717ceeedbd647053f560706 and it part of 2.0.19.rc4

HarrisonHeli, 2015-10-22:
Cool. 
It appears that I unfixed your fix and came up with my own fix
I cherry picked the files from https://github.com/diydrones/apm_planner/commit/d7c904ad1fe1c1e5d717ceeedbd647053f560706 and your solution works well.

I will be using mission planner to set-up a new quad this weekend so it will be an opportunity to test this.

billbonney, 2015-10-23:
It was bad merge. I should have took a little more time on it before merging. I suppose that's what RC candidates are for ;-)