ArduPilot/apm_planner #749: Connecting joystick with joystick dialog open causes crash

ronpandolfi, 2015-06-20:
Tested the joystick dialog on OS X 10.10.2 with XBOX 360 wired controller. When I have the joystick dialog open and the controller connected (either open dialog first or connect controller first) it crashes.

Crash log: http://pastebin.com/wYm2P6nG

billbonney, 2015-06-22:
What version of APM Planner 2.0 are you using, try using rc2, thanks

ronpandolfi, 2015-06-22:
I tested with rc2 built from source and the rc1 release dmg. If you have a built mac version you'd like me to test I can also do that.

billbonney, 2015-06-22:
Everytime we push to master brach a build will appear here http://firmware.diydrones.com/Tools/APMPlanner/daily/

beta (aka release candidate builds here) (based on the beta branch, tags are used to name the build)
http://firmware.diydrones.com/Tools/APMPlanner/beta/

stable builds are here (based on the stable branch, tags are used the name the build.)
http://firmware.diydrones.com/Tools/APMPlanner/

ronpandolfi, 2015-06-22:
Ah, great. I didn't see the rc2 dmg in the github releases. I'll try the rc2 build dmg tonight and let you know. Thanks!

ronpandolfi, 2015-06-23:
The rc2 dmg had the same behavior. Crash log below:
http://pastebin.com/FqysK39j

kenckh, 2015-09-15:
Hi, I'm facing the same issue as ronpandolfi. I have tried v 2.0.18 and 2.0.19 from beta. Pls advice.

imuguruza, 2015-12-12:
Me too, using 2.0.19-rc4 and 2.0.18

KalebClark, 2016-01-01:
I am having the same problem on OSX with latest from: /Tools/APMPlanner/daily/2015-12-31

Everything runs fine until the joystick dialog detects a joystick. Then immediate crash

billbonney, 2016-01-01:
Interesting. I tested this just a couple days ago in relation to #862 . It was working for me. The code isn't super robust, as it's not fully functional (experimental), but it should be stable enough to test.
what OSX version are you using?

imuguruza, 2016-01-07:
Yosemite 10.10.5

Drunkar, 2016-03-27:
I faced the same problem on mac (Yosemite 10.10.5) and windows.
And on APM planner 2.0.17, I can use joystick with no problems.
This issue was reported on ardupilot.com:
http://ardupilot.com/forum/viewtopic.php?f=82&t=13872