ArduPilot/apm_planner #751: Planner crashing while Armed

roldershaw, 2015-06-25:
Hi,

I'm running APM Planner v2.0.18-rc2 built from the latest commit on Xubuntu 14.04 and I'm getting crashes generally whilst in flight but also when I'm on the ground testing. I'm connecting using a third-party telemetry radio connected to an APM running the final firmware.

I've captured the terminal output from the program here http://pastebin.com/x6PrfWB0

Thanks.

billbonney, 2015-06-25:
What version of Qt are you using?

roldershaw, 2015-06-25:
Qt version 5.2.1

mrpilot2, 2015-06-25:
Seems to be a duplicate of #648, which states a a bug in libxcb on Ubuntu 14.04

roldershaw, 2015-06-25:
I'll try the workaround suggested in that issue and report back, thanks.

roldershaw, 2015-06-27:
Flew for around an hour today with no crashes, thanks @mrpilot2 

https://gist.github.com/slimsag/a26d838ccc4480ce21bc resolved the issue.

closed 2015-06-27 by roldershaw

billbonney, 2015-06-28:
Thanks for the update. I wonder if we will get an official fix for LTS

roldershaw, 2015-06-28:
Maybe at some point, there is an actual bug report for it here https://bugs.launchpad.net/ubuntu/+source/libxcb/+bug/1402966
So it should be actioned upon at some point, for now it may be an idea to write about this issue in the readme. The current release I'm on is Ubuntu 14.04.2 LTS