ArduPilot/apm_planner #754: Telemetry radio rssi/remrssi not available?

jasonmc44, 2015-07-01:
I am unable to find any telemetry radio messages (specifically rssi/remrssi) when graphing live data or anywhere else for that matter. Only place I have seen them available is when looking at a TLOG. Either i'm looking in the wrong place or it's not available when connected to a live vehicle. Running Copter 3.2.

Thanks

Jason

billbonney, 2015-07-01:
You need to enable the Heartbeat sending from APM Planner 2.0 for the messages to appear (it's in APM Planner 2.0 Config View). And it will only be in tlogs. It's on by default

jasonmc44, 2015-07-01:
I've got heartbeat on. Are you saying telemetry rssi is only in tlogs and
not live.

billbonney, 2015-07-01:
The MAVLink radio message is only in MAVlink message logs or in the graph view showing Tlogs or live data. Not data flash logs. 

I'll test tomorrow and check its working.

HarrisonHeli, 2015-09-05:
Hi @billbonney 
I am trying to get a live view of the MAVLINK_MSG_ID_RADIO messages in MP v2.0.18.
I can find it in the tlogs but MP wont display it in the live graph.

Why can it be recorded in the tlog file but not displayed in live view? I am trying to understand how the MAVlink messages are processed in MP by studying the code but I am a bit lost at the moment.

AndKe, 2015-09-12:
Mavlink Inspector does not show RADIO.* items
-Info view "quick" and "status" does not offer to show it.

There is basically no place to see current RADIO.rssi , remoterssi , noise, and remotenoise   Is that a fix that needs to be implemented many places, or is there simply one place that needs to be updated to "know" those items ?

billbonney, 2015-09-12:
this fix https://github.com/diydrones/apm_planner/commit/b0fa90c67ab94aeb837b0d936bbadff94687b3ed#diff-ff6f0e8f96c8b65ee73924f60debbb8e should allow it to be shown in the Info View. I was unaware that this was still a problem in the Inspector Widget

billbonney, 2015-09-13:
I had a look at this in some detail finally. The problem seems that the Radio send messages from a different component id system id and this is getting filtered out. I can capture the data on the stream. I just need to see why its not showing in MAV Link Inspector. I'm sure it used too work. I tested with older 1.6 SiK FW as I wasn't sure if the 1.9V was the issue.
```
Printing description of mavMessage:
 MavMsgRadio:
 rxerrors = 264
 fixed = 0
 rssi = 211
 remrssi = 209
 txbuf = 100
 noise = 39
 remnoise = 31
Printing description of msg:
(mavlink_message_t) msg = {
  checksum = 40076
  magic = '\xfe'
  len = '\t'
  seq = 'O'
  sysid = 51
  compid = 68
  msgid = '\xa6'
  payload64 = ([0] = 2838624369318560008, [1] = 13773809099665607711, [2] = 109964047679601, [3] = 123109826393889, [4] = 74, [5] = 18446505545498624000, [6] = 2634042836353286144, [7] = 85027964693155838, [8] = 134409035265, [9] = 0, [10] = 0, [11] = 0, [12] = 0)
}
```

HarrisonHeli, 2015-09-13:
Hi @billbonney 
I have figured out where the sysid filter is being applied by MP.

So far I have traced the code and seen that MAVLinkProtocol::handleMessage() is called by MAVLinkProtocol::receiveBytes() when a valid message is received.

Within the handelMessage() there is the line:
UASInterface* uas = m_connectionManager->getUas(message.sysid);

If the message.sysid does not match 1, the uas is returned as 0x00 and the message is discarded.
To prove it, I changed all incoming message.sysid's to always be 1 and now I can see the live radio messages in the log and status.
What I have done is nasty and not really a solution. I am hoping you will know the correct way to fix this.

billbonney, 2015-09-14:
I've had look and it's and interesting problem to solve 'cleanly'. When system connects it sends out a HB Message. This means a UAS object is created for that system. Since the Radio is sending on a Radio Message and not identifying HB message, a UAS Object is not created and the message doesn't make it to the higher levels. I'm trying to figure out a reasonable fix that would not create a UAS object but allow the the message to be seen. It's almost like it should re routed to each UAS object as though it was intended for them.

AndKe, 2015-09-14:
What about treating radio message like a heartbeat, and create UAS object
by it as well ?

HarrisonHeli, 2015-09-14:
Interesting that sysid = 51, compid = 68 is 0x33 and 0x44 in hex.
Is there some significance to the radio module using these numbers. Are they going to remain using them in future firmware updates or is this a bug that has made its way into the radio firmware?