ArduPilot/apm_planner #767: Windows compilation/running issues

mpu-creare, 2015-07-15:
Greetings! I'm hoping to add some functionality to APM Planner for my own projects. 

I've been trying to compile on windows, and I suspect the instructions in the readme are out of date. I'm running into a number of issues. 

Following the Instructions I get:
```
src\ui\CameraView.cc(240) : error C3861: 'glDrawPixels': identifier not found
src\ui\CameraView.cc(246) : error C2065: 'GL_PROJECTION' : undeclared identifier
src\ui\CameraView.cc(246) : error C3861: 'glMatrixMode': identifier not found
src\ui\CameraView.cc(247) : error C3861: 'glLoadIdentity': identifier not found
src\ui\CameraView.cc(248) : error C3861: 'glOrtho': identifier not found
src\ui\CameraView.cc(249) : error C2065: 'GL_MODELVIEW' : undeclared identifier
src\ui\CameraView.cc(249) : error C3861: 'glMatrixMode': identifier not found
```

Then I edit qgroundcontrol.pro removing 
```
#ifdef CAMERAVIEW
    src/ui/CameraView.h \
#endif
#ifdef CAMERAVIEW
    src/ui/CameraView.cc \
#endif
```
I clean and remove all temporary files (basically a clean clone with those lines removed. 
Then i get:
```
src\uas\LogDownloadDialog.cc(374) : error C2668: 'floor' : ambiguous call to overloaded function
        C:\Program Files (x86)\Microsoft Visual Studio 10.0\VC\INCLUDE\math.h(567): could be 'long double floor(long double)'
        C:\Program Files (x86)\Microsoft Visual Studio 10.0\VC\INCLUDE\math.h(519): or       'float floor(float)'
        C:\Program Files (x86)\Microsoft Visual Studio 10.0\VC\INCLUDE\math.h(136): or       'double floor(double)'
        while trying to match the argument list '(uint32_t)'
src\uas\LogDownloadDialog.cc(380) : error C2668: 'floor' : ambiguous call to overloaded function
        C:\Program Files (x86)\Microsoft Visual Studio 10.0\VC\INCLUDE\math.h(567): could be 'long double floor(long double)'
        C:\Program Files (x86)\Microsoft Visual Studio 10.0\VC\INCLUDE\math.h(519): or       'float floor(float)'
        C:\Program Files (x86)\Microsoft Visual Studio 10.0\VC\INCLUDE\math.h(136): or       'double floor(double)'
        while trying to match the argument list '(uint32_t)'
```

Looking at the code it looks like we're just diving two ints, which means the floor is just safe ,and not really needed. So i just take out the floor in those lines of code. 

Okay, then we get a lot further, but it now gives problems with SDL. For example:
```
JoystickInput.obj : error LNK2019: unresolved external symbol _SDL_JoystickGetButton referenced in function "protected: bool __thiscall JoystickInput::sdlJoystickUpdate(unsigned int)" (?sdlJoystickUpdate@JoystickInput@@IAE_NI@Z)
JoystickInput.obj : error LNK2019: unresolved external symbol _SDL_JoystickGetHat referenced in function "protected: bool __thiscall JoystickInput::sdlJoystickUpdate(unsigned int)" (?sdlJoystickUpdate@JoystickInput@@IAE_NI@Z)
JoystickInput.obj : error LNK2019: unresolved external symbol _SDL_JoystickGetAxis referenced in function "protected: bool __thiscall JoystickInput::sdlJoystickUpdate(unsigned int)" (?sdlJoystickUpdate@JoystickInput@@IAE_NI@Z)
JoystickInput.obj : error LNK2019: unresolved external symbol _SDL_GetTicks referenced in function "protected: bool __thiscall JoystickInput::sdlJoystickUpdate(unsigned int)" (?sdlJoystickUpdate@JoystickInput@@IAE_NI@Z)
JoystickInput.obj : error LNK2019: unresolved external symbol _SDL_JoystickUpdate referenced in function "protected: bool __thiscall JoystickInput::sdlJoystickUpdate(unsigned int)" (?sdlJoystickUpdate@JoystickInput@@IAE_NI@Z)
```

Then I downloaded SDL2 from http://www.libsdl.org/release/SDL2-devel-2.0.3-VC.zip
I unzip it then I edit qgroundcontrol.pro again adding the following lines (from https://www.youtube.com/watch?v=3KpD_to28YQ):
```
INCLUDEPATH += -L C:/SDL2-2.0.3/include
LIBS += -L C:/SDL2-2.0.3/lib/x86 -lSDL2 -lSDL2main
```

But then I get a: LNK1146: no argument specified with option '/LIBPATH:'

We we change the above lines to 
```
INCLUDEPATH += C:/SDL2-2.0.3/include
LIBS += -lSDL2 -lSDL2main
LIBPATH += C:/SDL2-2.0.3/lib/x86
```

And hurrah, we can compile into an .exe file. 

Next there's an error, saying the file does not exist for this line: 
```
copy /y C:\apm_planner\libs\lib\sdl\win32\SDL.dll C:\apm_planner\release
```

So, then I create a new file SDL.dll in the libs\lib\sdl\win32 directory. Now the build completes happily. Okay, we get past this step. 

When I try to run apmplanner2.exe it complains that SDL2.dll is missing. Ok, so then I copy SDL2.dll from libs\lib\sdl\win32 into the same directory as the .exe. 

NOW when trying to run it complains that libGLESv2.dll is missing. 

According to [stackoverflow](http://stackoverflow.com/questions/12823551/libglesv2-dll-cannot-be-found-while-compiling-code-that-uses-opengl)  this suggests that I'm linking against the wrong libgl? 

Okay, so then I change my build kit to something different than was listed in the README.md file. Now I'm using the MSVC2010 32bit OpenGL kit that was automatically detected. 

Now when I start apmplanner2.exe I get a "qml/PrimaryFlightDisplayQML.qml not found". So, I create the qml directory in the same place that the .exe is, and I copy FligthDisplayQML.qml into the 'qml' subdirectory, along with "ApmToolBar.qml" and the "components" directory

Now everything seems to work. 

I'm also using the 2.0.18-rc2 release (tag) to do the compilation, so it's not at the head of the repository. 

I can continue hacking this, but I'm new to QT, and I haven't programmed in c/c++ for quite some time. Seems like this could be more fixable? It seems like these are mostly path issues. I probably didn't have to download SDL and link by myself if the paths were set correctly. Anyone think they can fix this one for us Windows folks? 

Thanks!

dcarpy, 2015-08-03:
OK, I've figured out what's going on and will submit a pull request to fix.

billbonney, 2015-08-04:
the windows builds are cross compiled on linux using
`/usr/local/Qt-5.2.1/bin/qmake -r -spec win32-x-g++ qgroundcontrol.pro`
There is nothing special required in the build AFAIU

dcarpy, 2015-08-04:
I agree.  However, I assumed he was talking about the native windows build.

billbonney, 2015-08-04:
I would recommend following the instructions on QT site for building with MSVC http://doc.qt.io/qt-5/windows-requirements.html The will negate the Open GL errors that are being seen, and the integer conversion problems,

mpu-creare, 2015-08-04:
Thanks dcarpy, and yes, this was for a native windows build.

commit 3aa4f3d9 (2015-11-24): build: Fixes #767 Replace SDL2 x64 dll with x86 dll

Replaced SDL2 x64 dll and x64 libs for native Windows build
with x86 dll and libs.
commit 50b2f53f (2015-11-24): build: Fixes #767 Use SDL version 2.0.3 instead for native Windows builds

Replaced binary name from SDL to SDL2 in copy command.
closed 2015-11-24 by billbonney