ArduPilot/apm_planner #768: Can't reconnect over USB

nunojpg, 2015-07-16:
I can always connect fine to my board over USB, but If I press disconnect I can' connect again. I always have to reboot APM Planner.

Also, regardless if connected or disconnected I always keep receiving on the console every 3 seconds this messages:

 INFO 2015-07-16T14:37:48.685 Changed Link to: "ttyUSB0" 
 INFO 2015-07-16T14:37:51.710 Inserting  "ttyUSB0" 
 INFO 2015-07-16T14:37:51.711 Changed Link to: "ttyUSB0" 
 INFO 2015-07-16T14:37:54.684 Inserting  "ttyUSB0" 
 INFO 2015-07-16T14:37:54.684 Changed Link to: "ttyUSB0" 
 INFO 2015-07-16T14:37:57.685 Inserting  "ttyUSB0" 
 INFO 2015-07-16T14:37:57.685 Changed Link to: "ttyUSB0" 
 INFO 2015-07-16T14:38:00.684 Inserting  "ttyUSB0" 
 INFO 2015-07-16T14:38:00.684 Changed Link to: "ttyUSB0" 

Ubuntu 15.04
Tested master and 2.0.18-rc2.
Board is a Crius AIOP V2.

billbonney, 2015-07-16:
The messages are benign and not a problem. 

What USB type is the Cirus board. As it sounds like a driver issue.  I don't have a problem with PH, APM, or 3DR Radio.

nunojpg, 2015-07-16:
FT232RL

closed 2019-11-05 by nunojpg