ArduPilot/apm_planner #769: Can't load parameters from file

nunojpg, 2015-07-16:
The following params file was created with Save To File, but Load From File causes repeatedly error "Failed to convert number, please verify your input and try again."

Maybe it is considering localization number format in one and not in the other?

#NOTE: 7/16/2015 2:48:38 DA TARDE: second
ACRO_BAL_PITCH,1.00000000
ACRO_BAL_ROLL,1.00000000
ACRO_RP_P,4.50000000
ACRO_TRAINER,2
ACRO_YAW_P,4.50000000
AHRS_COMP_BETA,0.10000000
AHRS_GPS_DELAY,2
AHRS_GPS_GAIN,1.00000000
AHRS_GPS_MINSATS,6
AHRS_GPS_USE,1
AHRS_ORIENTATION,0
AHRS_RP_P,0.10000000
AHRS_TRIM_X,0.00000000
AHRS_TRIM_Y,0.00000000
AHRS_TRIM_Z,0.00000000
AHRS_WIND_MAX,0
AHRS_YAW_P,0.10000000
ANGLE_MAX,4500
ANGLE_RATE_MAX,18000
ARMING_CHECK,1
BATT_AMP_OFFSET,0.00000000
BATT_AMP_PERVOLT,17.00000000
BATT_CAPACITY,3300
BATT_CURR_PIN,-1
BATT_MONITOR,0
BATT_VOLT_MULT,10.10000038
BATT_VOLT_PIN,-1
CAM_DURATION,10
CAM_SERVO_OFF,1100
CAM_SERVO_ON,1300
CAM_TRIGG_DIST,0.00000000
CAM_TRIGG_TYPE,0
CH7_OPT,0
CH8_OPT,0
CIRCLE_RADIUS,10
CIRCLE_RATE,20.00000000
COMPASS_AUTODEC,1
COMPASS_DEC,0.00000000
COMPASS_EXTERNAL,0
COMPASS_LEARN,0
COMPASS_MOTCT,0
COMPASS_MOT_X,0.00000000
COMPASS_MOT_Y,0.00000000
COMPASS_MOT_Z,0.00000000
COMPASS_OFS_X,0.00000000
COMPASS_OFS_Y,0.00000000
COMPASS_OFS_Z,0.00000000
COMPASS_ORIENT,0
COMPASS_USE,1
ESC,0
FENCE_ACTION,1
FENCE_ALT_MAX,100.00000000
FENCE_ENABLE,0
FENCE_MARGIN,2.00000000
FENCE_RADIUS,150.00000000
FENCE_TYPE,3
FLOW_ENABLE,0
FLTMODE1,0
FLTMODE2,0
FLTMODE3,0
FLTMODE4,0
FLTMODE5,0
FLTMODE6,0
FRAME,1
FS_BATT_ENABLE,0
FS_BATT_MAH,0.00000000
FS_BATT_VOLTAGE,10.50000000
FS_GCS_ENABLE,0
FS_GPS_ENABLE,1
FS_THR_ENABLE,0
FS_THR_VALUE,975
GND_ABS_PRESS,100370.54687500
GND_ALT_OFFSET,0
GND_TEMP,34.29544830
GPSGLITCH_ACCEL,1000.00000000
GPSGLITCH_ENABLE,1
GPSGLITCH_RADIUS,200.00000000
GPS_HDOP_GOOD,200
HLD_LAT_I,0.00000000
HLD_LAT_IMAX,0
HLD_LAT_P,1.00000000
HLD_LON_I,0.00000000
HLD_LON_IMAX,0
HLD_LON_P,1.00000000
INAV_TC_XY,2.50000000
INAV_TC_Z,5.00000000
INS_ACCOFFS_X,0.00000000
INS_ACCOFFS_Y,0.00000000
INS_ACCOFFS_Z,0.00000000
INS_ACCSCAL_X,1.00000000
INS_ACCSCAL_Y,1.00000000
INS_ACCSCAL_Z,1.00000000
INS_GYROFFS_X,-0.04626963
INS_GYROFFS_Y,0.00353055
INS_GYROFFS_Z,0.03396460
INS_MPU6K_FILTER,0
INS_PRODUCT_ID,40
LAND_SPEED,50
LED_MODE,9
LOG_BITMASK,830
LOITER_LAT_D,0.00000000
LOITER_LAT_I,0.50000000
LOITER_LAT_IMAX,400
LOITER_LAT_P,1.00000000
LOITER_LON_D,0.00000000
LOITER_LON_I,0.50000000
LOITER_LON_IMAX,400
LOITER_LON_P,1.00000000
MAG_ENABLE,1
MNT_ANGMAX_PAN,4500
MNT_ANGMAX_ROL,4500
MNT_ANGMAX_TIL,4500
MNT_ANGMIN_PAN,-4500
MNT_ANGMIN_ROL,-4500
MNT_ANGMIN_TIL,-4500
MNT_CONTROL_X,0.00000000
MNT_CONTROL_Y,0.00000000
MNT_CONTROL_Z,0.00000000
MNT_JSTICK_SPD,0
MNT_MODE,0
MNT_NEUTRAL_X,0.00000000
MNT_NEUTRAL_Y,0.00000000
MNT_NEUTRAL_Z,0.00000000
MNT_RC_IN_PAN,0
MNT_RC_IN_ROLL,0
MNT_RC_IN_TILT,0
MNT_RETRACT_X,0.00000000
MNT_RETRACT_Y,0.00000000
MNT_RETRACT_Z,0.00000000
MNT_STAB_PAN,0
MNT_STAB_ROLL,0
MNT_STAB_TILT,0
MOT_SPIN_ARMED,70
MOT_TCRV_ENABLE,1
MOT_TCRV_MAXPCT,93
MOT_TCRV_MIDPCT,52
OF_PIT_D,0.12000000
OF_PIT_I,0.50000000
OF_PIT_IMAX,100
OF_PIT_P,2.50000000
OF_RLL_D,0.12000000
OF_RLL_I,0.50000000
OF_RLL_IMAX,100
OF_RLL_P,2.50000000
PILOT_VELZ_MAX,250
RATE_PIT_D,0.00400000
RATE_PIT_I,0.10000000
RATE_PIT_IMAX,500
RATE_PIT_P,0.15000001
RATE_RLL_D,0.00400000
RATE_RLL_I,0.10000000
RATE_RLL_IMAX,500
RATE_RLL_P,0.15000001
RATE_YAW_D,0.00000000
RATE_YAW_I,0.02000000
RATE_YAW_IMAX,800
RATE_YAW_P,0.20000000
RC10_DZ,0
RC10_FUNCTION,0
RC10_MAX,1900
RC10_MIN,1100
RC10_REV,1
RC10_TRIM,1500
RC11_DZ,0
RC11_FUNCTION,0
RC11_MAX,1900
RC11_MIN,1100
RC11_REV,1
RC11_TRIM,1500
RC1_DZ,30
RC1_MAX,1900
RC1_MIN,1100
RC1_REV,1
RC1_TRIM,1500
RC2_DZ,30
RC2_MAX,1900
RC2_MIN,1100
RC2_REV,1
RC2_TRIM,1500
RC3_DZ,30
RC3_MAX,1900
RC3_MIN,1100
RC3_REV,1
RC3_TRIM,1500
RC4_DZ,40
RC4_MAX,1900
RC4_MIN,1100
RC4_REV,1
RC4_TRIM,1500
RC5_DZ,0
RC5_FUNCTION,0
RC5_MAX,1900
RC5_MIN,1100
RC5_REV,1
RC5_TRIM,1500
RC6_DZ,0
RC6_FUNCTION,0
RC6_MAX,1900
RC6_MIN,1100
RC6_REV,1
RC6_TRIM,1500
RC7_DZ,0
RC7_FUNCTION,0
RC7_MAX,1900
RC7_MIN,1100
RC7_REV,1
RC7_TRIM,1500
RC8_DZ,0
RC8_FUNCTION,0
RC8_MAX,1900
RC8_MIN,1100
RC8_REV,1
RC8_TRIM,1500
RCMAP_PITCH,2
RCMAP_ROLL,1
RCMAP_THROTTLE,3
RCMAP_YAW,4
RC_SPEED,490
RELAY_PIN,46
RSSI_PIN,-1
RTL_ALT,1500
RTL_ALT_FINAL,0
RTL_LOIT_TIME,5000
SCHED_DEBUG,0
SERIAL1_BAUD,57
SIMPLE,0
SONAR_ENABLE,0
SONAR_GAIN,0.80000001
SONAR_TYPE,0
SR0_EXTRA1,10
SR0_EXTRA2,10
SR0_EXTRA3,2
SR0_EXT_STAT,2
SR0_PARAMS,10
SR0_POSITION,3
SR0_RAW_CTRL,0
SR0_RAW_SENS,2
SR0_RC_CHAN,2
SR1_EXTRA1,0
SR1_EXTRA2,0
SR1_EXTRA3,0
SR1_EXT_STAT,0
SR1_PARAMS,0
SR1_POSITION,0
SR1_RAW_CTRL,0
SR1_RAW_SENS,0
SR1_RC_CHAN,0
STB_PIT_I,0.00000000
STB_PIT_IMAX,0
STB_PIT_P,4.50000000
STB_RLL_I,0.00000000
STB_RLL_IMAX,0
STB_RLL_P,4.50000000
STB_YAW_I,0.00000000
STB_YAW_IMAX,0
STB_YAW_P,4.50000000
SUPER_SIMPLE,0
SYSID_MYGCS,255
SYSID_SW_MREV,120
SYSID_SW_TYPE,10
SYSID_THISMAV,1
TELEM_DELAY,0
THR_ACCEL_D,0.00000000
THR_ACCEL_I,1.50000000
THR_ACCEL_IMAX,500
THR_ACCEL_P,0.75000000
THR_ALT_I,0.00000000
THR_ALT_IMAX,300
THR_ALT_P,1.00000000
THR_MAX,1000
THR_MID,500
THR_MIN,130
THR_RATE_D,0.00000000
THR_RATE_I,0.00000000
THR_RATE_IMAX,300
THR_RATE_P,6.00000000
TRIM_THROTTLE,450
TUNE,0
TUNE_HIGH,1000
TUNE_LOW,0
WPNAV_ACCEL,100.00000000
WPNAV_LOIT_SPEED,500.00000000
WPNAV_RADIUS,200.00000000
WPNAV_SPEED,500.00000000
WPNAV_SPEED_DN,150.00000000
WPNAV_SPEED_UP,250.00000000
WP_INDEX,0
WP_TOTAL,0
WP_YAW_BEHAVIOR,2

billbonney, 2015-07-16:
What locale is your system set as? and are you using , as the decimal delimiter?

nunojpg, 2015-07-16:
LANG=en_US.UTF-8
LANGUAGE=en_US
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC=pt_PT.UTF-8
LC_TIME=pt_PT.UTF-8
LC_COLLATE="en_US.UTF-8"
LC_MONETARY=pt_PT.UTF-8
LC_MESSAGES="en_US.UTF-8"
LC_PAPER=pt_PT.UTF-8
LC_NAME=pt_PT.UTF-8
LC_ADDRESS=pt_PT.UTF-8
LC_TELEPHONE=pt_PT.UTF-8
LC_MEASUREMENT=pt_PT.UTF-8
LC_IDENTIFICATION=pt_PT.UTF-8
LC_ALL=

This are all the Ubuntu default installs, when I select my country or timezone, I guess. My decimal delimiter is probably the ",".

billbonney, 2015-07-16:
A fix was put in to stop a number such as 5,00 being interpreted as a zero. I'm going to remove it and change it make sure numbers are entered as x.xx only and if not they are not changed.

More work is really needed to support 0,00 format throughout amp_planner, for now we should just stick to .

nunojpg, 2015-07-16:
I don't get it. Is this a param file write or read problem? The file looks using the "." decimal separator. Isn't this a question of passing the "US locale" to "atof()" or friends so they will interpret the file correctly?

billbonney, 2015-07-16:
The problem is with the AdvParameterList.cc Line#99 QLocale calls.

We don't use atof() in Qt. We use QVariant types.

nunojpg, 2015-07-16:
Ok. What I mean is that I don't think there is a need to support "0,00 format". I think is best that Param files are universal, so they should always use the same locale regardless of the PC where they happen to be created.

Thanks for the support.

billbonney, 2015-07-16:
I've pushed this fix to master https://github.com/diydrones/apm_planner/commit/b996db6b9a2ae576dfd63c7bbf1bab5ee34a4ea6
It means that it only accepts 'C' language locale for numbers with is XXXX.XXXX no thousands separator.
If we add locale support for X.XXX,XX it needs to be more comprehensive.

This change maintains the original intent that adding a badly formatted number will not allow the value to be set to zero, but its previous value

You can test the latest daily build form http://firmware.diydrones.com/Tools/APMPlanner/daily/2015-07-16/ tr http://firmware.diydrones.com/Tools/APMPlanner/daily/2015-07-17 with SHA b996db6 in the name

closed 2015-07-16 by billbonney