ArduPilot/apm_planner #772: APM:Copter 3.3 Extended Tuning screen value refresh does not work properly

walmis, 2015-07-26:
When I hit refresh, values do not update properly. Usually fields display zeroes. The only way I can get the fields to update is to go to Full Parameter List, hit refresh there and quickly go back to extended tuning, as the values are downloading.

walmis, 2015-07-26:
This seems like a flow control issue, link gets flooded with PARAM_REQUEST requests and most of those requests get lost. I have tried proxying over UDP with MAVProxy, works better - about 80% params get returned, but when connecting over USB about 10% params get returned. Also tried connecting over UART to APM:Copter, situation is almost the same as over USB.

I'm working on a patch to queue parameter requests, and issue new requests only when requested parameters are returned.

walmis, 2015-07-26:
Patch #1 
UAS: Queue requested parameters

```
diff --git a/src/uas/UAS.cc b/src/uas/UAS.cc
index 4d1830f..b578f4d 100644
--- a/src/uas/UAS.cc
+++ b/src/uas/UAS.cc
@@ -2638,6 +2638,10 @@ void UAS::processParamValueMsg(mavlink_message_t& msg, const QString& paramName,
     default:
         QLOG_ERROR() << "INVALID DATA TYPE USED AS PARAMETER VALUE: " << rawValue.param_type;
     } //switch (value.param_type)
+
+    if(!paramRequestQueue.isEmpty()) {
+        requestNextParamFromQueue();
+    }
 }
 
 /**
@@ -2657,11 +2661,13 @@ void UAS::requestParameter(int component, int id)
     //QLOG_DEBUG() << __FILE__ << __LINE__ << "REQUESTING PARAM RETRANSMISSION FROM COMPONENT" << component << "FOR PARAM ID" << id;
 }
 
-/**
-* Request a parameter, use parameter name to request it.
-*/
-void UAS::requestParameter(int component, const QString& parameter)
-{
+void UAS::requestNextParamFromQueue() {
+    if(paramRequestQueue.isEmpty()) return;
+
+    QPair<int,QString> pr = paramRequestQueue.front();
+    int component = pr.first;
+    QString parameter = pr.second;
+
     // Request parameter, use parameter name to request it
     mavlink_message_t msg;
     mavlink_param_request_read_t read;
@@ -2681,6 +2687,24 @@ void UAS::requestParameter(int component, const QString& parameter)
     mavlink_msg_param_request_read_encode(systemId, componentId, &msg, &read);
     sendMessage(msg);
     QLOG_DEBUG() << __FILE__ << __LINE__ << "REQUESTING PARAM RETRANSMISSION FROM COMPONENT" << component << "FOR PARAM NAME" << parameter;
+
+    paramRequestQueue.pop_front();
+}
+
+/**
+* Request a parameter, use parameter name to request it.
+*/
+void UAS::requestParameter(int component, const QString& parameter)
+{
+    QLOG_DEBUG() << "Queuing param " << parameter;
+    QPair<int,QString> p = QPair<int,QString>(component, parameter);
+
+    paramRequestQueue.append(p);
+
+    if(paramRequestQueue.size() == 1) {
+       //requestNextParamFromQueue();
+       QTimer::singleShot(0, this, SLOT(requestNextParamFromQueue()));
+    }
 }
 
 /**
diff --git a/src/uas/UAS.h b/src/uas/UAS.h
index 0d16b9d..ace7372 100644
--- a/src/uas/UAS.h
+++ b/src/uas/UAS.h
@@ -1075,8 +1075,13 @@ protected:
     quint64 lastSendTimeGPS;     ///< Last HIL GPS message sent
     quint64 lastSendTimeSensors;
 
+    QList< QPair<int, QString> >  paramRequestQueue;
+
+
 
 protected slots:
+    void requestNextParamFromQueue();
+
     /** @brief Write settings to disk */
     void writeSettings();
     /** @brief Read settings from disk */
```

walmis, 2015-07-26:
Patch #2:
CopterPidConfig: Disable fields if parameters are not fetched properly. Add Z Accel D field.

```
diff --git a/src/ui/configuration/CopterPidConfig.cc b/src/ui/configuration/CopterPidConfig.cc
index d6a4d04..8c5abe6 100644
--- a/src/ui/configuration/CopterPidConfig.cc
+++ b/src/ui/configuration/CopterPidConfig.cc
@@ -82,6 +82,7 @@ CopterPidConfig::CopterPidConfig(QWidget *parent) : AP2ConfigWidget(parent)
 
     m_nameToBoxMap["ACCEL_Z_P"] = ui.accelZPSpinBox;
     m_nameToBoxMap["ACCEL_Z_I"] = ui.accelZISpinBox;
+    m_nameToBoxMap["ACCEL_Z_D"] = ui.accelZDSpinBox;
     m_nameToBoxMap["ACCEL_Z_IMAX"] = ui.accelZIMAXSpinBox;
     m_nameToBoxMap["ACCEL_Z_FILT_HZ"] = ui.accelZFiltHzSpinBox;
 
@@ -273,17 +274,21 @@ void CopterPidConfig::parameterChanged(int uas, int component, QString parameter
     if (m_nameToBoxMap.contains(parameterName))
     {
         m_nameToBoxMap[parameterName]->setValue(value.toDouble());
+        m_nameToBoxMap[parameterName]->setEnabled(true);
     }
     else if (parameterName == "TUNE_HIGH")
     {
         ui.ch6MaxSpinBox->setValue(value.toDouble() / 1000.0);
+        ui.ch6MaxSpinBox->setEnabled(true);
     }
     else if (parameterName == "TUNE_LOW")
     {
         ui.ch6MinSpinBox->setValue(value.toDouble() / 1000.0);
+        ui.ch6MinSpinBox->setEnabled(true);
     }
     else if (parameterName == "TUNE")
     {
+        ui.ch6OptComboBox->setEnabled(true);
         for (int i=0;i<m_ch6ValueToTextList.size();i++)
         {
             if (m_ch6ValueToTextList[i].first == value.toInt())
@@ -294,6 +299,7 @@ void CopterPidConfig::parameterChanged(int uas, int component, QString parameter
     }
     else if (parameterName == "CH7_OPT")
     {
+        ui.ch7OptComboBox->setEnabled(true);
         for (int i=0;i<m_ch78ValueToTextList.size();i++)
         {
             if (m_ch78ValueToTextList[i].first == value.toInt())
@@ -304,6 +310,7 @@ void CopterPidConfig::parameterChanged(int uas, int component, QString parameter
     }
     else if (parameterName == "CH8_OPT")
     {
+        ui.ch8OptComboBox->setEnabled(true);
         for (int i=0;i<m_ch78ValueToTextList.size();i++)
         {
             if (m_ch78ValueToTextList[i].first == value.toInt())
diff --git a/src/ui/configuration/CopterPidConfig.ui b/src/ui/configuration/CopterPidConfig.ui
index 1624a5d..c2a0b6e 100644
--- a/src/ui/configuration/CopterPidConfig.ui
+++ b/src/ui/configuration/CopterPidConfig.ui
@@ -7,7 +7,7 @@
     <x>0</x>
     <y>0</y>
     <width>677</width>
-    <height>575</height>
+    <height>588</height>
    </rect>
   </property>
   <property name="sizePolicy">
@@ -113,6 +113,9 @@
                <layout class="QVBoxLayout" name="verticalLayout_30">
                 <item>
[... 398 lines cut ...]
+                </property>
                 <property name="decimals">
                  <number>0</number>
                 </property>
@@ -1702,6 +1833,9 @@
               </item>
               <item>
                <widget class="QDoubleSpinBox" name="wpNavSpeedDownSpinBox">
+                <property name="enabled">
+                 <bool>false</bool>
+                </property>
                 <property name="decimals">
                  <number>0</number>
                 </property>
@@ -1715,6 +1849,9 @@
               </item>
               <item>
                <widget class="QDoubleSpinBox" name="wpNavLoiterSpeedSpinBox">
+                <property name="enabled">
+                 <bool>false</bool>
+                </property>
                 <property name="decimals">
                  <number>0</number>
                 </property>
```

billbonney, 2015-07-26:
How are you connecting and which OS?

Also https://help.github.com/articles/fork-a-repo and create a branch* and push your changes to your branch, I can then review your changes much easier.  You can even create a pull request to get your fixes merged to the repo.

*when you make changes in your fork always work on a branch and not master, it will make your life easier.

Thanks for looking at this issue :+1:

walmis, 2015-07-26:
Thanks, I have created a pull request.

I tried connecting with the following methods:
1. UART connection with FTDI cable, 57kbaud
2. USB ttyACM
3. Over MAVProxy UDP.
All of which have the same problem more or less. Refresh button grabs at most 5-7 parameters, all others are lost. I think the request are lost on APM side, since the MAVLink loop runs there at 50hz IIRC, therefore the MAVLink UART buffer fills up with requests faster than APM loop can handle.

OS is Ubuntu 14.10

billbonney, 2015-07-27:
fixed with PR #773

closed 2015-07-27 by billbonney