ArduPilot/apm_planner #779: AC3.3: real-time EKF view

rmackay9, 2015-07-29:
[image]

corresponding Mission Planner issue: https://github.com/diydrones/MissionPlanner/issues/949

rmackay9, 2015-08-20:
We're getting close to the official Copter-3.3 release.  My guess is it'll go out within 3 weeks.  This EKF viewer and the vibration viewer are really important I think because it should allow users to know before hand if the EKF or vibration levels are serious enough that a crash or flyaway could result.

Bill & Co, I know you guys are busy but I just want to highlight the importance of these two items.

billbonney, 2015-10-02:
This is in master

closed 2015-10-02 by billbonney

rmackay9, 2015-10-03:
great news!  thanks!