ArduPilot/apm_planner #781: Missing include in AP2DataPlotThread.cc

Salamandar, 2015-08-02:
Hi there !
I tried to compile the project on Linux (Manjaro Linux, Arch based).
There is a missing include in AP2DataPlotThread.cc. The errors I had before were : 

    src/ui/AP2DataPlotThread.cc: Dans la fonction membre ‘void AP2DataPlotThread::loadBinaryLog(QFile&)’:
    src/ui/AP2DataPlotThread.cc:168:56: erreur: variable ‘QDataStream packetstream’ has initializer but incomplete type
                         QDataStream packetstream(packet);
                                                            ^
    src/ui/AP2DataPlotThread.cc:169:51: erreur: incomplete type ‘QDataStream’ used in nested name specifier
                             packetstream.setByteOrder(QDataStream::LittleEndian);
                                                       ^
    src/ui/AP2DataPlotThread.cc:170:64: erreur: incomplete type ‘QDataStream’ used in nested name specifier
                             packetstream.setFloatingPointPrecision(QDataStream::SinglePrecision);
                                                                    ^
    Makefile:68315: recipe for target 'build-release/obj/AP2DataPlotThread.o' failed
    make: *** [build-release/obj/AP2DataPlotThread.o] Error 1
    make: *** Waiting for unfinished jobs....
    src/ui/dataselectionscreen.cpp:129:67: attention : unused parameter ‘column’ [-Wunused-parameter]
     void DataSelectionScreen::onItemChanged(QTreeWidgetItem* item,int column)
                                                                       ^
    src/ui/qcustomplot.cpp: Dans la fonction membre ‘virtual void QCPGraph::drawLinePlot(QCPPainter*, QVector<QPointF>*) const’:
    src/ui/qcustomplot.cpp:12206:12: attention : unused variable ‘msecs’ [-Wunused-variable]
         qint64 msecs = QDateTime::currentMSecsSinceEpoch();
                ^

and 

    #include <QtCore/QDataStream>

made the trick.

billbonney, 2015-08-02:
What version of Qt are you using? Qt5.5?

Salamandar, 2015-08-02:
Yeah !

billbonney, 2015-08-02:
There's a qt55-wip branch that tracks changes require to get master compiling. Can you create a PR against that branch ?

Salamandar, 2015-08-03:
So… Master does not compile out of the box ?
By the way, I cloned master and this only change was enough to make it compile + work without any problems.
I will do this pull request.

billbonney, 2015-08-03:
This branch tracks changes required to master to make Qt5.5 work https://github.com/diydrones/apm_planner/tree/qt55-wip
If you make changes that allow compilation with Qt5.2.1/Qt5.4 as well with the same code, we can merge it to master.

volalto, 2015-09-07:
I can confirm that adding this include solves the compilation problem. QDataStream is available since Qt 4.8, so it should be OK to include it anyway.

zehortigoza, 2015-09-08:
Same issue here with qt5.5, solved by qt55-wip.

billbonney, 2015-10-02:
This is merged, thx

closed 2015-10-02 by billbonney