ArduPilot/apm_planner #788: text message severity change

rmackay9, 2015-08-25:
For future releases of Copter, Tracker, Plane, Rover the severities of text messages sent from the flight controller will change to be consistent with the MAVLink standards.

Before After
SEVERITY_LOW =1 MAV_SEVERITY_WARNING = 4
SEVERITY_MEDIUM =2 MAV_SEVERITY_ALERT =1
SEVERITY_HIGH =3 MAV_SEVERITY_CRITICAL =2

During the transitional period in which the GCSs want to support firmware which could send either set of severities, the autopilot version could be used (http://mavlink.org/messages/common#AUTOPILOT_VERSION). The first version to support the new severities is below.
Copter: 3, 4, 0, FIRMWARE_VERSION_TYPE_DEV
Plane: 3,4,0,FIRMWARE_VERSION_TYPE_BETA
Rover:
Tracker: 0, 7, 3, FIRMWARE_VERSION_TYPE_DEV

equivalent MP issue: https://github.com/diydrones/MissionPlanner/issues/977

billbonney, 2015-08-26:
The only place I can see this as a real issue is when we calibrate the the access as we check the messages for accel calibration rotation instructions. Do you have a commit hash that is relevant for APM changes?

rmackay9, 2015-08-26:
@billbonney, here's the copter change: https://github.com/diydrones/ardupilot/commit/767b4da5b6852a2ba700ee1f97d83129618b04f1

Here's the PR that covers the rest of the changes. https://github.com/diydrones/ardupilot/pull/2757

DonLakeFlyer, 2015-09-25:
There are a number of messages which seem like they should be MAV_SEVERITY_INFO. For example: when you request the parameter list the firmware sends out mavlink messages with various firmware version info set to MAV_SEVERITY_WARNING. This in turn causes QGroundControl to thinks these are errors since they are marked as warning. Shouldn't these be MAV_SEVERITY_INFO? There may be others. I'm just getting started with APM Flight Stack support in QGC. This is what I've found so far.

pritamghanghas, 2015-10-08:
@rmackay9 Rover is at 2.50. How will Rover every have 2.6, that will be like decremeting the version number. Either rover this is fixed in 2.51 or 3.0. Those are the only two numbers that are in forward direction.

gmorph, 2015-10-08:
Hi.  You are correct.  However,  for the next release of Rover I am going to make it 2.6.0 so the versions fall in line with Plane and Copter.
Thanks, Grant.

pritamghanghas, 2015-10-08:
that will need a special case treatment for rover in groundstation. Why not
make it 3.0.0. Plane and Copter are already 3.x .

Regards,
Pritam

rmackay9, 2015-10-08:
@DonLakeFlyer, yes, I think you're probably right.  If you want to PR it I think we'd accept that without much hesitation.  If a PR is not something you're comfortable with we can add it to the to-do list to do at some point..

gmorph, 2015-10-08:
I'm not super bothered.  A change in the major version number would indicate a significant update which there isn't.  However, if it causes the GCS's a problem then I will figure something out.  I'll have a chat with them.
Thanks, Grant.

pritamghanghas, 2015-10-08:
There is no significant problem on the GCS side. It just looks a little awkward trying to figure out which is newer version and which is older, definitely confusing to somebody who reads the code lateron. 2.50 will become a version wherelogic reverses.

DonLakeFlyer, 2015-10-08:
@rmackay9 I'll add a separate issue for the MAV_SEVERITY_INFO thing.