ArduPilot/apm_planner #790: APM Planner Debian 8 (Jessie)

mijailr, 2015-08-29:
I am unable to compile apm_planner for Debian 8 (Jessie) because i cant solve the following dependencies:
* libopenscenegraph80
* libopenthreads14
* libqt4-declarative:amd64
* libqt4-svg:amd64
* libqt4-test:amd64
* libudev0

Any way to solve this issue?

billbonney, 2015-08-29:
Install an use Qt5, APM Planner 2.0 is not compatible with Qt4. Or download the pre-compiled version. There's more information on compiling for RPi on the support forum http://ardupilot.com/forum/viewforum.php?f=82 
I'm closing this issue, as it's know to work. Thx :+1:

closed 2015-08-29 by billbonney

mikolajzieba, 2016-07-04:
The solution is to compile QT package with a newer version.
There is a script that installs qt5 and apm planner on debian jessie
https://github.com/mikolajzieba/apm_planner_debian_jessie/