ArduPilot/apm_planner #793: Tx failsafe not updating when receiver is sending no pulse

batzkass, 2015-09-01:
As it is the first time I write here, I want first to thank all devs/contributors of this great software.
Now the issue :
I tested the failsafe configuration of my receiver, which is sending no pulses at all if I turn off the transmitter. In mission planner, the PWM for each channel is showing "0" in this case, confirming the right configuration. In APM Planner, the PWM just stops updating, sticking at the last known pulse.
This is embarrassing, as one could think that the receiver is configured to send the last known commands from the transmitter, which is a very bad behaviour ...
Thanks.

billbonney, 2015-09-01:
I have just tested this with Taranis and a FrSKY D4RII and I see the correct behaviour in AP2. All channels go to zero when the TX is turned off. This was using a Pixhawk FC.

Re-open if there is more information of TX/RX and FC used.

Though I recommend support issues are first discussed in the forum, we can raise an issue when we are sure its a defect see http://ardupilot.com/forum/viewforum.php?f=82 for support

closed 2015-09-01 by billbonney

batzkass, 2015-09-03:
Thanks for your answer, and sorry for not having provided much informations. I was using APM Planner 2.0.17 with a HobbyKing Crius clone (AIO V2), and a FrSky D8R-XP.
I tried yesterday with the same receiver but APM Planner 2.0.18 and a 3DR Pixhawk and it worked. I will just check that my Crius Clone works with 2.0.18 and keep you informed if there is an issue.