ArduPilot/apm_planner #794: Incorrect data from PX4FLOW in APM.

migvel, 2015-09-04:
Hello,

I'm using the PX4FLOW device attached to a Pixhawk, I'm having problems when visualizing the ground distance parameter in the APM planner or in the GCS.

The ground distance value is completely erratic, the reason that I post this here is because I checked that the Pixhawk is receiving correctly the data from the PX4FLOW, logging into the NSH console I can test the PX4FLOW application and the data is correct.

Is it possible that there is a problem in the APM or GCS? Otherwise might be in the Arducopter code running above the Pixhawk firmware.

Thanks.

billbonney, 2015-09-05:
APM Planner just reports the value sent by Ardupilot. Can you plot a Graph on the value or post the log?

migvel, 2015-09-07:
Here the graph. 

[image]

As I said the Pixhawk application Pix4FLOW receives the ground_distance correctly.

Thanks.

billbonney, 2015-09-08:
That graph is in QGC. I'd like to see the issue plotted in a graph in AP2 and can you post the log (.bin/tlog) so I can reproduce. I can then check if its the datasteram from the APM for a problem in AP2