ArduPilot/apm_planner #795: Graphing stops after 30mins

billbonney, 2015-09-04:
**keohr** see http://ardupilot.com/forum/viewtopic.php?t=13629&p=37106
I have a problem when using the live Graphs function in APM Planner 2.0. I recently tested it with a own compiled version (2015-08-27, Ubuntu 14.04) but I observed this problem for a longer time in older versions (1/2 to 1 year?). Even with precompiled versions.

When I connect my Pixhawk FC (AUAV-X2) via telemetry radio modem, I can display and watch the graphs for about 30 min after the first connect of telemetry. It does not depend on the start up time of APM Planner2.
Even when I disconnect telemetry, the max display time is still about 30 min after the first connect.

After the moment the graphs disappear, the only way to recover from this error is to restart AMP Planner2. 
Clearing all selections results in only displaying one graph for less than a second (flash) when selecting a single value.
'Show Values Under Mouse' gives an ERR (see pictures)

[image]
[image]

HarrisonHeli, 2015-10-22:
This is an intentional action in the AP2DataPlot2D::updateValue() function in AP2DataPlot2D.cpp.

When 1800000 ms have elapsed, the function drops the previous 30 min of data from the graph. I think this has been done to improve the QCustomPlots performance. QCustomPlot is the object that does the graph drawing for us.

I removed the offending code and have let mission planner log 7h of data. The graph still works and I am able to scroll, but the whole mission planner application is very slow to the point of being unusable.

I can see there is room to add a bit of optimisation to the range of data displayed. We can keep all the data points but just limit the graph to only display 1h of data at time. I will have a play over the weekend.

billbonney, 2015-10-22:
Thx, that would be awesome :+1: