ArduPilot/apm_planner #802: Segfault on existing UDP data

AndKe, 2015-09-21:
I run mavproxy like that:
./mavproxy.py --master=/dev/ttyACM1 --baudrate=115200 --out=udp:127.0.0.1:14550  

upon APM planner start, it arrarently recieves data, says "mode change ..." and crashes..
the funny part is that I did not connect, it "just works" on boot

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff4eaebcb in QCoreApplication::notifyInternal(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5

If I start APM Planner first, then go to UDP settings, and click "disconnect" - I get:

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff4eed394 in QSocketNotifier::type() const ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5

- most likely related ?

billbonney, 2015-09-21:
Odd. I use UDP all the time and o haven't been having that issue. But it's been on OSX. Another bug for Linux.

billbonney, 2015-09-25:
Ok, this is pretty bad... I haven't been able to discover the reason yet. but it's only on Linux.

AndKe, 2015-09-25:
yes, I think it's ironic, Linux being such a IP-orientated OS.  I suspect a
Qt bug.

billbonney, 2015-10-01:
I tested with Qt 5.4.1, 5.4.2, 5.5.0 and they all behave the same. I might check older releases of AP2 to see if there is an older one that works. Finding the seg fault by brute force is seemingly impossible at the mo.

AndKe, 2015-10-02:
Yes, I suspected it had to be a Qt bug, do we know where/how to submit it so the Qt devs are aware of it ?

commit 8f4a7efb (2015-10-11): UDP Link: Fixes #802 Dont use QObjects in different threads

The code added in 20f7b8f to workaround a Qt bug also created a new
issue: it constantly changed the thread that an object was created
without paying attention about the signal / slot connection nor the
thread ownership.

This code moves all object creation from within the UDPLink object
inside the UDPLink thread, and the communication for the thread
contol is done via boolean flags (_restart, _reconnect, _finish, etc
) that the thread mainloop will certainly hit and do what is requested.

Since a single approach is usually better th [... cut]
closed 2015-10-11 by billbonney