ArduPilot/apm_planner #804: scaling and units on dataflash graphs broken

tridge, 2015-09-23:
Using current apmplanner2 on Linux it seems that both scaling of values in DataFlash logs and units are broken.
I am trying to graph BARO.Alt, RFND.Dist1 and RFND.Dist2 in this log:
http://uav.tridgell.net/Lidar/SF10/sf10b-sf10c.bin
All 3 are in meters, but the rangefinder data is of type 'C' and the baro data is of type 'c'. That means both should be scaled by a factor of 100.
The BARO.Alt is scaled correctly by 100 and shows up in meters. The RFND.Dist1 and RFND.Dist2 are scaled by apmplanner2 graphing by a factor of 100*100, so the range of values shows up in the graph from zero to about 1.2, where it should be zero to 120.
There also doesn't seem to be any easy way to make all the graphs scale the same. A button for "same scaling" would be very useful. Otherwise it is difficult to compare values. It can be done now by setting BARO.Alt to 0 to 120, and RFND.Dist1 and RFND.Dist2 to 0 to 1.2, but that seems needlessly complex.
Thanks!

billbonney, 2015-09-23:
Related to #805

billbonney, 2015-09-24:
This is graph of the log, I am seeing the graph show 0-120 for RFND.dist2 and 0-45 for RFND.dist1 and 0-100 for BARO.alt. Is that not correct?
[image]

billbonney, 2015-09-25:
this fix is in this build http://firmware.diydrones.com/Tools/APMPlanner/daily/2015-09-16/ for the issue you describe

Arne-W, 2017-03-17:
fixed in current master

closed 2017-03-17 by Arne-W