ArduPilot/apm_planner #808: Corruption /segfault when loading mossion file more than once.

AndKe, 2015-09-27:
Mission part one:  https://gist.github.com/AndKe/6b4e56e22d3b32ca65d4
If you load this mission (open same file twice) - then AP2 will crash with 
'''
09-27T14:59:32.886 src/ui/map/QGCMapWidget.cc 736 UPDATING WP FUNCTION CALLED 
DEBUG 2015-09-27T14:59:32.886 UPDATING WAYPOINT 46 IN 2D MAP 
DEBUG 2015-09-27T14:59:32.887 src/ui/map/QGCMapWidget.cc 736 UPDATING WP FUNCTION CALLED 
DEBUG 2015-09-27T14:59:32.887 UPDATING WAYPOINT 47 IN 2D MAP 
DEBUG 2015-09-27T14:59:32.887 src/ui/map/QGCMapWidget.cc 736 UPDATING WP FUNCTION CALLED 
DEBUG 2015-09-27T14:59:32.887 UPDATING WAYPOINT 48 IN 2D MAP 
[Thread 0x7fffbd653700 (LWP 15145) exited]
[New Thread 0x7ffeb67fc700 (LWP 15191)]
[New Thread 0x7ffed0ff9700 (LWP 15190)]
[New Thread 0x7ffed1ffb700 (LWP 15173)]
[New Thread 0x7ffed27fc700 (LWP 15172)]
[New Thread 0x7ffed2ffd700 (LWP 15171)]
[New Thread 0x7ffed37fe700 (LWP 15170)]
[New Thread 0x7ffed3fff700 (LWP 15169)]
[New Thread 0x7fff68ff9700 (LWP 15168)]
[New Thread 0x7fff697fa700 (LWP 15167)]
[New Thread 0x7fff69ffb700 (LWP 15166)]
[New Thread 0x7fff6a7fc700 (LWP 15165)]
[New Thread 0x7fff6affd700 (LWP 15164)]
[New Thread 0x7fff6b7fe700 (LWP 15163)]
[New Thread 0x7fff6bfff700 (LWP 15162)]
[New Thread 0x7fff80c3e700 (LWP 15160)]
[New Thread 0x7fff8183f700 (LWP 15159)]
[New Thread 0x7fff827fc700 (LWP 15157)]
[New Thread 0x7fff837fe700 (LWP 15155)]
[New Thread 0x7fffbe757700 (LWP 15144)]
[New Thread 0x7fffbf8cf700 (LWP 15143)]
[New Thread 0x7fffcde1f700 (LWP 15139)]
[New Thread 0x7fffcf72a700 (LWP 15136)]
[New Thread 0x7fffd4906700 (LWP 15135)]
[New Thread 0x7fffe0c7d700 (LWP 15133)]

Program received signal SIGSEGV, Segmentation fault.
0x0000000000740e69 in WaypointEditableView::setCurrent(bool) ()
'''

also, after working on one mission, loading the other gives wrong UI for some items like takeoff (like QGCmissionNavTakeoff.ui)  is not shown, but another with less fields is used.

I suspect there is some "resetting" that needs to be done, because any mission loading/editing works 100% when used for he first time after application start.

https://gist.github.com/AndKe/fdf5d6351e3dff7aa449

(And before anyone mention that the attached missions can't be VLOS and think this small piece of the missions are a danger to other traffic, - please observe that I have a CAA approval and NOTAM's active https://www.ippc.no/ippc/aerodromes.jsp?AED=ENSR&FIR=ENOR)

AndKe, 2015-09-27:
additional info: clicking "delete all waypoints" before loading new file fixes the problem, so the fix may be very easy :)

tcanabrava, 2015-10-02:
I'll get this and fix it on monday - going home right now.

commit 6ca8ce60 (2015-11-24): Waypoint Manager: Fixes #808 - segfault when loading mission file

The waypoint list and the GUI elements need to be kept in sync. When all waypoint elements are deleted the GUI needs to be refreshed.
closed 2015-11-24 by billbonney