ArduPilot/apm_planner #809: apmplanner2 crash on Linux

0xCCD, 2015-09-29:
**Version:** APM Planner v2.0.19-rc1 (697d452c-2.0.18-29-g697d452-DEV) => Compiled on Ubuntu 15.04
**Firmware:** Latest stable v3.2.1 

If i connect through USB to the IRIS+ the application crashes with the following message! 

```
DEBUG 2015-09-29T00:02:14.211 APM: Text Message rx'd "PreArm: Need 3D Fix"
```

If i connect first through the Android App (Tower) and disable the PreArm check temporarily, APM Planner does not crash and works as expected ! 

**FYI:** I'm indoor and try to configure my controller, so i will never get a 3D Fix, looks like a bug to me, another problem is i can't start the app without sudo even though my user is in the dialout group and i've added the udev rules, ends in a loop, but thats another story 

```
 INFO 2015-09-29T09:05:33.114 Start Link Manager 
DEBUG 2015-09-29T09:05:33.115 Create MAVLinkDecoder:  MAVLinkDecoder(0x266be50) 
 INFO 2015-09-29T09:05:33.115 Start UAS Manager 
DEBUG 2015-09-29T09:05:33.186 Creating MainWindow 
 INFO 2015-09-29T09:05:33.355 Creating SubMainWindow:  SubMainWindow(0x2896a60) 
 INFO 2015-09-29T09:05:33.355 setting objectName:  "VIEW_MISSION" 
 INFO 2015-09-29T09:05:33.383 Creating SubMainWindow:  SubMainWindow(0x28a00e0) 
 INFO 2015-09-29T09:05:33.383 setting objectName:  "VIEW_FLIGHT" 
 INFO 2015-09-29T09:05:33.396 Creating SubMainWindow:  SubMainWindow(0x287df20) 
 INFO 2015-09-29T09:05:33.396 setting objectName:  "VIEW_HARDWARE_CONFIG" 
DEBUG 2015-09-29T09:05:33.564 fillPortsInfo  
DEBUG 2015-09-29T09:05:33.667 Creating MainWindow 
 INFO 2015-09-29T09:05:33.812 Creating SubMainWindow:  SubMainWindow(0x3fc7630) 
 INFO 2015-09-29T09:05:33.812 setting objectName:  "VIEW_MISSION" 
 INFO 2015-09-29T09:05:33.829 Creating SubMainWindow:  SubMainWindow(0x4088380) 
 INFO 2015-09-29T09:05:33.829 setting objectName:  "VIEW_FLIGHT" 
 INFO 2015-09-29T09:05:33.843 Creating SubMainWindow:  SubMainWindow(0x40ded30) 
 INFO 2015-09-29T09:05:33.843 setting objectName:  "VIEW_HARDWARE_CONFIG" 
DEBUG 2015-09-29T09:05:33.985 fillPortsInfo  
DEBUG 2015-09-29T09:05:34.073 Creating MainWindow 
 INFO 2015-09-29T09:05:34.298 Creating SubMainWindow:  SubMainWindow(0x56077e0) 
 INFO 2015-09-29T09:05:34.298 setting objectName:  "VIEW_MISSION" 
 INFO 2015-09-29T09:05:34.320 Creating SubMainWindow:  SubMainWindow(0x56b0c70) 
 INFO 2015-09-29T09:05:34.320 setting objectName:  "VIEW_FLIGHT" 
 INFO 2015-09-29T09:05:34.333 Creating SubMainWindow:  SubMainWindow(0x570b2a0) 
 INFO 2015-09-29T09:05:34.333 setting objectName:  "VIEW_HARDWARE_CONFIG" 
DEBUG 2015-09-29T09:05:34.479 fillPortsInfo  
DEBUG 2015-09-29T09:05:34.567 Creating MainWindow 
 INFO 2015-09-29T09:05:34.861 Creating SubMainWindow:  SubMainWindow(0x725d640) 
 INFO 2015-09-29T09:05:34.862 setting objectName:  "VIEW_MISSION" 
 INFO 2015-09-29T09:05:34.883 Creating SubMainWindow:  SubMainWindow(0x72f8300) 
 INFO 2015-09-29T09:05:34.883 setting objectName:  "VIEW_FLIGHT" 
 INFO 2015-09-29T09:05:34.897 Creating SubMainWindow:  SubMainWindow(0x73497e0) 
 INFO 2015-09-29T09:05:34.897 setting objectName:  "VIEW_HARDWARE_CONFIG" 
DEBUG 2015-09-29T09:05:35.049 fillPortsInfo  
DEBUG 2015-09-29T09:05:35.146 Creating MainWindow 
 INFO 2015-09-29T09:05:35.516 Creating SubMainWindow:  SubMainWindow(0x8f26500) 
 INFO 2015-09-29T09:05:35.516 setting objectName:  "VIEW_MISSION" 
 INFO 2015-09-29T09:05:35.538 Creating SubMainWindow:  SubMainWindow(0x8fba8c0) 
 INFO 2015-09-29T09:05:35.538 setting objectName:  "VIEW_FLIGHT" 
 INFO 2015-09-29T09:05:35.552 Creating SubMainWindow:  SubMainWindow(0x9010220) 
 INFO 2015-09-29T09:05:35.552 setting objectName:  "VIEW_HARDWARE_CONFIG" 
DEBUG 2015-09-29T09:05:35.697 fillPortsInfo  
DEBUG 2015-09-29T09:05:35.791 Creating MainWindow 
 INFO 2015-09-29T09:05:36.242 Creating SubMainWindow:  SubMainWindow(0xab6d0b0) 
 INFO 2015-09-29T09:05:36.242 setting objectName:  "VIEW_MISSION" 
 INFO 2015-09-29T09:05:36.264 Creating SubMainWindow:  SubMainWindow(0xac13900) 
 INFO 2015-09-29T09:05:36.264 setting objectName:  "VIEW_FLIGHT" 
 INFO 2015-09-29T09:05:36.276 Creating SubMainWindow:  SubMainWindow(0xac30d60) 
 INFO 2015-09-29T09:05:36.276 setting objectName:  "VIEW_HARDWARE_CONFIG" 
DEBUG 2015-09-29T09:05:36.420 fillPortsInfo  
DEBUG 2015-09-29T09:05:36.513 Creating MainWindow 
 INFO 2015-09-29T09:05:37.048 Creating SubMainWindow:  SubMainWindow(0xbf2f820) 
 INFO 2015-09-29T09:05:37.048 setting objectName:  "VIEW_MISSION" 
 INFO 2015-09-29T09:05:37.077 Creating SubMainWindow:  SubMainWindow(0xc85d4c0) 
 INFO 2015-09-29T09:05:37.077 setting objectName:  "VIEW_FLIGHT" 
 INFO 2015-09-29T09:05:37.089 Creating SubMainWindow:  SubMainWindow(0xc8aea90) 
 INFO 2015-09-29T09:05:37.089 setting objectName:  "VIEW_HARDWARE_CONFIG" 
DEBUG 2015-09-29T09:05:37.253 fillPortsInfo  
DEBUG 2015-09-29T09:05:37.350 Creating MainWindow 
 INFO 2015-09-29T09:05:37.967 Creating SubMainWindow:  SubMainWindow(0xdb73650) 
 INFO 2015-09-29T09:05:37.967 setting objectName:  "VIEW_MISSION" 
 INFO 2015-09-29T09:05:37.999 Creating SubMainWindow:  SubMainWindow(0xe4a2d50) 
 INFO 2015-09-29T09:05:37.999 setting objectName:  "VIEW_FLIGHT" 
 INFO 2015-09-29T09:05:38.016 Creating SubMainWindow:  SubMainWindow(0xe4f4530) 
 INFO 2015-09-29T09:05:38.016 setting objectName:  "VIEW_HARDWARE_CONFIG" 
DEBUG 2015-09-29T09:05:38.160 fillPortsInfo  
DEBUG 2015-09-29T09:05:38.251 Creating MainWindow 
```

billbonney, 2015-09-29:
Does the same happen with a pre-built version. I see yours has modifications http://firmware.diydrones.com/Tools/APMPlanner/daily/2015-09-26/

PS: Support is usually handled here, http://ardupilot.com/forum/viewforum.php?f=82

0xCCD, 2015-10-01:
I've tried the pre-built version earlier and encountered the same issue (does not start as root and crash), it was the main reason to try the self compiled version, i will move to the forum now, many thanks for the link

closed 2015-10-01 by 0xCCD