ArduPilot/apm_planner #817: Graphing logs not working on OS X El Capitan

jmachuca77, 2015-10-02:
I just upgraded my OS and it seems the log graphing is broken. Not 100% sure its due to the OS though.
[image]

jmachuca77, 2015-10-02:
That is a screenshot after loading a log.

billbonney, 2015-10-02:
I just tested on OSX 10.11 and its working ok for me. Do you have more details on how to reproduce. ie APM Planner 2.0 version.
Also, try this build as it has some fixes in graphing (to color slection) http://firmware.diydrones.com/Tools/APMPlanner/daily/2015-10-01/apm_planner_2.0.18-35-g8fdbd34_osx.dmg

jmachuca77, 2015-10-02:
Im using 2.0.17, and the log is a DF log from SOLO. I-ll try the new build.

jmachuca77, 2015-10-02:
I just checked with the new build and it works fine.

closed 2015-10-02 by jmachuca77