ArduPilot/apm_planner #822: Taranis telemetry stops working when AP2 connects

rmackay9, 2015-10-09:
Reported by Richard Kennedy on the Copter-3.3 beta testing discussion.
http://diydrones.com/forum/topics/copter-3-3-beta-testing?commentId=705844%3AComment%3A2107426&xg_source=msg_com_forum

rmackay9, 2015-10-09:
Here's the discussion in the APM:Forum.
http://ardupilot.com/forum/viewtopic.php?f=111&t=13967&sid=2db8f1e6cae5b23c2c69f7f238659309

billbonney, 2015-10-09:
You can set custom telemetry rates for AP2. I'm not sure why your teensy is failing, it could be that AP2 exposes messages it can't handle gracefully.
[image]

billbonney, 2015-10-11:
The solution is probably to lower the attitude message rate to 5Hz and Extra 2 to 5Hz, in line with what the Teensy expects. The issue is with the teensy getting two many updates and not requesting what it needs from the autopilot. It's probably just swamped with messages

Not a defect,

closed 2015-10-11 by billbonney