ArduPilot/apm_planner #823: connected by UDP, then reconnecting pixawk=crash

AndKe, 2015-10-11:
Reproduce: 
-connect  a Pixhawk to USB
-mavproxy.py --out=udp:127.0.0.1:14550
-start apmplanner2   (observe that it will connect to UDP and work great)
-disconnect Pixhawk USB - observe that MavProxy and apmplanner2 will handle that fine (connection lost)
-now reconnect Pixhawk USB - observe that Mavproxy will resume connection fine, but apmplanner will crash well after receiving data.

```
[Thread 0x7fff6b7fe700 (LWP 7707) exited]
qml: HBD: Heartbeat stopped
DEBUG 2015-10-11T21:24:35.753 DRAW LOITER 
DEBUG 2015-10-11T21:24:35.780 DRAW LOITER 
 INFO 2015-10-11T21:24:40.447 "/home/andre/apmplanner2/tmp_audio/link_lost_to_system_1.wav"  file exist, playing this file 
[Thread 0x7fffbd7fd700 (LWP 7682) exited]
DEBUG 2015-10-11T21:24:42.580 APM: Text Message rx'd "Demo Servos!" 
 INFO 2015-10-11T21:24:42.581 STATUS TEXT: 4 : "Demo Servos!" 
qml: HBD: Heartbeat started
DEBUG 2015-10-11T21:24:42.587 UAS: new system_status 2 
DEBUG 2015-10-11T21:24:42.587 UAS: new base mode  17 
DEBUG 2015-10-11T21:24:42.587 MODE: 17 
DEBUG 2015-10-11T21:24:42.588 "D|STABILIZED" 
DEBUG 2015-10-11T21:24:42.588 UAS: new custom mode  16 
DEBUG 2015-10-11T21:24:42.589 APM: getCustomModeText() 
DEBUG 2015-10-11T21:24:42.589 plotTextArrow: 21  to  "MODE" 
DEBUG 2015-10-11T21:24:42.589 APMToolBar::mode: "Initializing" 
DEBUG 2015-10-11T21:24:42.590 APM: getCustomModeText() 
DEBUG 2015-10-11T21:24:42.590 APM say: "Mode changed to Initializing for system 1" 
 INFO 2015-10-11T21:24:42.591 "/home/andre/apmplanner2/tmp_audio/mode_changed_to_initializing_for_system_1.wav"  file exist, playing this file 
DEBUG 2015-10-11T21:24:42.592 APM: Text Message rx'd "Initialising APM..." 
 INFO 2015-10-11T21:24:42.592 STATUS TEXT: 4 : "Initialising APM..." 
 INFO 2015-10-11T21:24:42.947 "/home/andre/apmplanner2/tmp_audio/link_regained_to_system_1_after_12_seconds.wav"  file not exist, create a new one 
VAL: tried to access cart in -1 type val
[Thread 0x7fffa2ffd700 (LWP 7693) exited]
[Thread 0x7fffcf7fe700 (LWP 7679) exited]
[Thread 0x7fffe43707c0 (LWP 7665) exited]
[Inferior 1 (process 7665) exited with code 0377]
```

billbonney, 2015-10-11:
Qt version? have tried a newer Qt version?

AndKe, 2015-10-11:
Qt5.4.1
I strongly doubt it have anything to do with QT, judging by when it happens,  I would guess something does not like the sudden rebooted device while apparently continously connected.

billbonney, 2015-10-11:
I've just tested on OSX, and other than the serial port naming changing in Qt5.5 again (I have a fix), I could not replicate. Need to test on Linux

AndKe, 2015-10-12:
tested something different, here are the results:
connected as:
Pixhawk->radio ... radio->USB->Mavproxy->UDP_out->apmplaner2
If I pull the USB radio, then reconnect, everything works fine, no crash

connected as:
Pixhawk USB->Mavproxy->UDP_out->apmplaner2
If I pull the USB, then reconnect, or command preflight reboot, apmplanner2 will crash (after recieving booting messages, about where the usual mavlink packets would start)

I think it's very strange it could be OS dependent.

billbonney, 2015-10-12:
The data stream is just connected to AP2 via UDP. Mmm why the serial connection effects it is weird.

AndKe, 2015-10-12:
Yes, running AP2  on another computer, using UDP, works perfectly. 
Another clue: 
Changing mode using AP2 on computer with physical USB connection, crashes APM planner after 0-2 changes.
Changing mode using AP2 on remote computer, works as expected.

billbonney, 2015-10-12:
It's the interaction of AP2 and MAVProxy using the serial device. can't say I test with a Pixhawk connected to a another program and then run AP2. I tried to replicate by connecting screen tothe serial port presented by the pixhawk, and then tested UDP coming from SITL on another machine. It worked. But then I would expected it two.
I should look at getting MAVProxy (SITL) running natively on OSX (I just run SITL on a headless linux machine)

billbonney, 2015-11-02:
@AndKe Can you verify if this is now fixed in latest master?

AndKe, 2015-11-02:
Yes, it works perfectly now.

closed 2015-11-02 by billbonney