ArduPilot/apm_planner #829: Empty widgets for EKF and VIBE on latest RC running on OS X (latest)

lvale, 2015-10-26:
they say a picture is worth a thousand words.....

[image]

lostcontrol, 2015-10-26:
Same on Linux (Ubuntu 15.04)

billbonney, 2015-10-27:
I tested this as built from source and it works. Not sure why the packaged builds fail with no widgets. Needs investigation

khancyr, 2015-10-29:
I confirmed that it is not working on ubuntu 15.04 , and windows (64bits)

dcarpy, 2015-11-05:
It has something to do with the following (at run time):
DEBUG 2015-11-04T10:07:08.625  QUrl( "file:///usr/share/APMPlanner2/qml/VibrationMonitor.qml" )  
file:///usr/share/APMPlanner2/qml/VibrationMonitor.qml:54:9: Type BarGauge unavailable 
             BarGauge { 
             ^
file:///usr/share/APMPlanner2/qml/components/BarGauge.qml:3:1: module "QtQuick.Controls.Styles" is not installed 
     import QtQuick.Controls.Styles 1.2 
     ^
file:///usr/share/APMPlanner2/qml/components/BarGauge.qml:2:1: module "QtQuick.Controls" is not installed 
     import QtQuick.Controls 1.2 
     ^
file:///usr/share/APMPlanner2/qml/components/BarGauge.qml:3:1: module "QtQuick.Controls.Styles" is not installed 
     import QtQuick.Controls.Styles 1.2 
     ^
file:///usr/share/APMPlanner2/qml/components/BarGauge.qml:2:1: module "QtQuick.Controls" is not installed 
     import QtQuick.Controls 1.2 
     ^

billbonney, 2015-11-05:
I'll take a look later. Please report Qt version when reporting Linux releases since that can be an issue. OSX Qt dependency is bundled with the package.

dcarpy, 2015-11-06:
I've tested both Qt 5.2.1 and Qt 5.4.1 with the apm_planner_2.0.19-rc4-7-g0ce96ed_ubuntu_trusty64.deb daily build and both have the same issue.

dcarpy, 2015-11-06:
Works great when I build on Windows using Qt 5.4.2.  Weird.

billbonney, 2015-11-08:
I have fixed it for OSX (tested), Windows (still needs to be verified the new changes work)

lvale, 2015-11-10:
@billbonney 
It does have content but it's not connected to data?

[image]
[image]

billbonney, 2015-11-10:
It's fixed for VIBE/EKF Monitor. The issue you are seeing is that the title bar gets out of sync with which link is connected. you may have another link connected. if you are connected with a real device, if you table you will see the vibration values change etc..

closed 2015-11-10 by billbonney

lvale, 2015-11-10:
@billbonney 
You lost me.
There is only one PixHawk connected via USB. Opened the widgets and moved and resized them to the previous screen cap.

When selected they appear squashed to the right side of the screen

[image]

so I have to move them and resize them.

[image]

And the widgets "follow" no matter which screen we go they come :)

[image]
[image]
[image]

lvale, 2015-11-10:
and even deselected they stay on screen:

[image]

btw:

OSX

[image]

billbonney, 2015-11-10:
This issue was closed. Please raise a new one for other issues. It's know issue, once the boxes have been resized, it remembered for the next run.

Try this build http://firmware.diydrones.com/Tools/APMPlanner/daily/2015-11-07/ as it seems to allow me to close the windows using the menu checklist (Please try latest daily first before posting an issue here)

You can resize the widgets when docked, and then they don't move to the other view, they only do that when undocked.

Post logs from ~/apmplanner2 folder is sometimes helpful.

If you write you use OSX10.10.1 I'll believe you, save yourself doing a screen grab everytime.