ArduPilot/apm_planner #834: data from .tlog playback crashes AP2 when speaking

AndKe, 2015-11-05:
mavplayback is playing back a .tlog
AP2 starts and hooks up to that just fine, but crashes whenever something should be spoken.
if AP2 needs to speak while connected by telemetry or USB, it works fine.

there is no crash if  "Mute audio output" is enabled.

```
 DEBUG 2015-11-04T14:59:58.557 APM: Text Message rx'd "Throttle armed!"
INFO 2015-11-04T14:59:58.559 STATUS TEXT: 3 : "Throttle armed!"
DEBUG 2015-11-04T14:59:58.934 ApmSoftwareConfig::populateTimerTick() - No Param names from param manager. Sleeping for one second...
INFO 2015-11-04T14:59:59.111 Arming State Changed: ARMED
DEBUG 2015-11-04T14:59:59.112 APM say: "armed"
INFO 2015-11-04T14:59:59.112 "/home/andre/apmplanner2/tmp_audio/system_10_is_armed.wav" file not exist, create a new one
VAL: tried to access cart in -1 type val
and the file created is 0 bytes.
```

0xCCD, 2015-11-05:
+1 
I have a similair issue, it always crashs if a file does not exist, after an restart it works until the next file needs to be created :) 

Workaround, mute all sounds within apmplanner:)

billbonney, 2015-11-05:
Qt versions? Platform? I don't see this on OSX. And it something that has happened recently, so I think ti a Qt related behaviour change exposing a race condition (I have random crashing if audio is enabled on linux)

AndKe, 2015-11-05:
Qt5.4.2 , Ubuntu 15.10 64bit
I don't see other/random crashing on voice, but *occasionally* (maybe 1/20) AP2 will crash shortly after getting telemetry link (this could be voice related)    - once I had an "feeling" that it crashed when I expected to hear more than one "new" message (generating more then one .wav ?)

however - this .tlog playback method is 100% reproducible , and may be a clue as to what's going on.

0xCCD, 2015-11-05:
I use the same Ubuntu and QT Version, but i had the same issues with Ubuntu 15.04 (unfortunately i don't know which QT was running their)

AndKe, 2015-11-05:
In general : this is what Ubuntu users normally have
Ubuntu 14.10 used Qt5.3.0
Ubuntu 15.04 used Qt5.4.1
Ubuntu 15.10 uses Qt5.4.2

Qt5.5.1 is in testing, with some issues.

billbonney, 2015-11-05:
I thinking of making Qt5.4.2 the minimum, since the video stuff that I have currently requires it. And i ask only that I use Qt5.4.2/5.5.1 only on the various builds. I need to pull Windows Installer Package up to Qt5.4.2 minimum as well (though may jump to Qt5.5.1 for better Win10 support)

What issue are you seeing with Qt5.5.1 on Linux?

AndKe, 2015-11-05:
https://bugs.launchpad.net/bugs/+bugs?field.searchtext=&orderby=-importance&search=Search&field.status%3Alist=NEW&field.status%3Alist=OPINION&field.status%3Alist=CONFIRMED&field.status%3Alist=TRIAGED&field.status%3Alist=INPROGRESS&field.status%3Alist=INCOMPLETE_WITH_RESPONSE&field.status%3Alist=INCOMPLETE_WITHOUT_RESPONSE&assignee_option=any&field.assignee=&field.bug_reporter=&field.bug_commenter=&field.subscriber=&field.tag=qt5.5&field.tags_combinator=ANY&field.status_upstream-empty-marker=1&field.has_cve.used=&field.omit_dupes.used=&field.omit_dupes=on&field.affects_me.used=&field.has_patch.used=&field.has_branches.used=&field.has_branches=on&field.has_no_branches.used=&field.has_no_branches=on&field.has_blueprints.used=&field.has_blueprints=on&field.has_no_blueprints.used=&field.has_no_blueprints=on

billbonney, 2015-11-05:
@AndKe sure software has bugs, but how does that effect APM Planner 2.0. It seems to be working and compiling great for me. Ubuntu Video Player may not be working with Qt5.5, but APM Planner 2.0 is.

AndKe, 2015-11-05:
As I understood it, it's not "Ubuntu Video player"  - but video playback in general that does not work in 5.5.1 - so I suspect AP2 's video streaming/playback/camera device will not work with 5.5.1  - where can I find any info about the feature, and how to test it ?

billbonney, 2015-11-05:
Checkout the Video-wip branch and there is some more info on the ardupilot forum

mrpilot2, 2015-11-28:
This bug was introduced when porting to flite2 in pull request #812 

I can't find a common linux distribution (Ubuntu, CentOs, Fedora, ..) that supports flite2. In Ubuntu flite2 is considered as unstable.

I'm trying to find a way to support both versions of flite.

billbonney, 2015-11-28:
@mrpilot2 Great find. It was starting to annoy me :-) if i get time I'll revert this change until we have a fix.

commit 6f4ec1fb (2015-12-09): Audio: Fix #834: Revert "Audio: Fixes #812 Port to flite2"

This reverts commit 5f0890a72bf227fc010c68517fe91b44efa850e1.
closed 2015-12-09 by billbonney