ArduPilot/apm_planner #835: Unable to flash pixhawk

AndKe, 2015-11-10:
Ubuntu 15.10 + AP 2.0.19-rc4 (current master)
Unable to upload firmware to any pixhawk (thought at first it was blank/new pixhawks only)

this is what I see:
[image]

This (contradicting) text appears below if I click "stable"  
[image]

and this is what I see when tryiong to upload "custom" firmware (after selecting firmware, no instructions to dis/reconnect pixhawk is given, and the progress bar is going nowhere.)
[image]

In earlier versions/before, flashing was not a problem.
Tested on three computers, with four different Pixhawks (what I had at hand at office/home+laptop) - all are real 3DR products.
There is no problem connecting to the Pixhawks using USB
Log:  (strangely many "inserting" lines, but ttyACM0 seems steady in GUI)
DEBUG 2015-11-10T20:05:56.774 "retrieve versionobject from server: http://firmware.diydrones.com/Tools/APMPlanner/apm_planner_version.json" 
DEBUG 2015-11-10T20:05:57.152 Downloading: 2815 / 2815 
DEBUG 2015-11-10T20:05:57.153 Downloading: 2815 / 2815 
DEBUG 2015-11-10T20:05:57.153 AutoUpdateCheck::httpFinished() 
DEBUG 2015-11-10T20:05:57.273 Requesting firmware: "stable" "" 
ERROR 2015-11-10T20:05:57.275 Unknown autopilot in ApmFirmwareConfig::requestFirmwares() "" 
DEBUG 2015-11-10T20:05:57.387 Inserting  "ttyS4" 
DEBUG 2015-11-10T20:05:57.387 Inserting  "ttyACM0" 
DEBUG 2015-11-10T20:05:57.388 Inserting  "ttyS0" 
DEBUG 2015-11-10T20:05:59.387 Inserting  "ttyS4" 
DEBUG 2015-11-10T20:05:59.387 Inserting  "ttyACM0" 
DEBUG 2015-11-10T20:05:59.388 Inserting  "ttyS0" 
DEBUG 2015-11-10T20:06:01.387 Inserting  "ttyS4" 
DEBUG 2015-11-10T20:06:01.388 Inserting  "ttyACM0" 
DEBUG 2015-11-10T20:06:01.388 Inserting  "ttyS0" 
DEBUG 2015-11-10T20:06:03.387 Inserting  "ttyS4" 
DEBUG 2015-11-10T20:06:03.388 Inserting  "ttyACM0" 
DEBUG 2015-11-10T20:06:03.388 Inserting  "ttyS0" 
DEBUG 2015-11-10T20:06:05.386 Inserting  "ttyS4" 
DEBUG 2015-11-10T20:06:05.386 Inserting  "ttyACM0"

AndKe, 2015-11-10:
the problem is with variable "autopilot" being empty:
logline
DEBUG 2015-11-10T20:05:57.273 Requesting firmware: "stable" "" 
is supposed to contain "autopilot"
QLOG_DEBUG() << "Requesting firmware:" << type << autopilot;
this variable is used to detect which firmware to request.

billbonney, 2015-11-10:
You have a radio connected at the same time?

AndKe, 2015-11-10:
nope - this are new pixhawks, (never flashed, and some programmed before) - no thing plugged ito them, no other ttyUSB or ttyACM devices on the computer(s)

billbonney, 2015-11-10:
That's what I means, what's plugged into the computer. AP2 looks like its reading a port that is not a PH, and then get confused. with only a PH connected it should flash correctly

AndKe, 2015-11-10:
not sure what you mean there, I just wrote that there are no ttyACM* or ttyUSB* connected 
(except of one pixhawk at a time) 
ttyACM0 pops up correctly every time pixhawk is connected, and is correctly described on the right side as "PX4 FMU 2.x 3D Robotics"  - yet ,the mentioned variable is empty, and device "unrecognised"

billbonney, 2015-11-10:
Sorry, trying to do too many thongs at the sometime ;) 

I need to test this on linux, It works as expected on OSX, and I am not near my Linux machine at the moment.

AndKe, 2015-11-24:
were you able to reproduce it ?

billbonney, 2015-11-26:
I looked at it and the Pixhawk now reports the product Identifier as 0x27CA on linux. I post a fix.

billbonney, 2015-11-26:
This should fix it, but I am not sure why the product identifier has changed 
https://github.com/diydrones/apm_planner/commit/16489dd112b498ae79b31eb993723d7fa69b3db1

closed 2015-11-26 by billbonney

AndKe, 2015-11-26:
sorry to say, a clean build of (16489dd1-2.0.19-rc4-19-g16489dd)
still does not work
```DEBUG 2015-11-26T14:08:32.335 Requesting firmware: "stable" "" 
ERROR 2015-11-26T14:08:32.336 Unknown autopilot in ApmFirmwareConfig::requestFirmwares() "" 
DEBUG 2015-11-26T14:08:32.851 Inserting  "ttyS4" 
DEBUG 2015-11-26T14:08:32.851 Inserting  "ttyACM0" 
DEBUG 2015-11-26T14:08:32.851 Inserting  "ttyS0" 
DEBUG 2015-11-26T14:08:34.851 Inserting  "ttyS4" 
DEBUG 2015-11-26T14:08:34.851 Inserting  "ttyACM0" 
DEBUG 2015-11-26T14:08:34.852 Inserting  "ttyS0" 
DEBUG 2015-11-26T14:08:36.851 Inserting  "ttyS4" 
DEBUG 2015-11-26T14:08:36.851 Inserting  "ttyACM0" 
DEBUG 2015-11-26T14:08:36.852 Inserting  "ttyS0" 
DEBUG 2015-11-26T14:08:37.490 flashButtonClicked 
DEBUG 2015-11-26T14:08:37.491 Flash button clicked without any HTTP links! 
DEBUG 2015-11-26T14:08:37.491 Autopilot: "" 
DEBUG 2015-11-26T14:08:37.491 Firmware: "stable" 
DEBUG 2015-11-26T14:08:37.492 Set COM port: "ttyACM0" 
```

I find it very strange that the right side of the serial port box says "PX4 FMU v2.x 3D Robotics" apparently that part of detection works fine, and something else fails. 
And " Requesting firmware: "stable" ""  is still supposed to contain "autopilot"

AndKe, 2015-11-26:
new comment just so you "discover" it.
I've added some debugging, and found that 
Detected: 7718 "PX4 FMU v2.x"

7718d , (0x1E26)   -while you see 0x27CA  
so I tried same Pixhawk on another system (same Ubuntu) sees 7206d   - also , same number while bootloader or booted. 

The productIdentifier is 100% reproducible, no amount of plugging/unplugging changes them , 
While the real VID/PID from lsusb shows always 26ac:0011  - on both systems... during boot or operation.
So if productidentifier is meant to show PID, (which is apparently normal) could this be some bizzare Qt bug ?

AndKe, 2015-11-26:
confirmed solved on two different systems (seee above) with this fix https://github.com/diydrones/apm_planner/pull/840

billbonney, 2015-11-26:
Yep. Looks like some Qt bug. I'll merger your fix later today. Thx

billbonney, 2015-11-27:
Fixed with https://github.com/diydrones/apm_planner/commit/98c08f19d30dcebc366f91e27130a8e6da4de369