ArduPilot/apm_planner #838: MSVC2012: Build failure with glu32.lib not found

billbonney, 2015-11-25:
Build Failure with MSVC2012 and glulib not found.
```
[23:27:37][Step 2/7] moc_Settings.cpp
[23:27:39][Step 2/7] 	link /NOLOGO /DYNAMICBASE /NXCOMPAT /INCREMENTAL:NO /SUBSYSTEM:CONSOLE "/MANIFESTDEPENDENCY:type='win32' name='Microsoft.Windows.Common-Controls' version='6.0.0.0' publicKeyToken='6595b64144ccf1df' language='*' processorArchitecture='*'" /MANIFEST /MANIFESTFILE:release\apmplanner2.exe.embed.manifest /OUT:release\apmplanner2.exe @C:\BuildAgent\temp\buildTmp\nm31F5.tmp
[23:27:42][Step 2/7] LINK : fatal error LNK1181: cannot open input file 'glu32.lib'
[23:27:42][Step 2/7] NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio 11.0\VC\BIN\link.EXE"' : return code '0x49d'
[23:27:42][Step 2/7] Stop.
[23:27:42][Step 2/7] NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio 11.0\VC\BIN\nmake.exe"' : return code '0x2'
[23:27:42][Step 2/7] Stop.
[23:27:42][Step 2/7] Process exited with code 2
[23:27:43][Step 2/7] Step Build (Command Line) failed
```

dcarpy, 2015-11-25:
Hi Bill.  While looking at my configuration I noticed I use jom instead of nmake.  Under Options, Build & Run General tab, I have checked (I didn't actually check it: it was checked by default.) Use jom instead of nmake.

dcarpy, 2015-11-25:
Is glu32.dll in C:\Windows\System32?  I wonder where that sucker comes from, either Windows or the video card manufacturer...hmm.

dcarpy, 2015-11-25:
Never mind about jom vs. nmake, that should not have anything to do with your linking problem.

dcarpy, 2015-11-25:
If you have glu32.dll in C:\Windows\System32 then I would check your Path variable in Environment Variables under System Properties.  Here's mine:
Path = C:\ProgramData\Oracle\Java\javapath;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;%SystemRoot%\system32;%SystemRoot%;%SystemRoot%\System32\Wbem;%SYSTEMROOT%\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\Common Files\Acronis\SnapAPI\;D:\Program Files\Calibre2\;C:\Program Files\Microsoft SQL Server\110\Tools\Binn\;C:\Program Files\Microsoft SQL Server\120\Tools\Binn\;C:\Program Files (x86)\Git\cmd;C:\Program Files (x86)\Git\bin;D:\Program Files (x86)\QuickTime\QTSystem\;C:\Program Files (x86)\Microsoft SDKs\TypeScript\1.0\

Although, if you did'nt have ;%SystemRoot%\system32; in there nothing would work...

billbonney, 2015-11-25:
it's glu32.lib with is a lib required for linking, not a .dll

dcarpy, 2015-11-25:
Sorry about that Bill--not much help there.  :-)
I have glu32.lib in 7 places:
C:\Program Files (x86)\Windows Kits\8.1\Lib\winv6.3\um\x64
C:\Program Files (x86)\Windows Kits\8.1\Lib\winv6.3\um\x86
C:\Program Files (x86)\Windows Kits\8.0\Lib\win8\um\x64
C:\Program Files (x86)\Windows Kits\8.0\Lib\win8\um\x86
C:\Program Files (x86)\Microsoft SDKs\Windows\v7.0A\Lib
C:\Program Files (x86)\Microsoft SDKs\Windows\v7.1A\Lib
C:\Program Files (x86)\Microsoft SDKs\Windows\v7.1A\Lib\x64

billbonney, 2015-12-06:
I having building now with msvc2012 (on windows 10) 32bit. I still need to look at packaging etc..
```
qmake -spec win32-msvc2012 qgroundcontrol.pro
nmake
```

closed 2015-12-06 by billbonney