ArduPilot/apm_planner #844: New version of protobuf requires update to mavlink

spaceotter, 2015-11-30:
I'm getting a problem when building on Manjaro (Arch Linux derivative) in pixhawk.pb.h:
```
#error This file was generated by an older version of protoc which is
#error incompatible with your Protocol Buffer headers.  Please
#error regenerate this file with a newer version of protoc.
```
Ok, so how to regenerate headers? APM planner seems to depend on pixhawk.pb.h which vanished in this commit: https://github.com/mavlink/mavlink/commit/4a71e082efb7ee5915ea4a12a2881b64a38d11a8

billbonney, 2015-11-30:
I think you are including files in your build that are not in the linux build for ubuntu. Check the .pro file and make sure you are not building protobufs in Arch Linux.

spaceotter, 2015-11-30:
The only thing I have added is `MAVLINK_CONF += pixhawk` in user_config.pri. I'm trying to get the Arch PKGBUILD for APM Planner to work.

Without that option I instead get a different ton of build errors...
```
apm_planner-2.0.18/src/Waypoint.h:48:109: error: ‘MAV_CMD’ has not been declared
              bool autocontinue = true, bool current = false, MAV_FRAME frame=MAV_FRAME_GLOBAL_RELATIVE_ALT, MAV_CMD action=MAV_C
                                                                                                             ^
apm_planner-2.0.18/src/Waypoint.h:118:5: error: ‘MAV_CMD’ does not name a type
     MAV_CMD getAction() const {
     ^
```

billbonney, 2015-11-30:
You don't need to add that to the build. What are you trying to build exactly? Are you trying to buidl fro ArduPilot Flight Stack autopilots or Pixhawk Flight stack. aka PX4 flight stack. (this is not the same as Pixhawk HW which runs ArduPilot or PX4 flight stacks)

spaceotter, 2015-11-30:
I was wondering about that. Got it mixed up with the HW. I want to build APM planner, for the ArduPilot stack (that's currently installed) so I don't need the option. But what about those build errors?

billbonney, 2015-11-30:
We should chat on gitter, it will be easier
[image]

billbonney, 2015-11-30:
let's track this also we a new issue #845 

Closing this one

closed 2015-11-30 by billbonney