ArduPilot/apm_planner #845: Enable Building on ArchLinux (Update the readme.md)

billbonney, 2015-11-30:
Enable the required information to build on ArchLinux

spaceotter, 2015-12-01:
Build result:
```
In file included from src/uas/UASWaypointManager.h:38:0,
                 from src/uas/UASInterface.h:43,
                 from src/ui/MainWindow.h:45,
                 from src/QGCCore.h:38,
                 from src/main.cc:31:
apm_planner-2.0.18/src/Waypoint.h:48:109: error: ‘MAV_CMD’ has not been declared
              bool autocontinue = true, bool current = false, MAV_FRAME frame=MAV_FRAME_GLOBAL_RELATIVE_ALT, MAV_CMD action=MAV_C
                                                                                                             ^
apm_planner-2.0.18/src/Waypoint.h:118:5: error: ‘MAV_CMD’ does not name a type
     MAV_CMD getAction() const {
     ^
apm_planner-2.0.18/src/Waypoint.h:143:5: error: ‘MAV_CMD’ does not name a type
     MAV_CMD action;
     ^
apm_planner-2.0.18/src/Waypoint.h:168:20: error: ‘MAV_CMD’ has not been declared
     void setAction(MAV_CMD action);
                    ^
apm_planner-2.0.18/src/Waypoint.h:168:10: error: ‘void Waypoint::setAction(int)’ cannot be overloaded
     void setAction(MAV_CMD action);
          ^
apm_planner-2.0.18/src/Waypoint.h:167:10: error: with ‘void Waypoint::setAction(int)’
     void setAction(int action);
          ^
apm_planner-2.0.18/src/Waypoint.h:48:124: error: ‘MAV_CMD_NAV_WAYPOINT’ was not declared in this scope
        bool autocontinue = true, bool current = false, MAV_FRAME frame=MAV_FRAME_GLOBAL_RELATIVE_ALT, MAV_CMD action=MAV_CMD_NAV
                                                                                                                      ^
In file included from apm_planner-2.0.18/src/ui/MainWindow.h:45:0,
                 from apm_planner-2.0.18/src/QGCCore.h:38,
                 from apm_planner-2.0.18/src/main.cc:31:
apm_planner-2.0.18/src/uas/UASInterface.h:306:33: error: ‘MAV_CMD’ has not been declared
     virtual void executeCommand(MAV_CMD command) = 0;
                                 ^
apm_planner-2.0.18/src/uas/UASInterface.h:308:33: error: ‘MAV_CMD’ has not been declared
     virtual void executeCommand(MAV_CMD command, int confirmation, float param1, float param2, float param3, float param4, float
                                 ^
apm_planner-2.0.18/src/uas/UASInterface.h:694:56: error: ‘mavlink_sensor_offsets_t’ has not been declared
     void sensorOffsetsMessageUpdate(UASInterface *uas, mavlink_sensor_offsets_t sensorOffsets);
                                                        ^
apm_planner-2.0.18/src/uas/UASInterface.h:696:48: error: ‘mavlink_radio_t’ has not been declared
     void radioMessageUpdate(UASInterface *uas, mavlink_radio_t radioMessage);
                                                ^
apm_planner-2.0.18/src/uas/UASInterface.h:698:32: error: ‘mavlink_compassmot_status_t’ has not been declared
     void compassMotCalibration(mavlink_compassmot_status_t* compassmot_status);
                                ^
In file included from apm_planner-2.0.18/src/ui/JoystickWidget.h:37:0,
                 from apm_planner-2.0.18/src/ui/MainWindow.h:57,
                 from apm_planner-2.0.18/src/QGCCore.h:38,
                 from apm_planner-2.0.18/src/main.cc:31:
apm_planner-2.0.18/src/input/JoystickInput.h:64:36: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
     const int getNumberOfButtons() const;
                                    ^
In file included from apm_planner-2.0.18/src/uas/SlugsMAV.h:27:0,
                 from apm_planner-2.0.18/src/ui/SlugsDataSensorView.h:38,
                 from apm_planner-2.0.18/src/ui/MainWindow.h:72,
                 from apm_planner-2.0.18/src/QGCCore.h:38,
                 from apm_planner-2.0.18/src/main.cc:31:
apm_planner-2.0.18/src/uas/UAS.h:803:25: error: ‘MAV_CMD’ has not been declared
     void executeCommand(MAV_CMD command);
                         ^
apm_planner-2.0.18/src/uas/UAS.h:805:25: error: ‘MAV_CMD’ has not been declared
     void executeCommand(MAV_CMD command, int confirmation, float param1, float param2, float param3, float param4, float param5,
                         ^
Makefile:24218: recipe for target 'build-release/obj/main.o' failed
make: *** [build-release/obj/main.o] Error 1
```

spaceotter, 2015-12-10:
I modified the PKGBUILD for flite and apm_planner.

[apm_planner.zip](https://github.com/diydrones/apm_planner/files/58767/apm_planner.zip)