ArduPilot/apm_planner #847: Build install problem in Ubuntu 15.10

khancyr, 2015-12-07:
Hi,

There are some problems in installing apmplanner build in ubuntu 15.10. (the building is fine )

But I can't launch apmplanner after sudo make install. 
I got this error : 
[image]
This is related to wrong loading path : DEBUG 2015-12-07T12:46:36.725  QUrl( "file:///home/khancyr/sncf_ws/qml/PrimaryFlightDisplayQML.qml" )  

If I search where is this missing file, I got 
````
sudo find / -iname PrimaryFlightDisplayQML.qml/home/khancyr/Workspace/apm_planner/debian/tmp/usr/share/APMPlanner2/qml/PrimaryFlightDisplayQML.qml
/home/khancyr/Workspace/apm_planner/release/qml/PrimaryFlightDisplayQML.qml
/home/khancyr/Workspace/apm_planner/qml/PrimaryFlightDisplayQML.qml
/share/APMPlanner2/qml/PrimaryFlightDisplayQML.qml
````

The first time I run sudo make install , it produce : 
````
echo Copying files && cp -f -R /home/khancyr/Workspace/apm_planner/qml /home/khancyr/Workspace/apm_planner/release && cp -f -R /home/khancyr/Workspace/apm_planner/sik_uploader /home/khancyr/Workspace/apm_planner/release
Copying files
install -m 755 -p "release/apmplanner2" "/home/khancyr/Workspace/apm_planner/debian/tmp/usr/bin/apmplanner2"
cp -f -R /home/khancyr/Workspace/apm_planner/sik_uploader /home/khancyr/Workspace/apm_planner/debian/tmp/usr/bin/
install -m 644 -p /home/khancyr/Workspace/apm_planner/debian/apmplanner2 /home/khancyr/Workspace/apm_planner/debian/tmp/usr/share/menu/
strip "/home/khancyr/Workspace/apm_planner/debian/tmp/usr/bin/apmplanner2"
install -m 644 -p /home/khancyr/Workspace/apm_planner/debian/apmplanner2.desktop /home/khancyr/Workspace/apm_planner/debian/tmp/usr/share/applications/
cp -f -R /home/khancyr/Workspace/apm_planner/files /home/khancyr/Workspace/apm_planner/debian/tmp/usr/share/APMPlanner2/
cp -f -R /home/khancyr/Workspace/apm_planner/data /home/khancyr/Workspace/apm_planner/debian/tmp/usr/share/APMPlanner2/
cp -f -R /home/khancyr/Workspace/apm_planner/qml /home/khancyr/Workspace/apm_planner/debian/tmp/usr/share/APMPlanner2/
````

So file are not copied to /usr/share ....
Running sudo make install a second time do the job 
````
khancyr@khancyr-N551JQ:~/Workspace/apm_planner$ sudo make install -j4 -l4
install -m 755 -p "release/apmplanner2" "/bin/apmplanner2"
cp -f -R /home/khancyr/Workspace/apm_planner/sik_uploader /bin/
cp -f -R /home/khancyr/Workspace/apm_planner/files /share/APMPlanner2/
install -m 644 -p /home/khancyr/Workspace/apm_planner/debian/apmplanner2 /share/menu/
install -m 644 -p /home/khancyr/Workspace/apm_planner/debian/apmplanner2.desktop /share/applications/
strip "/bin/apmplanner2"
cp -f -R /home/khancyr/Workspace/apm_planner/data /share/APMPlanner2/
cp -f -R /home/khancyr/Workspace/apm_planner/qml /share/APMPlanner2/
````

But launching apmplanner2 from another directory that build directory doesn't work (cf QURL error)

I think it is related to apmplanner code more than install command. If you need more log, or that I test more commands, I will be glad to help

billbonney, 2015-12-07:
Which commit are you using?

the rules for loading qml files on linux are https://github.com/diydrones/apm_planner/blob/master/src/globalobject.cc#L199

khancyr, 2015-12-07:
I build from master with last commit : https://github.com/diydrones/apm_planner/commit/8a29a0dc780cdddb1784ead8bdb6fff35bad3109

khancyr, 2015-12-07:
Humm according to readme file are copied to /share/ ... But Qml loading doesn't look in it

billbonney, 2015-12-07:
And where are you executing apmplanner from?

khancyr, 2015-12-07:
Form home I get the error. If I am in the apmplanner build directory, it works

billbonney, 2015-12-07:
does it work when you run apmplanner from the /share/APMPlanner2/ dir?

khancyr, 2015-12-07:
Yep , it works. And I got
DEBUG 2015-12-07T18:06:55.301 qmlBaseDir "/share/APMPlanner2" 
DEBUG 2015-12-07T18:06:55.302  QUrl( "file:///share/APMPlanner2/qml/ApmToolBar.qml" )  

It works also from / , /tmp, /home...
But not from /home/khancyr ....

billbonney, 2015-12-09:
I've rasied an enhancement issue here #852 to update the lading of qml files when installed to the homedir.

closed 2015-12-09 by billbonney

felixsteghofer, 2016-04-06:
On Linux (Arch Linux) apmplanner2 is installed to `/share` and the binary directly to `/bin`.
On Linux this should be `/usr/`.. (`/usr/share, /usr/bin`) for user applications.
I checked the Makefile and found `INSTALL_ROOT` not to be defined.
If I define INSTALL_ROOT with `/usr` it is installed to `/usr/share`, `/usr/bin`.

After this modification the starting issue was fixed