ArduPilot/apm_planner #852: Linux: Fix loading of qml when executed from users home dir

billbonney, 2015-12-09:
see #847 Build issue when running apmplanner2 from home directory, qml files are not located

the rules for loading qml files on linux are https://github.com/diydrones/apm_planner/blob/master/src/globalobject.cc#L199

mrpilot2, 2015-12-11:
In my opinion the current implementation (using QDir::currentPath) is basically wrong. It allows to start a development build using installed config files and vice versa. e.g:
<code>cd /usr/share/APMPlanner2 && ~/workspace/apm_planner/release/apmplanner2/</code> loads the installed config files but uses the development application.

I think it should be possible to start the application from any directory and use the correct config files. Two use cases:
- Start installed application and use installed config files
- Start development application and use development config files

I'll have a look at this.

billbonney, 2015-12-11:
I agree, the loading on UNIX systems in AP2 is convoluted. I like your proposal