ArduPilot/apm_planner #857: Display servo outputs for channel 9 ~ 16

R-Lefebvre, 2015-12-22:
These 3 Commits:

https://github.com/R-Lefebvre/ardupilot/commits/MAVlink_Servo_Output_WIP

To fix this issue:

https://github.com/diydrones/ardupilot/issues/1968

Appear to break AP2.

Related issue on MP:

https://github.com/diydrones/MissionPlanner/issues/1090

billbonney, 2015-12-22:
How does this break APM Planner 2.0. It supports MAVLINK msg 35 and 65 (16 channel variant)

R-Lefebvre, 2015-12-24:
Bill, message 65 appears to be for RC Inputs, not Servo Outputs.

This change breaks AP2 in that, before the change, the servo outputs are shown correctly for channels 1-8 (ie: 1000-2000 on channels 1-4 and 8 for a helicopter).  After loading the change, AP2 does not display the servo outputs properly.  On my helicopter test-case, I'm not getting sensible values.  Channels 1-6 are 0, and 7-8 are 65000-something

billbonney, 2015-12-24:
That's going to be challenging since its essentially plotting values based on the internal state of one parameter in a message. We do have a 'filtered' message types that show in the InfoView. May that can be leveraged in the Graph View

billbonney, 2015-12-24:
MAVLINK msg 65 is for outputs, https://pixhawk.ethz.ch/mavlink/#RC_CHANNELS . Both 35 and 65 says "The PPM values of the RC channels received."

Maybe you can modify the code to send msg 65 instead for when more than 8 ch are used?

RC_CHANNELS_OVERRIDE is for inputs https://pixhawk.ethz.ch/mavlink/#RC_CHANNELS_OVERRIDE

billbonney, 2015-12-24:
My mistake you mean msg36 which is https://pixhawk.ethz.ch/mavlink/#SERVO_OUTPUT_RAW

See the above comment about decoding the message