ArduPilot/apm_planner #862: Program crashes when Joystick widget is called w/ joystick attached

zlite, 2015-12-31:
Using an XBox 360 controller and this driver (https://github.com/360Controller/360Controller/releases), APM Planner crashes when Joystick is selected. 

Here's the crash report: 

Process:               apmplanner2 [697]
Path:                  /Applications/APM Planner 2.0.app/Contents/MacOS/apmplanner2
Identifier:            com.diydrones.apmplanner2
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           apmplanner2 [697]
User ID:               501

Date/Time:             2015-12-30 16:12:03.948 -0800
OS Version:            Mac OS X 10.11.2 (15C50)
Report Version:        11
Anonymous UUID:        0A3DBD73-8F69-2B0B-C58B-01865D39E0E5

Time Awake Since Boot: 770 seconds

System Integrity Protection: disabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x00007ffc1168dd70
Exception Note:        EXC_CORPSE_NOTIFY

VM Regions Near 0x7ffc1168dd70:
    Stack                  00007000019aa000-0000700001a2c000 [  520K] rw-/rwx SM=COW  thread 20
--> 
    MALLOC_TINY            00007ffef9400000-00007ffef9800000 [ 4096K] rw-/rwx SM=PRV  

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_platform.dylib      	0x00007fffa0306851 _platform_memmove$VARIANT$Ivybridge + 49
1   com.diydrones.apmplanner2     	0x000000010b69445f QList<QLabel*>::detach_helper() + 95
2   com.diydrones.apmplanner2     	0x000000010b68f22d JoystickWidget::clearKeys() + 317
3   org.qt-project.QtCore         	0x000000010d46f3b3 QObject::event(QEvent*) + 755
4   org.qt-project.QtWidgets      	0x000000010c6dda4d QWidget::event(QEvent*) + 3677
5   org.qt-project.QtWidgets      	0x000000010c69dddb QApplicationPrivate::notify_helper(QObject*, QEvent*) + 251
6   org.qt-project.QtWidgets      	0x000000010c6a10f8 QApplication::notify(QObject*, QEvent*) + 8136
7   org.qt-project.QtCore         	0x000000010d444a7b QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) + 971
8   libqcocoa.dylib               	0x0000000112731c4e 0x112711000 + 134222
9   libqcocoa.dylib               	0x00000001127324d1 0x112711000 + 136401
10  com.apple.CoreFoundation      	0x00007fff9e4535c1 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
11  com.apple.CoreFoundation      	0x00007fff9e44541c __CFRunLoopDoSources0 + 556
12  com.apple.CoreFoundation      	0x00007fff9e44493f __CFRunLoopRun + 927
13  com.apple.CoreFoundation      	0x00007fff9e444338 CFRunLoopRunSpecific + 296
14  com.apple.HIToolbox           	0x00007fff93ea0935 RunCurrentEventLoopInMode + 235
15  com.apple.HIToolbox           	0x00007fff93ea0677 ReceiveNextEventCommon + 184
16  com.apple.HIToolbox           	0x00007fff93ea05af _BlockUntilNextEventMatchingListInModeWithFilter + 71
17  com.apple.AppKit              	0x00007fff92d910ee _DPSNextEvent + 1067
18  com.apple.AppKit              	0x00007fff9315d943 -[NSApplication _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 454
19  com.apple.AppKit              	0x00007fff92d86fc8 -[NSApplication run] + 682
20  libqcocoa.dylib               	0x00000001127313bd 0x112711000 + 132029
21  org.qt-project.QtCore         	0x000000010d44139d QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 381
22  org.qt-project.QtCore         	0x000000010d44437a QCoreApplication::exec() + 346
23  com.diydrones.apmplanner2     	0x000000010b5c8609 main + 601
24  com.diydrones.apmplanner2     	0x000000010b435894 start + 52

Thread 1:: Dispatch queue: com.apple.libdispatch-manager
0   libsystem_kernel.dylib        	0x00007fff9ab4dff6 kevent_qos + 10
1   libdispatch.dylib             	0x00007fff92981099 _dispatch_mgr_invoke + 216
2   libdispatch.dylib             	0x00007fff92980d01 _dispatch_mgr_thread + 52

Thread 2:: com.apple.NSEventThread
0   libsystem_kernel.dylib        	0x00007fff9ab47386 mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff9ab467c7 mach_msg + 55
2   com.apple.CoreFoundation      	0x00007fff9e445624 __CFRunLoopServiceMachPort + 212
3   com.apple.CoreFoundation      	0x00007fff9e444aec __CFRunLoopRun + 1356
[... 680 lines cut ...]
===========                     =======  ======= 
TOTAL                            882.5M      865 
TOTAL, minus reserved VM space   882.5M      865 

Model: MacBookAir5,2, BootROM MBA51.00EF.B04, 2 processors, Intel Core i7, 2 GHz, 8 GB, SMC 2.5f9
Graphics: Intel HD Graphics 4000, Intel HD Graphics 4000, Built-In
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1600 MHz, 0x802C, 0x384B54463531323634485A2D314736443120
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1600 MHz, 0x802C, 0x384B54463531323634485A2D314736443120
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0xE9), Broadcom BCM43xx 1.0 (7.21.94.136.1a1)
Bluetooth: Version 4.4.3f4 16616, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM512E, 500.28 GB
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: FaceTime HD Camera (Built-in)
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: Hub
USB Device: Card Reader
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: USB 3.0 Bus
USB Device: Controller
Thunderbolt Bus: MacBook Air, Apple Inc., 23.4

billbonney, 2015-12-31:
I just check out using the Taranis as the USB controller and it works. I don't have a XBOX 360 controller to test with unfortunately. I'll see if I can find one from somebody to test with

timdelbruegger, 2016-05-06:
I can confirm this behaviour on Windows 10 and the Xbox 360 Wireless controller.
1. Start APM Planner 2.0.20
2. "File"->"Joystick
3. Connect gamepad to PC by long-pressing the big X button -> APM Planner crashes

If the gamepad is connected before opening the Joystick view, the crash happens instantly when opening the Joystick view.

Toffe87, 2016-08-21:
+1 xbox 360 wired, when connected and selecting joystick option it crashes. Plugging it in after joystick window is opened, insta crash.. Kinda a while since this has been looked at?