ArduPilot/apm_planner #864: Graph speedup.

AndKe, 2016-01-01:
This commit works great:
https://github.com/diydrones/apm_planner/commit/1092d9af4ed40a5f13da0ffdde388820d333b3a6

please speed up the sort feature (which still takes the same time as before patch.)
and .log export.

If a certain log loaded in 21s before, filtered using "show sort"  in 21s before  , it takes now 8s to load, but still 21s to filter.

Arne-W, 2016-01-01:
As I wrote in comment on the commit - the export feature should be much faster already.
Regarding the speed of the filter I will have a look.

AndKe, 2016-01-01:
I commented on that comment, please check if the measured time is what you expected - too me it's seems like a very slow export.

Arne-W, 2016-01-01:
Okay - try the old version before #856 and you will see a slow export :-)
Regarding the filter speed I have an idea which could have an effect on exporting too...

Arne-W, 2016-01-02:
Using the actual master I got the following results:
Plot Log loading took 10.957 seconds - 15051872 of 15051872 bytes used (from log)
Exporting this log took round about 41 seconds. (measured with stopwatch)

If your log export is much slower we should have a look!

AndKe, 2016-01-02:
Apparently  I  fail to think of something that can be imported, and stashed into cells in 11 seconds, as anything else than *very slow* , when it takes 41s to just parse the same data to a file...  :)

Arne-W, 2016-01-03:
All right I am done..... Take a look at #867 
I could speed up the log exporting again now I am down to 22 sec instead of 41 sec. Okay I don't think this is incredibly fast - but fair :-)
The sorting feature should be faster now too...

Arne-W, 2016-01-21:
@billbonney If AndKe does not mind I think this issue can be closed.

AndKe, 2016-01-21:
I'll close it - it's working great.

closed 2016-01-21 by AndKe

Arne-W, 2016-01-21:
You are welcome - nice to hear that you like my changes!