ArduPilot/apm_planner #868: Graph: IMU.Acc? data not visible.

AndKe, 2016-01-05:
please see this post: http://ardupilot.com/forum/viewtopic.php?t=14691
The Acc data is not plotted on current master.

Arne-W, 2016-01-06:
On Linux I can not reproduce this (Kubuntu 15.10)

mboland, 2016-01-07:
This seems intermittent, as I downloaded some logs and the IMU data did display properly.
Same machine, same copter, next day flights and the IMU data did not display again.
Mac OS X 10.11.2

Arne-W, 2016-01-07:
@mboland could you test if the problem persists with the actual master and if it is still there upload a log witch shows the problem? I could test again then.

AndKe, 2016-01-07:
@Arne-W   - the log is here: http://ardupilot.com/forum/download/file.php?id=4850&sid=6f6ba2b99101805859f769fd34d75567  - and it's still an issue in current master - I had no seeing those data, from that particular file plotted, ever - tried at least 6 times.

mboland, 2016-01-07:
Sorry for dumb question but where would I download Master from?

AndKe, 2016-01-07:
@mboland right here, git clone https://github.com/diydrones/apm_planner.git  
- in case you look for compiled binary (may I guess that's the question?) 
it's supposed to be here, but something apparently broke and the build server makes only mac versions ?

billbonney, 2016-01-08:
prebuilt here for OSX http://firmware.diydrones.com/Tools/APMPlanner/daily/2016-01-06/

Arne-W, 2016-01-10:
@mboland - Just tested your log and you are right - no IMU data. The main problem is that your log data is corrupted! While loading your log the APM-Planner-logging says:
DEBUG 2016-01-10T12:59:46.617 AP2DataPlotThread::run(): Non packet bytes found in log file 1606 bytes filtered out. This may be a corrupt log 
DEBUG 2016-01-10T12:59:46.617 AP2DataPlotThread::run(): Non packet bytes found in log file 3526 bytes filtered out. This may be a corrupt log 

Opening your log with an HEX-editor shows a corrupt end (only FF FF FF FF....) and the APM-Planner table view shows "nan" - Not A Number in the last table line.
[image]
[image]

The second issue is in the code inserting the data into the internal database. The logline showing the "nan" shouldn't be stored because the generic plotting function interprets the "nan" as a string and therefore only an Arrow with "nan" is plotted and the rest of the data is discarded.

Perhaps we should open a popup - window if data is corrupt cause everybody should be aware that something is wrong with his logs.
Regarding the insertion bug I will have a look.

AndKe, 2016-01-10:
Yep, I did not doubt there were something fishy with that log, (which, BTW is not mine, just found in a support case)
The thing is, the corruption is handled well by the grid, allegedly it plots fine in MissionPlanner, and also is processed ok by mavlogdump.

So it would be best if AP2 been able to handle/ignore such issues. 
And yes, a visible notification, or popup, would be useful.

Arne-W, 2016-01-10:
Okay I made a bugfix for it see #874. The data should be plotted now because the corrupted data will not be inserted any more. If loading a corrupt file a popup will be shown.
Any further Ideas??

billbonney, 2016-01-10:
I think its good. My only comment is that if its the end of the file, lets not show the current suggested warning, but be more specific that it was the end of the file. i.e. if it's the last insert we say "file truncated - warning not all data may have been saved" as this error is more likely and less problematic than corruption within the file which is potentially a more serious problem. (loss of data mid log indicates power issue for example at the end it's just likely the power was pulled after landing)

Thanks for fixing though, this is great. :+1:

Arne-W, 2016-01-10:
Good Idea to make the error message more specific. I have already started implementation....
Created a class for delivering an error collection to the callee.
I think I will have enough time to finish it during this week.

commit 2c950461 (2016-01-10): Graph View: Better handling of corrupted logfiles Fixes #868

* Increased severity of popup window.
* Added handling for "not a number" while parsing binary logs.
* Make use of m_errorCount to feed back number of errors detected.
closed 2016-01-10 by billbonney