ArduPilot/apm_planner #870: Graph: not all Events/error are marked in the log

AndKe, 2016-01-05:
please see attached log.
only the first event/log are graphed in the log, but there are many more.

[10.BIN.zip](https://github.com/diydrones/apm_planner/files/78516/10.BIN.zip)

billbonney, 2016-01-05:
@Arne-W Maybe you would like to look at this issue, as I'm a bit busy for the next few days.

AndKe, 2016-01-05:
I kind of remember enough of that code to have a clue what might be wrong,
will check later today.

Arne-W, 2016-01-05:
@billbonney I will have a look - hopefully tomorrow

AndKe, 2016-01-05:
additional info:
it seems only first error shows, and *no* events. another log full of events (none plotted) made by another user is here: https://www.dropbox.com/s/nzo3ds50w2pfii5/16-1-4_15-14-34.zip?dl=0

Arne-W, 2016-01-06:
At the moment only plotting a graph of events is implemented. The "show EV" button has no effect, it is the next feature on my list :-)
Regarding the error plotting issue the implementation suppresses identical error messages following each other directly to prevent polluting. 
In the 10.bin I found 4 identical error events - so only the first has an arrow.
Perhaps the suppression is not a good solution. I derived it from the mode implementation where it makes sense because there are often repeated mode events.
 I think the best solution could be to use the relations between Mode, Err and EV while plotting - but until then a lot work has to be done :-)

AndKe, 2016-01-07:
the repeat should be disabled, it can hide important errors and only display one, by only displaying them one, before takeoff.

billbonney, 2016-01-07:
I don't think we should remove any messages from the log, including mode changes as they are indicative of an issue and sometime are there for good reason. I'll look at ArduPilot code and try and understand the extra mode change logging.

Arne-W, 2016-01-07:
Good idea. But as mentioned above I think the only way is to plot  all Err, Mode and EV messages in one run and handle their relation.
It looks like the relation between those messages show which of them can be suppressed because their content is redundant.

closed 2016-02-06 by billbonney

AndKe, 2016-02-06:
Works great, I like the use of colours.
would you please add MSG messages (strings) in the same way ?  very useful for planes. 
attaching screenshot to illustrate the kind of data we would see. 
(please say so if you wish this feature request as a new issue)
[image]

Arne-W, 2016-02-07:
Nice to hear you like it :-)
Should not be a problem to realize this. It would be great if you could open a new issue for this as it becomes more transparent who is working on which features then. 

@billbonney If AndKe creates a new issue you can assign it to me.

AndKe, 2016-02-07:
https://github.com/diydrones/apm_planner/issues/898 created.   :)

commit 0b579a81 (2016-02-06): Graph View: Implemented Event plotting and unified handling of "MODE,ERR,EV" messages Fixes #870

* Added funtionality to 'show EV' button
* Unified handling of 'MODE,ERR,EV' messages by using simple inheritance
  * Added abstract base class for all messages
  * Implemeted special types for 'MODE, ERR and EV' messages
* Moved old Mode enums into new namespace
* Colorized plotted arrows
* Verified all printed messages against apm code. MODE for all types (copter, plane, rover), ERR and EV for copter only.

* Removed Bugfix for 'Graph View: Fixed TableView does not start at 0' -> does not work - [... cut]