ArduPilot/apm_planner #880: Can not build with Ubuntu 14.04 LTS

mirkix, 2016-01-25:
Unfortunately I can not build apm_planner with Ubuntu 14.04 LTS. Always get the error message `Project ERROR: Unknown module(s) in QT: quickwidgets` when running `qmake qgroundcontrol.pro`. `sudo apt-get install libqt5quickwidgets5` does not fix it.

billbonney, 2016-01-26:
I'll need to create a vm to hcek this out, but I don't see the issue. Are you using Qt5.4.2?

mirkix, 2016-01-27:
@billbonney Seems to be 5.2.1 comes with Ubuntu 14.04 LTS. Followed the instruction here https://github.com/diydrones/apm_planner#linux- .

billbonney, 2016-01-27:
Fixed to refer to correct version. Will update with instructions on how to install Qt5.4.2 or greater soon

mirkix, 2016-01-27:
@billbonney Thanks!

billbonney, 2016-01-27:
@mirkix Let's keep this open until i fully update the README.md

ustramooner, 2016-02-04:
i got this working on 14.04 - but needed to remove quickwidgets from qgroundcontrol.pro

mirkix, 2016-02-04:
@ustramooner Thanks!

mstoker, 2016-02-08:
I'm also trying to build on 14.04 and get this error. libqt5quickwidgets5 does not appear to be available for 14.04. Does it require a later version, or is there a version of the library for 14.04 somewhere?

billbonney, 2016-02-08:
download.qt.io for downloading newer Qt5.4.2 or even Qt5.5.1 works

14.04 LTS comes with Qt5.2.1, which isn't supported anymore by this project.

markwbrown, 2016-04-22:
@billbonney Is  qt-everywhere-opensource-src-5.4.2.tar.gz the appropriate src to build from?

billbonney, 2016-04-22:
Yep, you can build Qt from source http://doc.qt.io/qt-5/linux-building.html

Also refer to https://wiki.qt.io/Building_Qt_5_from_Git as it can be helpful

billbonney, 2016-04-22:
Ps, you can also install QT from a ppa https://launchpad.net/~beineri

markwbrown, 2016-04-22:
@billbonney ppa looks nice, but it looks like Stephan's builds are for i386 and AMD64 only. Does not look like there is something for an ARM-based Chromebook

billbonney, 2016-04-22:
I doubt it, building Qt from source isn't so hard :)

markwbrown, 2016-04-23:
@billbonney On a Samsung Chromebook with Ubuntu trusty & kde installed through the crouton script, and the source pulled from git on the 5.4.2 branch, the chromebook runs out of space compiling using the wiki.qt.io link instructions. That seems unreasonable. Perhaps you have a suggestion, or I should take that problem to the Qt repo?

markwbrown, 2016-04-23:
I guess: are there specific dependencies apm planner needs from qt? To save space on the chromebook, would it be possible to just configure/make only what is needed to compile apm planner?

markwbrown, 2016-04-23:
I installed Qt on my mac, but it seems it is only able to cross compile for ARM if it is android, not ubuntu. So, I put a 32 gig SD in my chromebook, and got the 5.4.2 branch. It has been compiling Qt for about 10 hours now, and I'm at 6 gigs used on the SD card. Does APM planner depend on QuickWidgets? It seems from other threads there is some problem getting that in the 5.4.2 branch.

markwbrown, 2016-04-24:
Well, it's compiled and opens up. Need to add user to dialout to see if it talks to the telemetry / if an arm chromebook is up to the task.

billbonney, 2016-04-24:
Here's instructions on how to cross compile Qt5 Apps for Pi https://wiki.qt.io/RaspberryPi_Beginners_Guide

markwbrown, 2016-04-24:
Just plugged in the telemetry module, which shows up as ttyUSB0 in apm planner. Does not seem to work.

Arne-W, 2020-05-17:
outdated

closed 2020-05-17 by Arne-W