ArduPilot/apm_planner #881: When 'Use time on x-axis selected, no h-scrollbar

billbonney, 2016-01-26:
After this pull request https://github.com/diydrones/apm_planner/pull/879

When swicthing to time base x-axis no h-scrollbar is avaiable.
Also it clears the current graphing selection

@Arne-W WOuld you like to take a look?

Arne-W, 2016-01-27:
Yes for sure I will have a look ;-)
The horizontal scrollbar was not easy to handle during development and so I am not really surprised that there are some issues about it.
On Kubuntu 15.10 I do not have any problems with the h-scrollbar. After switching to time-axis it gets centered and scales correctly when using zoom.
So just let me know on which system you observed the bug.

The clearing of the graph is done by intention because when switching to time-axis all graphs have to be replotted due to the new x-scaling. It should be possible to store the selection and use it again after switching - I thought about that too but switching between Index and time axis is not the main use case I thought. Normally you will switch the Axis only once and use time or index for analysis.
But I can implement it if it is wanted.

billbonney, 2016-01-27:
it was Xubuntu 15.10
its _not_ an issue OSX haven't tested Windows.
I'll do some looking. Strange

Arne-W, 2016-01-28:
Ahh yes you are right - for finding the position in the Table the Index notation would be better. 
But wouldn't it be better if you could click into the graph and the table then jumps to the position where you clicked? I'll have a look if it is possible.

Arne-W, 2016-01-28:
yes it is possible :-) If using index view it is very simple due to the fact that the graph index corresponds to the table index. 
For the time view we have to extend the index of the data model to include the timestamp to get a relation between index and time.

Arne-W, 2016-01-28:
I made a quick hack to make the graph click able. At the moment it does only work in index view, but if you like it I will make it for the time view too.
[MakeGraphClickable_wip](https://github.com/Arne-W/apm_planner/tree/MakeGraphClickable_wip)

Arne-W, 2016-01-29:
I made a PR #885 which fixes the graph clearing. Also I did some intensive testing oft scroller but it always works on my machine.

billbonney, 2016-01-29:
I'll check it out later. I think it maybe a Xubuntu (Xfce) issue.

commit b0d0fa09 (2016-02-01): Graph View: Graph selection will be saved when switching x-Axis. Partly fixes #881
closed 2016-02-01 by billbonney