ArduPilot/apm_planner #882: Accel calibration does not work (ArduPlane 3.4.0-release)

AndKe, 2016-01-27:
I discovered earlier today that none of three newly built planes goes thru calibration.
The timeout timer is running, but no instructions for different directions appear.

Then I see  rmackay9    write this at 12:07 today:
------------
Random tidbit of knowledge from nowhere.. I think our accel calibration feature is too accepting of ACKs before it progresses to the next step of the calibration. It seems that any ACK will cause it to proceed but it should probably be more selective. For example it should probalby at least check that the "result" field is "MAV_RESULT_ACCEPTED". I think this lack of checks leads to some complaints from users that they can't do the accel calibration because it just keeps progressing through the stages before they've had a chance to position the vehicle properly.
----------

not sure if this is related, because I don't see any progress, but I've seen a step be "skipped" depending on click/press. 

To reproduce, try do accel calibration on USB connection (maybe telemetry is slower/more forgiving)

Sulter, 2016-02-04:
For me, this issue only comes up when using the keyboard. It can be avoided by using the mouse to press the "continue" button.

AndKe, 2016-02-08:
same problem, a little different look for Arducopter - 3.2 (release)
APM: Initialising APM...
APM: Initialising APM...
then it fails.

billbonney, 2016-02-08:
Everytime I check this, annoyingly it works, mouse or spacebar

```
DEBUG 2016-02-08T11:26:15.956 APM: Text Message rx'd "SUCCESS: Executed CMD: 241" 
DEBUG 2016-02-08T11:26:15.956 Rejecting message, check mismatch: current_state:  0  ==  3 , system id  0  ==  1 , comp id  1  ==  1 
 INFO 2016-02-08T11:26:15.957 STATUS TEXT: 2 : "Place vehicle level and press any key." 
DEBUG 2016-02-08T11:26:15.957 APM: Text Message rx'd "Place vehicle level and press any key." 
 INFO 2016-02-08T11:26:17.254 STATUS TEXT: 2 : "PreArm: RC not calibrated" 
DEBUG 2016-02-08T11:26:17.255 APM: Text Message rx'd "PreArm: RC not calibrated" 
 INFO 2016-02-08T11:26:19.154 STATUS TEXT: 2 : "Place vehicle on its LEFT side and press any key." 
DEBUG 2016-02-08T11:26:19.154 APM: Text Message rx'd "Place vehicle on its LEFT side and press any key." 
 INFO 2016-02-08T11:26:23.558 STATUS TEXT: 2 : "Place vehicle on its RIGHT side and press any key." 
DEBUG 2016-02-08T11:26:23.558 APM: Text Message rx'd "Place vehicle on its RIGHT side and press any key." 
 INFO 2016-02-08T11:26:25.960 STATUS TEXT: 2 : "Place vehicle nose DOWN and press any key." 
DEBUG 2016-02-08T11:26:25.961 APM: Text Message rx'd "Place vehicle nose DOWN and press any key." 
 INFO 2016-02-08T11:26:29.063 STATUS TEXT: 2 : "Place vehicle nose UP and press any key." 
DEBUG 2016-02-08T11:26:29.063 APM: Text Message rx'd "Place vehicle nose UP and press any key." 
 INFO 2016-02-08T11:26:31.765 STATUS TEXT: 2 : "Place vehicle on its BACK and press any key." 
DEBUG 2016-02-08T11:26:31.765 APM: Text Message rx'd "Place vehicle on its BACK and press any key." 
DEBUG 2016-02-08T11:26:34.455 Param: "INS_ACCOFFS_X" :  QVariant(double, 0.074614) 
DEBUG 2016-02-08T11:26:34.455 Param: "INS_ACCOFFS_Y" :  QVariant(double, -1.63021) 
DEBUG 2016-02-08T11:26:34.456 Param: "INS_ACCOFFS_Z" :  QVariant(double, -0.213929) 
DEBUG 2016-02-08T11:26:34.456 Param: "INS_ACCSCAL_X" :  QVariant(double, 1.01177) 
DEBUG 2016-02-08T11:26:34.456 Param: "INS_ACCSCAL_Y" :  QVariant(double, 1.0406) 
DEBUG 2016-02-08T11:26:34.457 Param: "INS_ACCSCAL_Z" :  QVariant(double, 1.01246) 
DEBUG 2016-02-08T11:26:34.457 Param: "INS_ACC2OFFS_X" :  QVariant(double, 0.00151297) 
DEBUG 2016-02-08T11:26:34.457 Param: "INS_ACC2OFFS_Y" :  QVariant(double, 0.230846) 
DEBUG 2016-02-08T11:26:34.457 Param: "INS_ACC2OFFS_Z" :  QVariant(double, -0.0224458) 
DEBUG 2016-02-08T11:26:34.458 Param: "INS_ACC2SCAL_X" :  QVariant(double, 0.998479) 
DEBUG 2016-02-08T11:26:34.458 Param: "INS_ACC2SCAL_Y" :  QVariant(double, 0.99603) 
DEBUG 2016-02-08T11:26:34.458 Param: "INS_ACC2SCAL_Z" :  QVariant(double, 0.994564) 
DEBUG 2016-02-08T11:26:34.458 Param: "AHRS_TRIM_X" :  QVariant(double, -0.0373994) 
DEBUG 2016-02-08T11:26:34.458 Param: "AHRS_TRIM_Y" :  QVariant(double, 0.0149773) 
DEBUG 2016-02-08T11:26:34.459 Param: "AHRS_TRIM_Z" :  QVariant(double, 0) 
 INFO 2016-02-08T11:26:34.553 STATUS TEXT: 2 : "Calibration successful" 
DEBUG 2016-02-08T11:26:34.554 APM: Text Message rx'd "Calibration successful" 
 INFO 2016-02-08T11:26:47.261 STATUS TEXT: 2 : "PreArm: RC not calibrated" 
DEBUG 2016-02-08T11:26:47.261 APM: Text Message rx'd "PreArm: RC not calibrated" 
```

AndKe, 2016-02-08:
just installed AP 3.5.0 (release) on an (3DR) Pixhawk. (no peripherals, just a pixhawk.)
changed FORMAT_VERSION to force a default, clean setup.
Yes, I successfully completed calibration. , both acc and mag.
-then I loaded a valid setup - completed acc calibration fine.

uploaded Arducopter. 3.3.2(release)  (default parameters)
```
now something interesting happened:
*before* timeout I git this log:
qml: APMToolBar: INITIAL SETUP unselected
qml: APMToolBar: INITIAL SETUP SELECTED
 INFO 2016-02-08T22:49:28.218 Unable to decode message from system  1  with message id: 178 
DEBUG 2016-02-08T22:49:34.851 UAS::executeCommand 241 conf 0 param1 0 param2 0 param3 0 param4 0 param5 1 param6 0 param7 0 
" NFO 2016-02-08T22:49:34.867 STATUS TEXT: 3 : "Place vehicle level and press any key.
" BUG 2016-02-08T22:49:34.868 APM: Text Message rx'd "Place vehicle level and press any key.
 INFO 2016-02-08T22:49:34.870 STATUS TEXT: 1 : "Initialising APM..." 
DEBUG 2016-02-08T22:49:34.870 APM: Text Message rx'd "Initialising APM..." 
 INFO 2016-02-08T22:49:39.867 STATUS TEXT: 1 : "Initialising APM..." 
DEBUG 2016-02-08T22:49:39.868 APM: Text Message rx'd "Initialising APM..." 
 INFO 2016-02-08T22:49:44.868 STATUS TEXT: 1 : "Initialising APM..." 
DEBUG 2016-02-08T22:49:44.868 APM: Text Message rx'd "Initialising APM..." 
 INFO 2016-02-08T22:49:49.868 STATUS TEXT: 1 : "Initialising APM..." 
DEBUG 2016-02-08T22:49:49.868 APM: Text Message rx'd "Initialising APM..." 
 INFO 2016-02-08T22:49:54.869 STATUS TEXT: 1 : "Initialising APM..." 
DEBUG 2016-02-08T22:49:54.869 APM: Text Message rx'd "Initialising APM..." 
 INFO 2016-02-08T22:49:59.869 STATUS TEXT: 1 : "Initialising APM..." 
DEBUG 2016-02-08T22:49:59.869 APM: Text Message rx'd "Initialising APM..." 
 INFO 2016-02-08T22:50:04.869 STATUS TEXT: 1 : "Initialising APM..." 
DEBUG 2016-02-08T22:50:04.870 APM: Text Message rx'd "Initialising APM..." 
" NFO 2016-02-08T22:50:04.870 STATUS TEXT: 3 : "Calibration FAILED
" BUG 2016-02-08T22:50:04.871 APM: Text Message rx'd "Calibration FAILED
DEBUG 2016-02-08T22:50:04.872 APM: Text Message rx'd "FAILURE: Failed CMD: 241" 
DEBUG 2016-02-08T22:50:04.873 Rejecting message, check mismatch: current_state:  0  ==  3 , system id  0  ==  1 , comp id  1  ==  1 
qml: HBD: Heartbeat stopped
qml: HBD: Heartbeat started
 INFO 2016-02-08T22:50:10.436 STATUS TEXT: 3 : "PreArm: RC not calibrated" 
DEBUG 2016-02-08T22:50:10.436 APM: Text Message rx'd "PreArm: RC not calibrated"
```
also, the "place level and press any key" did not show up in GUI
reproduced two times in row (restarted APM_Planner inbetween.)
- please note that pixhawk is *not* rebooting (based on LED flashing) - and "initilizing APM" is being displayed only until the calibration times out.

andriko13, 2016-02-19:
We're getting the same problem with UDP connection. Starting acceleration calibration only brings up the timeout timer and doesn't show any instructions.

AndKe, 2016-02-19:
Extra information:
I see now that this is random, sometimes it will succeed, most times not. Tested on >4 pixhawks with newly flashed firmware and default settings, as well as pixhawks loaded with working configuration from other RPAS.

minid33, 2016-02-19:
Same for me using v2.0.18 and a erle pxfmini on an rpi. This is a direct wifi connection where the rpi is the wifi AP. Connecting in APM planner using UDP. I see it every time without exception.

AndKe, 2016-02-19:
Another clue:  
I observe similar "initializing APM..." in mavproxy too, but it works fine there. (but you do need to remember that it's not enough to "press any key" in mavlink, only enter sends..)
```
Ready to FLY fence breach
APM: APM:Copter V3.3.2 (7f16e4d6)
APM: PX4: 34e1d543 NuttX: 7c5ef883
APM: Frame: QUAD
APM: PX4v2 0022002D 35324718 36343032
Received 505 parameters
Saved 505 parameters to mav.parm
APM: PreArm: RC not calibrated
accl
STABILIZE> Unknown command 'accl'
accelcal
STABILIZE> APM: Place vehicle level and press any key.
APM: Initialising APM...
 APM: Initialising APM...
 
STABILIZE> APM: Initialising APM...

STABILIZE> APM: Place vehicle on its LEFT side and press any key.
APM: Initialising APM...
APM: Initialising APM...

STABILIZE> APM: Place vehicle on its RIGHT side and press any key.
APM: Initialising APM...

STABILIZE> APM: Place vehicle nose DOWN and press any key.
APM: Initialising APM...

STABILIZE> APM: Place vehicle nose UP and press any key.
APM: Initialising APM...

STABILIZE> APM: Place vehicle on its BACK and press any key.
APM: Initialising APM...

STABILIZE> APM: AccelRange: 19.7 19.8 19.8
APM: Offsets[0]: -0.05 -0.23 -0.23
APM: Scaling[0]: 0.99 0.99 0.98
APM: AccelRange: 19.3 19.2 19.2
APM: Offsets[1]: 0.19 0.24 0.55
APM: Scaling[1]: 1.01 1.02 1.02
APM: Calibration successful
Trim OK: roll=-0.06 pitch=0.88
Got MAVLink msg: COMMAND_ACK {command : 241, result : 0}
Calibrated

```

Also, "FAILED cmd 241" does not stop the timer countdown.

AndKe, 2016-02-19:
I found the bug, fill submit a fix very soon.

AndKe, 2016-02-19:
https://github.com/diydrones/apm_planner/pull/902   fixes it.

AndKe, 2023-07-31:
fixed long time ago :)

closed 2023-07-31 by AndKe