ArduPilot/apm_planner #883: Compile error on Linux (CentOS 7 64bit)

hsteinhaus, 2016-01-28:
Compilation of master fails:

src/GAudioOutput.cc:401:12: error: 'flite_voice_list' was not declared in this scope
     for (v=flite_voice_list; v; v=val_cdr(v)) {

Regarding to grep, the symbol "flite_voice_list", does not appear in the whole source tree again, nor seem they to be an exported symbol of flite. 

I guess the problem was introduced by commit 4d67d34d7b6752b796198342f2de62c9c96be5b8 .

froller, 2016-09-01:
Same issue on Fedora 24

Evidlo, 2017-01-27:
I have this issue on Fedora 25.

iamjkp, 2017-01-30:
Same issue on Fedora 25. 

I am removing 2 lines in apm_planner.pro file and then compiling is successful. 

I reference above guys comments. Just remove `DEFINES += FLITE_AUDIO_ENABLED` code in apm_planner.pro. Everything will work out. 

Thanks guys.

larsks, 2017-06-16:
@madflower I just ran into the same issue (`error: 'flite_voice_list' was not declared...`).  Is this just some sort of version skew in flite? Fedora 25 has flite 1.3.

cafeclimber, 2017-09-05:
Has this issue been fixed? Experiencing the same issue on Fedora 26

larsks, 2017-09-06:
This is fixed in #1065, but so far nobody has reviewed or merged that patch.

closed 2023-07-15 by Arne-W