ArduPilot/apm_planner #889: v2.0.19-rc5 looks for wrong firmware for PX4 board

mboland, 2016-02-04:
I have a couple of PixRacer boards and tried to load firmware onto them.
The dialogue:
There was an error while downloading the firmware.
Error number: 203
Error text: Error downloading http://firmware.diydrones.com/Copter/stable/PX4-quad/ArduCopter-v4.px4 - server replied: Not Found

The only firmware on the server is ArduCopter-v1 and ArduCopter-v2

I was able to load ArduPlane but nothing else.

There was an error while downloading the firmware.
Error number: 203
Error text: Error downloading http://firmware.diydrones.com/Rover/stable/PX4/APMrover2-v4.px4 - server replied: Not Found

The board is identified when plugged in as PX4 FMU v4.x
[image]

billbonney, 2016-02-04:
Right click on the beta button to load trunk, there's is no beta or stable builds yet

closed 2016-02-04 by billbonney

mboland, 2016-02-04:
Yes, I loaded Trunk, and the routine times out no matter how quickly I unplug and plug in the USB.
This is what I am left with:

[image]

It downloads the firmware but goes no further.
Do I HAVE to use QGroundControl for this?

billbonney, 2016-02-04:
Don't think QGC will download the latest APM FW yet, I'll have to look. 

I just re-tested on OSX/Linux using Qt5.4.2/5.5.1 builds and it works, maybe you need to wait for the erasing step to complete, it seems to take longer on v4.

Another option is http://firmware.diydrones.com/Tools/APMPlanner/daily/2016-02-03/ build today

You can always install using the custom button and download from http://firmware.diydrones.com/Copter/latest and that will work.

mboland, 2016-02-04:
Thanks Bill,
QGC did the job straight up but I would like to keep using APM Planner instead of switching between apps.
Downloaded the APMP and refreshed the board without incident.
It is very slow to download parameters, about 1/3 to 1/2 speed of the Pixhawk.

billbonney, 2016-02-04:
I'm glad you have it working. (I hope they have a beta release soon to make it safe to test fly)
That said, QGC has the same issue as AP2, you need to check the Advanced box, then select the 'developer (master)' fw option for it to work. Otherwise the same error occurs as it cannot find stable/beta releases
```
QGroundControl can upgrade the firmware on Pixhawk devices, 3DR Radios and PX4 Flow Smart Cameras.
Found device: Pixhawk
Connected to bootloader:
Version: 5
Board ID: 11
Flash size: 2080768
Downloading firmware...
From: http://firmware.diydrones.com/Copter/stable/PX4-quad/ArduCopter-v4.px4
Error during download. Error: 203
Upgrade cancelled
------------------------------------------
If upgrade failed, make sure to connect directly to a powered USB port on your computer, not through a USB hub. Also make sure you are only powered via USB not battery.
Download complete
```