ArduPilot/apm_planner #897: Some metrics can be hard to read or are cut off

ustramooner, 2016-02-06:
On the quick screen and status screen I've noticed some metrics can be very hard to read due to their colours (light grey on ever so slightly darker grey) or are cut off due to the title being too long. 

[image]
[image]
[image]

I thought perhaps putting tool tips over them would help:
https://github.com/diydrones/apm_planner/pull/892
https://github.com/diydrones/apm_planner/pull/890

billbonney, 2016-02-07:
Hi,
Maybe the solution is to show the name only for variables that names do not fit?

(NOTE: Control_Sensors_present is a bit flag is not used often in Quick Info View parameter in anycase)

billbonney, 2016-02-07:
The other point is that the Status Info View is a poor cousin to the MAVLinkInspector widget. It's a carry over from making APM planner 2.0 familiar to Mission Planner user. Maybe we should remove the Status view?

[image]