ArduPilot/apm_planner #899: APM Planner 2 loses UDP connection after 30s or so

antonvh, 2016-02-07:
I have a wired connection to the Raspberry Pi sending the UDP packets so the network should be the issue. The funny thing is that UDP is a broadcast so there is no real 'connection'. Hitting 'disconnect' and 'connect' restores the connection immediately.  It would be nice if APM Planner did that automatically when some data is lost. (Running APM planner on mac os x)

billbonney, 2016-02-07:
This sounds strange. As there nothing making the connection close.  Sounds more like data from the Pi has stopped and restarting the socket triggers the send if request data and the Pi to start sending data again. Use the MAVLinkInspector to see data on the wire.

antonvh, 2016-02-08:
Thanks. Will do tonight.

antonvh, 2016-02-08:
I opened mavlink inspector  and made screenshot when the connection was lost and another one when the connection was regained after hitting connect/disconnect. What should I look for?

https://dl.dropboxusercontent.com/u/2042810/Screen%20Shot%202016-02-08%20at%2019.11.08.png
https://dl.dropboxusercontent.com/u/2042810/Screen%20Shot%202016-02-08%20at%2019.13.50.png

antonvh, 2016-02-08:
When using mavproxy I don't have disconnects. But the problem with mavproxy is that the GUI doesn't load on mac...

billbonney, 2016-02-08:
can you post the log files for AP2 from ~/apmplanner2/log.txt

antonvh, 2016-02-09:
There you go:
```
 INFO 2016-02-08T20:58:57.579 QGCCore::initialize() 
 INFO 2016-02-08T20:58:57.582 Current Build Info 
 INFO 2016-02-08T20:58:57.582 Git Hash: 51b6aa9218a0042744c31a7ea6f03752e5c9baa9 
 INFO 2016-02-08T20:58:57.582 Git Commit: 2.0.18 
 INFO 2016-02-08T20:58:57.582 APPLICATION_NAME: "APM Planner" 
 INFO 2016-02-08T20:58:57.582 APPLICATION_VERSION: "v2.0.18" 
 INFO 2016-02-08T20:58:57.582 APP_PLATFORM: osx 
 INFO 2016-02-08T20:58:57.582 APP_TYPE: stable 
 INFO 2016-02-08T20:58:58.215 Start Link Manager 
DEBUG 2016-02-08T20:58:58.216 Create MAVLinkDecoder:  MAVLinkDecoder(0x10cdcc000) 
 INFO 2016-02-08T20:58:58.224 Start UAS Manager 
DEBUG 2016-02-08T20:58:58.238 Creating MainWindow 
 INFO 2016-02-08T20:58:58.385 Creating SubMainWindow:  SubMainWindow(0x600000256560) 
 INFO 2016-02-08T20:58:58.385 setting objectName:  "VIEW_MISSION" 
 INFO 2016-02-08T20:58:58.449 Creating SubMainWindow:  SubMainWindow(0x60800024cb70) 
 INFO 2016-02-08T20:58:58.449 setting objectName:  "VIEW_FLIGHT" 
 INFO 2016-02-08T20:58:58.469 Creating SubMainWindow:  SubMainWindow(0x60800024ed00) 
 INFO 2016-02-08T20:58:58.469 setting objectName:  "VIEW_HARDWARE_CONFIG" 
DEBUG 2016-02-08T20:58:58.722 fillPortsInfo  
 INFO 2016-02-08T20:58:58.725 Inserting  "Rigo-COM1" 
 INFO 2016-02-08T20:58:58.725 Inserting  "Rigo-NokiaPCSuite" 
 INFO 2016-02-08T20:58:58.725 Inserting  "NXT-DevB" 
 INFO 2016-02-08T20:58:58.725 Inserting  "Bluetooth-Modem" 
 INFO 2016-02-08T20:58:58.725 Inserting  "Bluetooth-Incoming-Port" 
 INFO 2016-02-08T20:58:58.725 Inserting  "iPhonevanAnton-Wireless" 
 INFO 2016-02-08T20:58:58.725 Changed Link to: "Rigo-COM1" 
 INFO 2016-02-08T20:58:58.915 Creating SubMainWindow:  SubMainWindow(0x60000025d520) 
 INFO 2016-02-08T20:58:58.915 setting objectName:  "VIEW_SOFTWARE_CONFIG" 
DEBUG 2016-02-08T20:58:58.927 Disable Controls 
 INFO 2016-02-08T20:58:59.087 Creating SubMainWindow:  SubMainWindow(0x608000a4a890) 
 INFO 2016-02-08T20:58:59.087 setting objectName:  "VIEW_ENGINEER" 
 INFO 2016-02-08T20:58:59.114 Creating SubMainWindow:  SubMainWindow(0x608000a516a0) 
 INFO 2016-02-08T20:58:59.114 setting objectName:  "VIEW_SIMULATOR" 
DEBUG 2016-02-08T20:58:59.139 MODE: 0 
DEBUG 2016-02-08T20:58:59.140 "D|PREFLIGHT" 
DEBUG 2016-02-08T20:58:59.140 MODE: 80 
DEBUG 2016-02-08T20:58:59.140 "D|STABILIZED" 
DEBUG 2016-02-08T20:58:59.140 MODE: 64 
DEBUG 2016-02-08T20:58:59.141 "D|MANUAL" 
DEBUG 2016-02-08T20:58:59.141 MODE: 88 
DEBUG 2016-02-08T20:58:59.141 "D|VECTOR|STABILIZED" 
DEBUG 2016-02-08T20:58:59.141 MODE: 92 
DEBUG 2016-02-08T20:58:59.141 "D|AUTO|STABILIZED" 
DEBUG 2016-02-08T20:58:59.141 MODE: 66 
DEBUG 2016-02-08T20:58:59.142 "D|TEST" 
DEBUG 2016-02-08T20:58:59.197  QUrl( "file:///Users/anton/Applications/APM Planner 2.0.app/Contents/MacOS/qml/PrimaryFlightDisplayQML.qml" )  
DEBUG 2016-02-08T20:58:59.314 QML Status: 1 
DEBUG 2016-02-08T20:58:59.316  QUrl( "file:///Users/anton/Applications/APM Planner 2.0.app/Contents/MacOS/qml/PrimaryFlightDisplayQML.qml" )  
DEBUG 2016-02-08T20:58:59.349 QML Status: 1 
 INFO 2016-02-08T20:58:59.355 UASActionsWidget creating  UASActionsWidget(0x115521ec0) 
DEBUG 2016-02-08T20:58:59.358 QuickView: load settings 
DEBUG 2016-02-08T20:58:59.358 QuickView: loading key:  "GCS Status.Altitude (GPS) (m)"  type: "text" 
DEBUG 2016-02-08T20:58:59.360 QuickView: loading key:  "GCS Status.Altitude (REL) (m)"  type: "text" 
DEBUG 2016-02-08T20:58:59.364 QuickView: loading key:  "GCS Status.Battery (%)"  type: "text" 
DEBUG 2016-02-08T20:58:59.365 QuickView: loading key:  "GCS Status.Climb (m/s)"  type: "text" 
DEBUG 2016-02-08T20:58:59.367 QuickView: loading key:  "GCS Status.Current (A)"  type: "text" 
DEBUG 2016-02-08T20:58:59.368 QuickView: loading key:  "GCS Status.GPS Fix ()"  type: "text" 
DEBUG 2016-02-08T20:58:59.370 QuickView: loading key:  "GCS Status.GPS HDOP (m)"  type: "text" 
DEBUG 2016-02-08T20:58:59.373 QuickView: loading key:  "GCS Status.GPS Sats ()"  type: "text" 
DEBUG 2016-02-08T20:58:59.375 QuickView: loading key:  "GCS Status.Pitch (deg)"  type: "text" 
DEBUG 2016-02-08T20:58:59.378 QuickView: loading key:  "GCS Status.Roll (deg)"  type: "text" 
DEBUG 2016-02-08T20:58:59.380 QuickView: loading key:  "GCS Status.Voltage (V)"  type: "text" 
DEBUG 2016-02-08T20:58:59.383 QuickView: loading key:  "GCS Status.Yaw (deg)"  type: "text" 
DEBUG 2016-02-08T20:58:59.387 QuickView: load settings END 
DEBUG 2016-02-08T20:58:59.467 qmlBaseDir "/Users/anton/Applications/APM Planner 2.0.app/Contents/MacOS" 
DEBUG 2016-02-08T20:58:59.467  QUrl( "file:///Users/anton/Applications/APM Planner 2.0.app/Contents/MacOS/qml/ApmToolBar.qml" )  
DEBUG 2016-02-08T20:58:59.490 QML Status: 1 
DEBUG 2016-02-08T20:58:59.496 Loading widget: "PRIMARY_FLIGHT_DISPLAY_QML_DOCKWIDGET" 
[... 194 lines cut ...]
DEBUG 2016-02-08T20:59:11.874 FlightModeConfig:: name: "FLTMODE2"  value: QVariant(int, 11) 
DEBUG 2016-02-08T20:59:11.874 Param: "FLTMODE2" :  QVariant(int, 11) 
DEBUG 2016-02-08T20:59:11.975 Param: "RC7_MAX" :  QVariant(int, 1900) 
DEBUG 2016-02-08T20:59:11.977 Param: "RC7_REV" :  QVariant(int, 1) 
DEBUG 2016-02-08T20:59:11.978 Param: "RC7_DZ" :  QVariant(int, 0) 
DEBUG 2016-02-08T20:59:11.979 Param: "RC7_FUNCTION" :  QVariant(int, 0) 
DEBUG 2016-02-08T20:59:11.980 Param: "RC8_MIN" :  QVariant(int, 1100) 
DEBUG 2016-02-08T20:59:11.981 Param: "RC8_TRIM" :  QVariant(int, 1500) 
DEBUG 2016-02-08T20:59:11.982 Param: "RC8_MAX" :  QVariant(int, 1900) 
[... cut ...]
DEBUG 2016-02-08T20:59:12.210 Param: "RLL2SRV_IMAX" :  QVariant(int, 3000) 
DEBUG 2016-02-08T20:59:12.387 Param: "RLL2SRV_FF" :  QVariant(double, 0) 
DEBUG 2016-02-08T20:59:12.389 Param: "PTCH2SRV_TCONST" :  QVariant(double, 0.5) 
DEBUG 2016-02-08T20:59:12.390 Param: "PTCH2SRV_P" :  QVariant(double, 0.4) 
DEBUG 2016-02-08T20:59:12.391 Param: "PTCH2SRV_D" :  QVariant(double, 0.02) 
DEBUG 2016-02-08T20:59:12.391 Param: "PTCH2SRV_I" :  QVariant(double, 0.04) 
DEBUG 2016-02-08T20:59:12.392 Param: "PTCH2SRV_RMAX_UP" :  QVariant(int, 0) 
DE

ZeyadTamimi, 2016-02-15:
I have the same issue when receiving a UDP connection from mavproxy

antonvh, 2016-02-15:
Not much to see in this log. This is where the disconnect happens:
```
DEBUG 2016-02-08T21:08:03.625 APMToolBar: connectMAV  ArduPilotMegaMAV(0x10628b200) with sLink 1 
 INFO 2016-02-08T21:08:03.626 UDP disconnect 
 INFO 2016-02-08T21:08:03.626 ArduPilotMegaMAV APM disconnected 
DEBUG 2016-02-08T21:08:03.627 AHC: uasDisconnected() 
 INFO 2016-02-08T21:08:03.627 UASActionsWidget::disconnected() ArduPilotMegaMAV(0x10628b200)
```

I have also exported a transfer log. It's here: https://dl.dropboxusercontent.com/u/2042810/planelog.log
Looks pretty normal. Is there a way to have a timestamp in the transfer log so we can relate it to the events in the APM planner log?

Arne-W, 2020-05-17:
somehow outdated

closed 2020-05-17 by Arne-W