ArduPilot/apm_planner #901: Graph: use time on X-Axis should use smaller values than microseconds.

AndKe, 2016-02-18:
X-Axis time is great, but when a flashlog is loaded , it tends to display , for example values like:
[image]

please reduce the values down to seconds with one or two decimals, that will improve readability of the scale.

An flight, zoomed out, contains absurdly large and round numbers, like   0 , 800000000 160000000 240000000  and so on a further improvement would be to roughly scale the time according to x-scale, so the above example could show: 1,3min , 2,6min , 4min etc.

An ultimate feature would also be to be able to offset the time by GPS time, so user could switch between viewing time as "seconds since boot"  or  "actual time"

Arne-W, 2016-02-19:
Yes - I know the numbers are absurd but I had no Idea how to make a good scaling, as it should change when zooming in and out. Meanwhile I saw that we could get a much newer version of the graphic framework we are using. Perhaps we should first update the framework before implementing new features with the old one.... but I haven't verified whether we could simply change the framework or if we have to redesign - and if we have to redesign - how long will it take?!
But you are definitely right the scaling is terrible. 

Your 'ultimate feature idea' - nice idea by the way :-) - should work the same way as the scaling issue as it IS only another scaling. So implementing should not be a problem. 
But first we should update the framework. I will raise an issue about that.

Arne-W, 2016-03-06:
If you want you could check my new PR #912 - its a first simple approach. The time line is now in seconds which is far better than micro secs. The good thing is that this solution is very simple due to the fact the time line is handled still by the framework. Your feature requires to handle the x-axis values by ourselves....

closed 2016-05-24 by AndKe