ArduPilot/apm_planner #904: Graph View: Update graphing framework to actual version

Arne-W, 2016-02-19:
The version of the graphing framework we are using in Graph View is a little bit outdated. At the moment we are using V1.1.1 released at 09.12.2013 and the actual version is V1.3.2 released at 22.12.2015.
I think we should update to the new version before implementing new features on the old one. 

@billbonney What Do you think ?

billbonney, 2016-02-19:
If you submit a pull request with it updated I will merge it in. Just make sure you test the Elevation Widget as that also uses the graph lib

Arne-W, 2016-02-20:
That was easy :-)
I tested the Elevation widget and it looked normal to me - but I only have one tlog which I could use for testing. Perhaps somebody else should take a look.

commit cf444848 (2016-03-05): Merge pull request #906 from Arne-W/Update_Graphic_framwork

Updated graphic framework to QCustomPlot V1.3.2 - Fixes #904
closed 2016-03-05 by billbonney