ArduPilot/apm_planner #916: OSX APM - Crash on Log Open

Fireycold, 2016-03-13:
Any time I try and open a log that contains data, APM crashes. If the log is blank, it will open the log and display a blank log, but if there is any information contained in the file, the program will crash. Mission Planner on Windows does not have this issue, this issue occurs with about a dozen different logs, its not specific to a given log.

Using the 2.0.19 nightly build, and the current 2.0.18 release, the error still happens. 

Any help is much appreciated, thanks!

Process:               apmplanner2 [60811]
Path:                  /Applications/APM Planner 2.0.app/Contents/MacOS/apmplanner2
Identifier:            com.diydrones.apmplanner2
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           apmplanner2 [60811]
User ID:               501

Date/Time:             2016-03-13 13:14:38.883 -0700
OS Version:            Mac OS X 10.11.3 (15D21)
Report Version:        11
Anonymous UUID:        190A88C6-B204-C29A-57B4-C304406C2A49

Sleep/Wake UUID:       FA51ADA1-63B2-445F-96E0-DF7744E4AC88

Time Awake Since Boot: 1200000 seconds
Time Since Wake:       910000 seconds

System Integrity Protection: enabled

Crashed Thread:        42  AP2DataPlotThread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       EXC_I386_GPFLT
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
Performing @selector(ok:) from sender NSButton 0x7fbe98db7a50

Thread 0:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff87827386 mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff878267c7 mach_msg + 55
2   com.apple.CoreGraphics        	0x00007fff92120dfa _CGSGetWindowDepth + 113
3   com.apple.CoreGraphics        	0x00007fff92120d77 CGSWindowGetDepth + 70
4   com.apple.CoreGraphics        	0x00007fff924d7242 CGSGetWindowDepth + 38
5   com.apple.AppKit              	0x00007fff9d5d7772 -[_NSCGSWindowBackingStore depth] + 62
6   com.apple.AppKit              	0x00007fff9d5d817f -[_NSCGSWindowBackingStore lockBackBuffer] + 96
7   com.apple.AppKit              	0x00007fff9d1531fc backing_store_delegate + 71
8   com.apple.AppKit              	0x00007fff9d1524f5 backing_store_DrawImage + 525
9   com.apple.CoreGraphics        	0x00007fff9227718f CGContextDrawImageWithOptions + 684
10  com.apple.CoreGraphics        	0x00007fff9210190f CGContextDrawImage + 73
11  libqcocoa.dylib               	0x0000000105d67bf9 0x105d51000 + 93177
12  com.apple.AppKit              	0x00007fff9cd80c22 -[NSView _drawRect:clip:] + 3626
13  com.apple.AppKit              	0x00007fff9cd7f501 -[NSView _recursiveDisplayAllDirtyWithLockFocus:visRect:] + 1873
14  com.apple.AppKit              	0x00007fff9cd7f8de -[NSView _recursiveDisplayAllDirtyWithLockFocus:visRect:] + 2862
15  com.apple.AppKit              	0x00007fff9cd7d427 -[NSView _recursiveDisplayRectIfNeededIgnoringOpacity:isVisibleRect:rectIsVisibleRectForView:topView:] + 838
16  com.apple.AppKit              	0x00007fff9cd7cc0c -[NSThemeFrame _recursiveDisplayRectIfNeededIgnoringOpacity:isVisibleRect:rectIsVisibleRectForView:topView:] + 334
17  com.apple.AppKit              	0x00007fff9cd791ef -[NSView _displayRectIgnoringOpacity:isVisibleRect:rectIsVisibleRectForView:] + 2449
18  com.apple.AppKit              	0x00007fff9cd33971 -[NSView displayIfNeeded] + 1950
19  com.apple.AppKit              	0x00007fff9cd3319c -[NSWindow displayIfNeeded] + 232
20  com.apple.AppKit              	0x00007fff9d4af6da ___NSWindowGetDisplayCycleObserver_block_invoke6367 + 476
21  com.apple.AppKit              	0x00007fff9d0f60d6 __37+[NSDisplayCycle currentDisplayCycle]_block_invoke + 738
22  com.apple.QuartzCore          	0x00007fff888b3b5d CA::Transaction::run_commit_handlers(CATransactionPhase) + 85
23  com.apple.QuartzCore          	0x00007fff8874dc7c CA::Context::commit_transaction(CA::Transaction*) + 160
24  com.apple.QuartzCore          	0x00007fff8874da24 CA::Transaction::commit() + 508
25  com.apple.AppKit              	0x00007fff9ceed611 -[NSMoveHelper _doAnimation] + 67
26  com.apple.AppKit              	0x00007fff9cf07f21 -[NSMoveHelper(NSSheets) _closeSheet:andMoveParent:] + 559
27  com.apple.AppKit              	0x00007fff9cf07c93 -[NSWindow(NSSheets) _orderOutRelativeToWindow:] + 105
28  com.apple.AppKit              	0x00007fff9cdcffc0 -[NSWindow _reallyDoOrderWindow:relativeTo:findKey:forCounter:force:isModal:] + 2097
29  com.apple.AppKit              	0x00007fff9cdcf20d -[NSWindow _doOrderWindow:relativeTo:findKey:forCounter:force:isModal:] + 894
30  com.apple.AppKit              	0x00007fff9cdcee35 -[NSWindow orderWindow:relativeTo:] + 159
31  com.apple.AppKit              	0x00007fff9d33e4e9 -[NSSavePanel orderWindow:relativeTo:] + 80
32  com.apple.AppKit              	0x00007fff9d3323bc -[NSSavePanel orderOut:] + 77
[... 874 lines cut ...]
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1600 MHz, 0x802C, 0x384A54463235363634485A2D3147364D3120
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0xEF), Broadcom BCM43xx 1.0 (7.21.94.136.1a1)
Bluetooth: Version 4.4.3f4 16616, 3 services, 27 devices, 1 incoming serial ports
Network Service: Thunderbolt Ethernet, Ethernet, en3
PCI Card: Apple 57762-A0, Ethernet Controller, Thunderbolt@195,0,0
Serial ATA Device: APPLE SSD SM256E, 251 GB
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: FaceTime HD Camera (Built-in)
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: Hub
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: USB 3.0 Bus
USB Device: 4-Port USB 3.0 Hub
USB Device: HD-GDU3
USB Device: BUP Slim SL
USB Device: USB2.0 Hub
USB Device: Keyboard Hub
USB Device: SpaceMouse Wireless Receiver
USB Device: Apple Keyboard
Thunderbolt Bus: MacBook Pro, Apple Inc., 23.4
Thunderbolt Device: Thunderbolt to Gigabit Ethernet Adapter, Apple Inc., 3, 5.5

Arne-W, 2016-03-14:
Could you please upload one of the logs that leads to a crash?
It looks like you are loading a *.tlog. Does this happen on loading flash logs (*.bin) too?

I will take a look. On my Linux machine everything seems to work - but perhaps with your logfile I can reproduce the crash.

billbonney, 2016-03-14:
@Fireycold Please post a log, I just tried a 60Mbyte tlog file on OSX and it was fine

You can email me a link to log, if you want to keep it private. (see profile for email address)

Fireycold, 2016-03-14:
I don't have any .bin files to test it with, I have only been working with 3DR Solo generated .tlogs

If you guys have a log you know is working on your end, I can test it on mine.

[4 Logs.zip](https://github.com/diydrones/apm_planner/files/172797/4.Logs.zip)

Arne-W, 2016-03-15:
I could reproduce it and made a small fast fix - is not perfect but fixes the crash.
Thanks for the logs.

Fireycold, 2016-03-15:
When will these modifications be reflected in the daily build?

billbonney, 2016-03-15:
@Fireycold now ;)

Arne-W, 2017-01-30:
As the log handling is completely rewritten this bug should be fixed...

closed 2017-01-30 by Arne-W